wlcrs/huawei_solar #93: SmartLogger 300A

bdkacz, 2022-07-21:
### System Health details

## System Health

version | core-2022.7.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.45
arch | x86_64
timezone | Europe/Warsaw
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4870
Installed Version | 1.26.2
Stage | running
Available Repositories | 1072
Downloaded Repositories | 2

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.2
-- | --
update_channel | stable
supervisor_version | supervisor-2022.07.0
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 30.8 GB
disk_used | 2.9 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | SSH & Web Terminal (11.0.2), Log Viewer (0.14.0), File editor (5.3.3), Samba share (10.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 21 lipca 2022 06:31
-- | --
current_recorder_run | 21 lipca 2022 09:12
estimated_db_size | 0.43 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Huawei Solar Setup

2x Huawei SUN-60-KTL-M0 inverters connected via rs485 to SmartLogger 3000A with modbus-tcp enabled  on logger

### Describe the issue

I have a challenge for the author :)

I have a SolarLogger 3000A and 2 inverters connected to it (at address 1 and 2,  logger is at address 0), link to the modbus  [documentation](https://drive.google.com/file/d/1f_bBiPUk75cGkGFUHUnYxXfKOR-L1DCM/view?usp=sharing)

If I understand the provisions of chapter "2.7 Remapped Modbus definitions", the registers are shifted in relation to the situation when we connect directly to the inverter

Currently, an attempt to establish a connection under any slaveID (0,1,2) results in such records in the logs

### Reproduction steps

1. Install Addon
2. Try to add device

### Relevant debug logs

```text
2022-07-21 09:13:40 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting to 131.87.12.200:502.
2022-07-21 09:13:40 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-07-21 09:13:40 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server
2022-07-21 09:13:40 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-07-21 09:13:40 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 131.87.12.200:502.
2022-07-21 09:13:41 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2022-07-21 09:13:41 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0xa7 0xfe 0x0 0x1
2022-07-21 09:13:41 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2022-07-21 09:13:41 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x1 0x0 0x0 0x0 0x3 0x1 0x83 0x3
2022-07-21 09:13:41 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x1 0x0 0x0 0x0 0x3 0x1 0x83 0x3
2022-07-21 09:13:41 DEBUG (MainThread) [pymodbus.factory] Factory Response[131]
2022-07-21 09:13:41 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1
2022-07-21 09:13:41 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 123, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 88, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 217, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 259, in get_multiple
    response = await self._read_registers(
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 350, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 133, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 332, in _do_read
    raise ReadException(
huawei_solar.exceptions.ReadException: Got error while reading from register 43006 with length 1: Exception Response(131, 3, IllegalValue)
2022-07-21 09:13:41 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None
2022-07-21 09:13:41 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
```

wlcrs, 2022-07-21:
That is not straightforward, but should be doable. Any chance that we can work out a way for me to access your installation (VPN, Wireguard, whitelisting of an IP)? That would immensely speed up the development cycle.

wlcrs, 2022-07-21:
What happens if you only mention slave ID 1 and 2 during the installation wizard? Can you post the logs of what happens then?

bdkacz, 2022-07-21:
With slaveid =1:

```
2022-07-21 11:10:54 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting to 131.87.12.200:502.
2022-07-21 11:10:54 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-07-21 11:10:54 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server
2022-07-21 11:10:54 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-07-21 11:10:54 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 131.87.12.200:502.
2022-07-21 11:10:55 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2022-07-21 11:10:55 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0xa7 0xfe 0x0 0x1
2022-07-21 11:10:55 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2022-07-21 11:10:55 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x1 0x0 0x0 0x0 0x3 0x1 0x83 0x3
2022-07-21 11:10:55 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x1 0x0 0x0 0x0 0x3 0x1 0x83 0x3
2022-07-21 11:10:55 DEBUG (MainThread) [pymodbus.factory] Factory Response[131]
2022-07-21 11:10:55 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1
2022-07-21 11:10:55 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 123, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 88, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 217, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 259, in get_multiple
    response = await self._read_registers(
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 350, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 133, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 332, in _do_read
    raise ReadException(
huawei_solar.exceptions.ReadException: Got error while reading from register 43006 with length 1: Exception Response(131, 3, IllegalValue)
2022-07-21 11:10:55 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None
2022-07-21 11:10:55 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
```

with slaveid=2:

```
2022-07-21 11:13:09 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting to 131.87.12.200:502.
2022-07-21 11:13:09 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-07-21 11:13:09 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server
2022-07-21 11:13:09 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-07-21 11:13:09 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 131.87.12.200:502.
2022-07-21 11:13:09 DEBUG (MainThread) [zeroconf] IPv6 scope_id 0 associated to the receiving interface
2022-07-21 11:13:09 DEBUG (MainThread) [zeroconf] Received from '::ffff:192.168.2.72':5353 [socket 12 (('::', 5353, 0, 0))]: <DNSIncoming:{id=0, flags=0, truncated=False, n_q=1, n_ans=1, n_auth=0, n_add=0, questions=[ptr[question,QM,in,_anywhereusb._tcp.local.]], answers=[record[ptr,in,_anywhereusb._tcp.local.]=4205/4204,AW02-004853 [0004F31FC7AA] (AnywhereUSB)._anywhereusb._tcp.local.]}> (96 bytes) as [b'\x00\x00\x00\x00\x00\x01\x00\x01\x00\x00\x00\x00\x0c_anywhereusb\x04_tcp\x05local\x00\x00\x0c\x00\x01\xc0\x0c\x00\x0c\x00\x01\x00\x00\x10m\x00+(AW02-004853 [0004F31FC7AA] (AnywhereUSB)\xc0\x0c']
2022-07-21 11:13:10 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2022-07-21 11:13:10 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x2 0x3 0xa7 0xfe 0x0 0x1
2022-07-21 11:13:10 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2022-07-21 11:13:10 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x1 0x0 0x0 0x0 0x3 0x2 0x83 0x3
2022-07-21 11:13:10 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x1 0x0 0x0 0x0 0x3 0x2 0x83 0x3
2022-07-21 11:13:10 DEBUG (MainThread) [pymodbus.factory] Factory Response[131]
2022-07-21 11:13:10 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1
2022-07-21 11:13:10 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 123, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 88, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 217, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 259, in get_multiple
    response = await self._read_registers(
[... cut ...]
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 332, in _do_read
    raise ReadException(
huawei_solar.exceptions.ReadException: Got error while reading from register 43006 with length 1: Exception Response(131, 3, IllegalValue)
2022-07-21 11:13:10 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None
2022-07-21 11:13:10 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
```

wlcrs, 2022-07-21:
Ok, I have a suspicion that the issue is not related to the setup itself, but with those commercial inverters using other Modbus addresses, making them incompatible with this integration.

Can you download the register definitions, as described [in this topic on the Huawei forum](https://forum.huawei.com/enterprise/en/huawei/m/ViewThread.html?tid=751541&lang=en) and upload them here? That way I can compare and check.

bdkacz, 2022-07-21:
Here [click](https://drive.google.com/file/d/191spzbpXhpKIVskeT-h5lkuoTNlnQZRZ/view?usp=sharing)

bdkacz, 2022-07-21:
Ok. I make some discover.

So as on 2.7 Remapped Modbus definitions acrive power of inverters i found  when connecting on slaveid=0 (logger)  and  read address

51001 for active power of inverter no 1
51026 for active power of inverter no 2

bdkacz, 2022-07-21:
Some readed from slave =1  [data](https://drive.google.com/file/d/1kxpYf-AreoEkUT5Nhpp12b-MMYPkb-Ga/view?usp=sharing)

Filename  is number od start of register  and contains 100-125   registers. So 3000.csv include data fro9m registers 30000-30125,   320000.csv from 32000 to 32100 etc

wlcrs, 2022-07-21:
> Here [click](https://drive.google.com/file/d/191spzbpXhpKIVskeT-h5lkuoTNlnQZRZ/view?usp=sharing)

The strange thing is that the register that I try to read is mentioned in this document, but your logs clearly indicate that an "Illegal data value" exception is being thrown when reading the register:

[image]

Can you please try to read the registers in the 43xxx range with your modbuspoll tool?

bdkacz, 2022-07-21:
43000 to 43100  on all 3 slaveid 0,1,2 return 0 on all registers
[43000.csv](https://github.com/wlcrs/huawei_solar/files/9160944/43000.csv)

bdkacz, 2022-07-21:
Time Zone I can read from logger  registry 40005 (as in documentation for logger)  and  slaveid=0. Looks like registers 43000 of inverters  are not acccessible  over logger ??

wlcrs, 2022-07-21:
Ok, thanks for the info. I'll need to think a bit over the approach for this. I could ignore the error on the time zone register, but this will have two consequences:
- reported times (startup time etc) can be wrong
- editable entities (mostly  for battery management) will not show up. This is because I use the time zone register as a way to test if I have elevated permissions
- optimizer entities will not show up as these need those elevated permissions

bdkacz, 2022-07-21:
For me, it will not matter too much, because I do not have any batteries or optimizers :) But if you want to do something universal, it is probably most sensible to do as I wrote, i.e. reading the time from two registers and depending on which success will be different futher proceeding.
You can add in the documentation that if user  have/use  a logger, user  should also provide the logger's slaveid to be able to read  the time

wlcrs, 2022-07-28:
Can you try again with release 1.1.2?

bdkacz, 2022-07-28:
I het error "Setup failed for custom integration huawei_solar: Requirements for huawei_solar not found: ['huawei-solar==2.1.2']"

tomderidder, 2022-07-28:
> After upgrade to 1.1.2 I get error "Setup failed for custom integration huawei_solar: Requirements for huawei_solar not found: ['huawei-solar==2.1.2']",also "Unable to install package huawei-solar==2.1.2"

I have the same issue with the 1.1.2 release

bdkacz, 2022-07-28:
> I have the same issue with the 1.1.2 release

Looks like typo mistake - somewhere in code should be 1.1.2 but is 2.1.2 :)

tomderidder, 2022-07-28:
> > I have the same issue with the 1.1.2 release
> 
> Looks like typo mistake - somewhere in code should be 1.1.2 but is 2.1.2 :)

I guess so, the 2.1.2 release is used in manifest.json

"requirements": [
    "pyserial==3.5",
    "pyserial-asyncio==0.6",
    "huawei-solar==2.1.2"

wlcrs, 2022-07-28:
Whoops, the release of the huawei_solar library version 2.1.2 went wrong. Fixed now!

[image]

tomderidder, 2022-07-28:
Thanks for this prompt solution :)

bdkacz, 2022-07-29:
> Can you try again with release 1.1.2?
Unfortunately still not working, below the logs :

With device id=0 (logger)

```
2022-07-29 07:02:18 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting to 131.87.12.200:502.
2022-07-29 07:02:18 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-07-29 07:02:18 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server
2022-07-29 07:02:18 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-07-29 07:02:18 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 131.87.12.200:502.
2022-07-29 07:02:19 DEBUG (MainThread) [zeroconf] IPv6 scope_id 0 associated to the receiving interface
2022-07-29 07:02:19 DEBUG (MainThread) [zeroconf] Received from '::ffff:192.168.2.72':5353 [socket 12 (('::', 5353, 0, 0))]: <DNSIncoming:{id=0, flags=0, truncated=False, n_q=1, n_ans=1, n_auth=0, n_add=0, questions=[ptr[question,QM,in,_anywhereusb._tcp.local.]], answers=[record[ptr,in,_anywhereusb._tcp.local.]=2520/2519,AW02-004853 [0004F31FC7AA] (AnywhereUSB)._anywhereusb._tcp.local.]}> (96 bytes) as [b'\x00\x00\x00\x00\x00\x01\x00\x01\x00\x00\x00\x00\x0c_anywhereusb\x04_tcp\x05local\x00\x00\x0c\x00\x01\xc0\x0c\x00\x0c\x00\x01\x00\x00\t\xd8\x00+(AW02-004853 [0004F31FC7AA] (AnywhereUSB)\xc0\x0c']
2022-07-29 07:02:19 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2022-07-29 07:02:19 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0xa7 0xfe 0x0 0x1
2022-07-29 07:02:19 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2022-07-29 07:02:19 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x1 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x0
2022-07-29 07:02:19 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x1 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x0
2022-07-29 07:02:19 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-07-29 07:02:19 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1
2022-07-29 07:02:19 DEBUG (MainThread) [pymodbus.payload] [0]
2022-07-29 07:02:19 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000
2022-07-29 07:02:19 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x2 0x0 0x0 0x0 0x6 0x0 0x3 0xb7 0x98 0x0 0x1
2022-07-29 07:02:19 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 2
2022-07-29 07:02:19 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x2 0x0 0x0 0x0 0x3 0x0 0x83 0x3
2022-07-29 07:02:19 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x2 0x0 0x0 0x0 0x3 0x0 0x83 0x3
2022-07-29 07:02:19 DEBUG (MainThread) [pymodbus.factory] Factory Response[131]
2022-07-29 07:02:19 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 2
2022-07-29 07:02:19 ERROR (MainThread) [huawei_solar.huawei_solar] Got error Got error while reading from register 47000 with length 1: Exception Response(131, 3, IllegalValue) while trying to determine battery.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 119, in _initialize
    self.battery_type = (await self.get(rn.STORAGE_UNIT_1_PRODUCT_MODEL)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 245, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 287, in get_multiple
    response = await self._read_registers(
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 378, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 133, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 360, in _do_read
    raise ReadException(
huawei_solar.exceptions.ReadException: Got error while reading from register 47000 with length 1: Exception Response(131, 3, IllegalValue)
2022-07-29 07:02:19 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 150, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 130, in _initialize
    raise rerr
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 119, in _initialize
    self.battery_type = (await self.get(rn.STORAGE_UNIT_1_PRODUCT_MODEL)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 245, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 287, in get_multiple
    response = await self._read_registers(
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 378, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 133, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 360, in _do_read
    raise ReadException(
[... cut ...]
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 360, in _do_read
    raise ReadException(
huawei_solar.exceptions.ReadException: Got error while reading from register 43006 with length 1: Exception Response(131, 3, IllegalValue)
2022-07-29 07:02:26 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None
2022-07-29 07:02:26 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
```

bdkacz, 2022-08-10:
> Can you try again with release 1.1.2?

Hi.  What IDE You use ? To easy debug, I will try correct something myself but I neded tip what tools use :)

wlcrs, 2022-08-10:
Hi,

I use VS Code, but feel free to use anything else :-)

I would suggest cloning the following repo:
https://gitlab.com/Emilv2/huawei-solar.git on the develop-branch.

Run `pip install -e .` in the directory with setup.py to install the
development version in your Python installation.

Have a look at the bridge_tst.py file on how to use the library. (this
revision might be more interesting for you:
https://gitlab.com/Emilv2/huawei-solar/-/blob/0b4c70a5e28984dd23c25c33e17cb7e339780416/bridge_tst.py
)

bdkacz, 2022-08-11:
When I try debug in vs code I get error "No module named 'huawei_solar.register_names'; 'huawei_solar' is not a package"

wlcrs, 2022-08-11:
That is why you need to run `pip3 install -e .` in the directory with
setup.py

bdkacz, 2022-08-11:
> That is why you need to run `pip3 install -e .` in the directory with setup.py
> […](#)

Ok from beggining

When I try clone from vs code I can't/don't see option  select develop branch - it clone me  master branch.

So via command line I clone develop branch using git clone -b option

Then go to folder with setup.py and run pip install-e, but it want full path so I enter path to folder where the setup.py file is and it Successfully installed huawei-solar-2.1.3.dev0+ga4a5341.d20220811 but still the same error "No module named 'huawei_solar.register_names'; 'huawei_solar' is not a package"

bdkacz, 2022-08-13:
> That is why you need to run `pip3 install -e .` in the directory with setup.py

Ok after second try it works . 

 So my suggestion is to add the "I'm using smartlogger" checkbox to the adding integration window. I attach the changed code, where I use this information - an additional parameter in the create method and then checking it in the time zone, I also added a register to the registers, where the logger keeps the time zone.

I imagine it so that the user adding the integration selects the checkbox and as the slave id he enters the ID of the inverters, e.g. 1,2. And the integration, seeing the selected checkbox, downloads the time zone from the logger, while the remaining data is read directly from the inverters using their slave_id
[huawei-solar.zip](https://github.com/wlcrs/huawei_solar/files/9331240/huawei-solar.zip)

wlcrs, 2022-08-13:
Can you try to improve your contribution to automatically detect the
smartlogger instead? I don't want to add checkboxes that cause confusion
unless absolutely necessary.

bdkacz, 2022-08-13:
> Can you try to improve your contribution to automatically detect the smartlogger instead? I don't want to add checkboxes that cause confusion unless absolutely necessary.
> […](#)

Ok I will try :)

bdkacz, 2022-08-13:
And we have a new code :)
I added a check with slave_id = 0 if the connected device is a smartlogger and I set the self.detected_smart_logger = 1 flag when it detects a smartlogger. But I do not know how to add security, that when the user adds the integration, he will give slave_id 0,1,2, then if slave_id = 0 detects a smartlogger, devices under slave_id 1,2 .... should be treated as connected via a smartlogger, not directly via modbu
[huawei-solar.zip](https://github.com/wlcrs/huawei_solar/files/9331486/huawei-solar.zip)
s

wlcrs, 2022-08-16:
I cleaned up your code and committed it to a new branch: https://gitlab.com/Emilv2/huawei-solar/-/tree/smartlogger-support

Can you please verify that you can successfully run the changes to `bridge_tst.py` that I made? Only if that works I can start looking into updating the integration itself.

bdkacz, 2022-08-16:
Hi

First my mistake - the register should be 51000 not 51001

With register 51000 it detect smartlogger correctly,but  it calls  await HuaweiSolarBridge.__populate_fields(bridge) and fails with reading register 3000, which not exist in solarlog modbus  -  so the communication with smartlogger should end on detecting timezone

bdkacz, 2022-08-16:
[smartlogger deifnisitons ](https://photomate.zendesk.com/hc/en-gb/article_attachments/4693295482781/SmartLogger_ModBus_Interface_Definitions__3_.pdf) unfortunatelly does not contain any register that hold model name,serial number etc of solar logger

wlcrs, 2022-08-16:
Thanks for the feedback, I'll work on this a bit further when I have more time.

wlcrs, 2022-10-29:
I have decided to not continue working on this, as it is too hard for me to develop this feature without being able to test it against a real SmartLogger.

I welcome all pull requests for this feature here and in the underlying library by people that do have a SmartLogger. I won't do any more manual work like incorporating changes contributed via ZIP-files, as has been done earlier in this issue.

jbhobson, 2022-11-13:
Wonder how many people are running Data Logger + multiple inverters, and would have benefit from this support being added?
If there's enough, raises the question of what the cost of a logger in Belgium for wlcrs (EU$300?) and gofundme?  ;-)

closed 2023-01-19 by stale[bot]

fsdd87, 2023-05-17:
Hi, just read through your comments and sadly realized support was stopped at some point. Any chance to revive it? I do have a SmartLogger 3000 running with 4 Inverters plus battery in our home.

Roving-Ronin, 2023-07-16:
@bdkacz Just wondering if you ever kept on with trying to get this working with the Smart Logger 3000A ? Revisting this issue and installing the lastest version of the integration I noticed that this now works, with the caveat that it detects the inverter(S) and the optimisers connected to the inverters. This is with option Network connection and Slave ID's of the primary and secondary inverters (i.e. 1, 2), if you try adding the logger ( 0,1,2) then it fails.

The one thing it hasn't been able to detect is the smart meter, which I assume its trying to either expecting to find attached to the primary inverter (when its attached to the logger on RS485B1 / B2 that appears as COM3 Address 11), or else its by default scanning for addresses on COM1 only by default, hence the smart meter being on COM3 sees it not detected.

SmartLogger 3000A - S/W = V300R023C00SPC132  (waiting to get access to V300R023C00SPC152 released end 6/2023)
M1 6kW Inverters - S/W  V100R001C00SPC155 (worked also with SPC152 previously)

cc: @fsdd87

Roving-Ronin, 2023-08-05:
If of use to anyone... Here's the latest firmware, modbus definitions and release documents:

### Firmware:
**SmartMBUS**
  - V100R001C00SPC330 - Available to install via Huawei SolarFusion portal.
  - Release Documents SPC330 - https://1drv.ms/u/s!Asd_ZbD6uBqvgr45W2Z-HRly25bDtw?e=8tQVfc

**Inverters**
  - Firmware V100R001C00SPC155 (i.e. SPC155) - https://1drv.ms/u/s!Asd_ZbD6uBqvgr4yp4J392UPozTY8g?e=pehHFT
  - Release Documents SPC155 - https://1drv.ms/u/s!Asd_ZbD6uBqvgr4zfRgi1RsK8ZNlew?e=nIbeTy
 
**Smart Logger**
  -  Firmware version SPC152  - Available to install via Huawei SolarFusion portal.

### Modbus Definitions
  -  **Inverter** Modbus Interface Definitions - v05 - Dated 2023-02-16. Released with firmware SPC155
https://1drv.ms/b/s!Asd_ZbD6uBqvgr4tAfnx6uzQgB2tiQ?e=nZ1MUn

  - **SmartLogger** ModBus Interface Definitions - v41 - Dated 2023-06-02. Released with firmware  SPC152
Note: SmartLogger firmware SPC152 has been updated to be able to be deployed via the FusionSolar portal, to your logger.
https://1drv.ms/b/s!Asd_ZbD6uBqvgr4uLMdyf62_CGrWbw?e=rH3CEL

Roving-Ronin, 2023-08-05:
Using the current (v1.2.6) release of the integration with the above firmware works with the SmartLogger (3000A) detecting and configuring the Inverter (2 x M1 cascaded via RS485 to each other and then the SmartLogger) and Optimizers (19 x 600W per inverter).

Given the inverters are able to be detected and communicated with, I would assume an attached LUNA battery would also work with the integration, however I am unable to test this as I don't have an battery.

Currently the only issue would appear to be that when connecting via a SmartLogger, the Meter (DDSU-666H 1phase or DTSU-666H 3phase) is unable to be detected. This would be as the meter is connected via RS485 to the SmartLogger (default is using COM3 on the SmartLogger with address 11 [communication mode addressing, not logical addressing] being assigned to it).

In this situation the modbus registers that are available to be queried are different (i.e. SmartLogger has 6 new registers for 'Total / Negative / Positive...  Active and Reactive Electricity') but loses some, such as not having 'Meter Status' (address: 37100).

To allow comparision, here's the modbus registers / definitions for the:

**Power meter when queried / attached to an inverter (SPC155).**
[image]

___________________________________________________________

**Power meter when queried / attached to a Smart Logger (SPC152).**

[image]

@andresserveriekspert

scrichab, 2023-08-13:
hi,
i also "play" with smart logger - have setup with two inverter and battery
what happens when power meter is still connected to the first Inverter as in dongle configuration instead of connected to rs485 on SL?

Roving-Ronin, 2023-08-14:
If you have power/smart meter connected to the primary inverter, and the inverters are then cascaded (SmartLogger COM1 --> Inverter 1 --> Inverter 2) then the SmartLogger is unable to 'see' the power meter.
Due to this the Smart Logger will then be unable to monitor/record the:
- Power pulled from the grid
- Power exported to the grid
- Overall consumption of the site
- and also be unable to control PV production / export (i.e. if you required to limit export to 5kW on a 10kW site, the Smart Logger reads the meter and then tells the inverters to throttle production to 5kW + <whatever your using in the house>).

Would anyone in this discussion have any programming experience and be able to assist with developing the updates needed to add Smart Logger support? All the registers are detailed above.

scrichab, 2023-08-14:
thank you for the quick answer 
i get impatient and tried it yesterday - with the same result :-(
as a noob it thought that "only" the register number must be adjusted in a file ie active power from 37113 to 32278 
but as i see IDE is needed to access the sources - and python skills

scrichab, 2023-08-29:
@Roving-Ronin  if there is only one Inverter connected to the smart logger  (power sensor connected to that inverter) will then the registers of the power sensor recognized by the smart logger?

Roving-Ronin, 2023-08-29:
If you have a Smart Logger based deployment, be it with 1 or (up to) 80 inverters attached to it, the Smart/Consumption Meter (ie DTSU-666H) will ALWAYS be connected to the Smart Logger (COM3 port). Given this the Smart Logger has visibility of the meter (not the inverter) and the Smart Logger then uses this to control functions such as throttling PV production to limit grid exports (such as if you have a 10kW system but only allowed to export 5kW maximum to the grid).

Additionally the Modbus registers for the meter are different that when connected to the inverter (such as when system is using WiFi/FE Dongle method), hence WLCRS will not have access/visibility of the power meter (unless this functionality is added by the addition of these different registers).

scrichab, 2023-08-29:
i try to find a way to help for a solution
i have 2 setups
1) 2 x SUN2000 10KTL M1 cascaded,  Luna2000 10kW, DTSU-666H - connected with Dongle - beneath a SL3000A were i can easy switch to - tried this already but then i "lose" Home Assistant connectivity wich i need for Wallbox 
2) 1 x SUN2000 10KTL M1, 1 x SUN2000 12KTL, 1 x SUN2000 17KTL (old Modell), HUAWEI Smart Power Sensor DTSU666-HW/YDS60-80A - connected to SL3000A if necessary i can setup a LUN2000 to
for both i can setup remote access

Roving-Ronin, 2023-08-30:
As above, unless WLCRS decides to add support for the Smart Loggers (or someone with programming skills continues the Fork to add this functionality) then you will NOT be able to use the Smart Logger. As it currently stands the integration can see the inverters (would assume the batteries also..but I don't have a LUNA ESS so can't confirm) but can not see the DTSU-666H, as it must be wired into the SL instead of the primary inverter.

If you want full functionaly available with HA integration, the ONLY way for you to get this is to REMOVE the SmartLogger and swap over to using the WiFi/FE Dongle for connectivity instead.

For your two setups...

#1 - Remove SmartLogger, rewire DTSU-666H to PIN 7 & 9 of the Primary inverter and remove connect from PIN 2 & 4 (that connect it to SmartLogger). You should then be able to connect dongle to home WiFi AP (or ethernet) and add into Home Assistant.

#2 - You don't mention what model the SUN2000 12KTL is... I assume this is also an M1 ? You can also re-wire and connect these up using a WiFi/FE Dongle, replacing the SmartLogger, and it will then be able to be added into HA.  

The 17KTL that you mention is an 'old model' .. is this an 'SUN2000 17KTL M0' ?

According to https://support.huawei.com/enterprise/en/doc/EDOC1100167258/6f202a37/grid-tied-ess-networking?fbclid=IwAR0rHPVZ-4Jr9X9HZtpM8RkhtRtNYkJ5Y_6OWOUweYZx_R7BYGeFj8kkV3U#EN-US_TOPIC_0279010007__section570213564716 this is not supposed to work in a Grid ESS Tied setup, as only M1 / M2 / M3 (and with limitations L1 single phase) inverters may be tied into a cascade.

Roving-Ronin, 2023-08-31:
If the '1 x SUN2000 10KTL M1, 1 x SUN2000 12KTL, 1 x SUN2000 17KTL (old Modell), HUAWEI Smart Power Sensor DTSU666-HW/YDS60-80A' setup is working and connects up to FusionSolar via the M1, that then has the old inverters attached to it via RS485... then fine. 
But, it still comes down to the choice mentioned above...

a) Do you want to have access full to the setup in HA using WCLRS ? If yes, then rewire the DTSU-666H to the primary inverter and remove the SmartLoggers and use the dongle to connect to FusionSolar instead, then assign the dongle a reserved/static IP and connect to it over the home network.

b) If you only want to have limited visibility within HA and not be able to see the Power Consumed / Import from Grid / Export / Optimisers etc than continue using SmartLogger. You will also not be able to control much using HA.  This will only change if SmartLogger support is added (which seems unlikely given developers comments and no reply to even a post offering to pay for adding this functionality). 

IMHO given release soon of EMMA hardware that will superceded SmartLogger / Dongle / Power Meters it would also be more useful for Dev to use any efforts to add hardware support to this device (that will be moving to be the default hardware for all new installs).

wlcrs, 2025-10-23:
Hi all, the integration has evolved quite a bit since I last visited this issue. I've taken the work of @TomHarm and have integrated it in v2 of this integration (and v3 of the underlying huawei-solar-lib).

You can test it in v2.0.0 alpha 6 . Please note that it's completely untested against an actual SmartLogger device, but as it's very similar to how the EMMA device is supported now, I'm hopeful that issues should be minor. 

If things go wrong, I would appreciate a clear debug log showing the errors (or you can try to fix it yourself and open a PR instead of course).