wlcrs/huawei_solar #97: [Feature Request]: Optimizer naming convention

luzik, 2022-07-26:
### Describe your feature request

Fusion Solar is using optimizer names as number in DC chain
1.1.1
1.1.2
1.1.3 and so on
Now names are defaults to SN, and it is hard to find one you want to find.
If 1.X.X is not avaiable via modbus, then it should be possible to discover this index using "Voltage to ground" - as every PV have higher voltage to the ground. This is how FusionSolar calculate its index (number in chain)

BTW look how great it looks! I have few PV facing SE and few with azimuth SW. Awesome update!
[image]

wlcrs, 2022-07-26:
> If 1.X.X is not avaiable via modbus, then it should be possible to discover this index using "Voltage to ground" - as every PV have higher voltage to the ground. This is how FusionSolar calculate its index (number in chain)

Do you have a source for that?

luzik, 2022-07-26:
No, but this is the only way it can be done as every optimizer can only speak for themself and voltage to ground is something it can measure because it is connected to grounded frame

luzik, 2022-07-26:
I believe that second chain is using 1.2.X naming, but someone should confirm this.

wlcrs, 2022-07-28:
While it would be nice to have a consistent naming with FusionSolar, it is not easily doable: the `voltage_to_ground` will be unavailable when the integration is initialized when its dark outside. 

I will therefore not pursue this any further, but I'm open to review PR's for this functionality.

closed 2022-07-28 by wlcrs