wlcrs/huawei_solar #103: [Bug]: direct serial connection to COM port, disconnected continually

turkano, 2022-07-30:
### System Health details

## System Health

version | core-2022.7.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.32-v8
arch | aarch64
timezone | Europe/Madrid
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.26.2
Stage | running
Available Repositories | 1073
Downloaded Repositories | 12

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.07.0
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 58.7 GB
disk_used | 10.2 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Home Assistant Google Drive Backup (0.108.2), Duck DNS (1.15.0), File editor (5.3.3), Terminal & SSH (9.6.0), eWeLink Smart Home (1.2.8), AdGuard Home (4.7.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 7
views | 9
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 20 de julio de 2022, 16:28
-- | --
current_recorder_run | 30 de julio de 2022, 10:21
estimated_db_size | 416.71 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Huawei Solar Setup

inverter SUN2000-6KTL-L1
firmware  V200R001C00SPC119
direct serial connection to COM port

### Describe the issue

I’ve connected [direct serial connection to the RS485A1 and RS485B1 pins of the COM port, and change the 'Device Commissioning' function of the FusionSolar app

Baud rate: 9600
Protocol type: MODBUS
Com address: 1
**RS485 Bus Frame capture: 1** This option not appear in my Fusion Solar App

The data of inverter only appear somtimes.

Last week I’d Connectes to the inverter using AP (SUN2000-<serial_no> wifi), without problems, but in this way the inverter lost the connection with solar fusion app although it continues to be connected and working with HA

Sorry about my english :(

### Reproduction steps

1. Add integration Huawei Solar, connect Serial
2. Sometimes appear data but a lot of time not show anything.
3. I try restart HA, I’ve check back fisical connection  and “Divice Commissioning” parameters. I’m using the same “USB to RS485 converter" os wiki.
...

### Relevant debug logs

```text
Logger: custom_components.huawei_solar
Source: helpers/update_coordinator.py:237 
Integration: Huawei Solar 
First occurred: 10:22:57 (1 occurrences) 
Last logged: 10:22:57

Error fetching 102130101473_data_update_coordinator data: Could not update 102130101473 values: Modbus client is not connected to the inverter.
```

wlcrs, 2022-07-30:
I need more info to make a proper diagnosis.

Can you add the following to your `configuration.yaml`:

```
logger:
  logs:
    pymodbus: debug # only include this if you're having connectivity issues
    huawei_solar: debug
    homeassistant.components.huawei_solar: debug
```

Then restart HA, wait until the "Full logs" contain some interesting messages and share those logs here please?

Which cabling did you use? How long is the cabling?

turkano, 2022-07-30:
The cabling 2 meters more or less, and I’m using the original Huawei Meter cabling. should i try another?
[image]

I’m trying know what log you need…

2022-07-30 15:18:11 WARNING (MainThread) [homeassistant.components.number] custom_components.huawei_solar.number is setting deprecated attributes on an instance of NumberEntityDescription, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom component author.
2022-07-30 15:18:11 WARNING (MainThread) [homeassistant.components.number] custom_components.huawei_solar.number is setting deprecated attributes on an instance of NumberEntityDescription, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom component author.
2022-07-30 15:18:11 WARNING (MainThread) [homeassistant.components.number] custom_components.huawei_solar.number is setting deprecated attributes on an instance of NumberEntityDescription, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom component author.
2022-07-30 15:18:11 WARNING (MainThread) [homeassistant.components.number] custom_components.huawei_solar.number is setting deprecated attributes on an instance of NumberEntityDescription, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom component author.
2022-07-30 15:18:11 WARNING (MainThread) [homeassistant.components.number] custom_components.huawei_solar.number is setting deprecated attributes on an instance of NumberEntityDescription, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom component author.
2022-07-30 15:18:11 WARNING (MainThread) [homeassistant.components.number] custom_components.huawei_solar.number is setting deprecated attributes on an instance of NumberEntityDescription, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom component author.
2022-07-30 15:18:11 WARNING (MainThread) [homeassistant.components.number] custom_components.huawei_solar.number::HuaweiSolarNumberEntity is overriding deprecated methods on an instance of NumberEntity, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom component author.
2022-07-30 15:18:11 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-07-30 15:18:14 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 1 tries
2022-07-30 15:18:19 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 2 tries
2022-07-30 15:18:21 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform huawei_solar is taking over 10 seconds.
2022-07-30 15:18:24 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 3 tries
2022-07-30 15:18:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 4 tries
2022-07-30 15:18:31 ERROR (MainThread) [custom_components.huawei_solar] Timeout fetching 102130101473_data_update_coordinator data
2022-07-30 15:18:41 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-07-30 15:18:44 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 1 tries
2022-07-30 15:18:49 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 2 tries
2022-07-30 15:18:54 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 3 tries
2022-07-30 15:18:59 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 4 tries
2022-07-30 15:19:31 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-07-30 15:19:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 1 tries
2022-07-30 15:19:39 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 2 tries
2022-07-30 15:19:41 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-07-30 15:19:41 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101

wlcrs, 2022-07-30:
Please update to v1.1.2.

turkano, 2022-07-30:
Ohh shit!! I’m sorry, seems to work fine now

wlcrs, 2022-07-30:
Glad to hear :-)

closed 2022-07-30 by wlcrs