wlcrs/huawei_solar #105: [Bug]:

DD-bep, 2022-07-31:
### System Health details

## System Health

version | core-2022.7.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.55
arch | x86_64
timezone | Europe/Rome
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4520
Installed Version | 1.26.2
Stage | running
Available Repositories | 1134
Downloaded Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.07.0
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 30.8 GB
disk_used | 14.8 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Grafana (7.6.0), InfluxDB (4.5.0), Terminal & SSH (9.6.0), Mosquitto broker (6.1.2), Zigbee2MQTT (1.26.0-1), Samba share (10.0.0), AppDaemon (0.9.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 1
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 29 luglio 2022 09:49
-- | --
current_recorder_run | 31 luglio 2022 19:22
estimated_db_size | 689.69 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Huawei Solar Setup

Inverter SUN2000-6KTL-M1 (tree phase) with SDongle and Power Meter, no battery. Latest firmware version.

### Describe the issue

Connecting the inverter to the home assistant via a rs485 usb converter and using modbus rtu, when I go to add the inverter I get the error "Reading from the inverter failed."
I have already checked the connections and they seem correct to me.

### Reproduction steps

1. Go to "Add Integration" and select "Huawei solar"
2. Select connection type: Serial
3. Select device (Usb - RS485 converter), Slave ID = 1
4. Click on "Send".

### Relevant debug logs

```text
File "/config/.local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 323, in backoff_giveup
raise ReadException(
huawei_solar.exceptions.ReadException: Failed to read register 43006 after 5 tries
```

wlcrs, 2022-07-31:
What version are you using? There is a bug in 1.1.0, but 1.1.1 and up should work fine.

DD-bep, 2022-07-31:
Hi, I have tried both version 1.1.1 and 1.1.2

wlcrs, 2022-07-31:
I need more information to diagnose the issue.

What type of cable are you using to connect the inverter and the usb-adapter?

Please add the following to your `configuration.yaml`, restart HA, try again and then get all relevant lines from the "Full logs":

```yaml
logger:
  logs:
    pymodbus: debug # only include this if you're having connectivity issues
    huawei_solar: debug
    homeassistant.components.huawei_solar: debug
```

DD-bep, 2022-07-31:
I am using a pair of ethernet cable, 2 meters long. I only connected the RS485 A and B signals, not the ground.

2022-07-31 20:55:55 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x1 0x3 0xa7 0xfe 0x0 0x1 0xc6 0x8e
2022-07-31 20:55:55 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2022-07-31 20:55:55 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [140288220676256] Sending {"id":23,"type":"result","success":true,"result":[{"name":"huawei_solar.huawei_solar","message":["Aborting client creation due to error."],"level":"ERROR","source":["/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py",183],"timestamp":1659293596.0631225,"exception":"Traceback (most recent call last):\n File \"/usr/local/lib/python3.10/asyncio/tasks.py\", line 456, in wait_for\n return fut.result()\nasyncio.exceptions.CancelledError\n\nThe above exception was the direct cause of the following exception:\n\nTraceback (most recent call last):\n File \"/usr/local/lib/python3.10/site-packages/backoff/_async.py\", line 133, in retry\n ret = await target(*args, **kwargs)\n File \"/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py\", line 347, in _do_read\n response = await self._client.protocol.read_holding_registers(\n File \"/usr/local/lib/python3.10/site-packages/pymodbus/client/asynchronous/async_io/__init__.py\", line 35, in execute\n resp = await asyncio.wait_for(req, timeout=self._timeout)\n File \"/usr/local/lib/python3.10/asyncio/tasks.py\", line 458, in wait_for\n raise exceptions.TimeoutError() from exc\nasyncio.exceptions.TimeoutError\n\nDuring handling of the above exception, another exception occurred:\n\nTraceback (most recent call last):\n File \"/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py\", line 177, in create_rtu\n await huawei_solar._initialize()\n File \"/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py\", line 115, in _initialize\n self.time_zone = (await self.get(rn.TIME_ZONE)).value\n File \"/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py\", line 245, in get\n return (await self.get_multiple([name], slave))[0]\n File \"/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py\", line 287, in get_multiple\n response = await self._read_registers(\n File \"/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py\", line 378, in _read_registers\n result = await _do_read()\n File \"/usr/local/lib/python3.10/site-packages/backoff/_async.py\", line 141, in retry\n await _call_handlers(on_giveup, *details)\n File \"/usr/local/lib/python3.10/site-packages/backoff/_async.py\", line 34, in _call_handlers\n await hdlr(details)\n File \"/usr/local/lib/python3.10/site-packages/backoff/_async.py\", line 16, in f\n return coro_or_func(*args, **kwargs)\n File \"/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py\", line 323, in backoff_giveup\n raise ReadException(\nhuawei_solar.exceptions.ReadException: Failed to read register 43006 after 5 tries\n","count":2,"first_occurred":1659293334.5951612},{"name":"backoff","message":["Giving up _do_read(...) after 5 tries 

In the log I found these lines about the huawei solar integration, I hope I have not missed anything. Thank you

wlcrs, 2022-07-31:
Please test if you are able to read something via the RS485-to-USB adapter directly, as explained on the [Wiki: Testing the serial connection](https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter#testing-the-serial-connection)

DD-bep, 2022-07-31:
Hi, I did several tests with QModbus and finally managed to get it to work.
I used a thicker and shorter cable to connect the inverter to the Usb-RS485 converter and had to remove the WLAN / FE Dongle.
When the dongle is connected to the inverter something strange happens on the rs485 bus, on QModBus I had many errors such as "Invalid CRC" and "Response not from requested slave" and also the integration on home assistant returned a generic error.
Being able to read all the data from the home assistant it is not a problem for me to give up the dongle.
Thank you so much for your help and for your amazing work

wlcrs, 2022-08-01:
Thank you for describing what you did to solve the problem. Very useful for me to know!

closed 2022-08-01 by wlcrs