wlcrs/huawei_solar #106: What entities to use?

webdados, 2022-08-04:
I have a Huawei SUN 2000 + a Power Meter connected to it and I'm having difficulties understanding what entities to use on the Energy dashboard.

This is what I see on both devices:

[image]

[image]

The energy dashboard requests entities in kWh, so for "Grid consumption" it seems that it's "Power meter Consumption" from the power meter, and for "Return to grid" it seems that it's "Power meter Exported" from the power meter also.

The problem is with "Solar production", as compared to the values I'm reading from Fusion Solar, it seems like it can only be "Active power" or "Input power" but those are not in kWh but in W, and of course, they do not show up on the options:

[image]

Any hints?

wlcrs, 2022-08-04:
HA will correctly deduce the Solar Production by selecting either "Inverter Daily yield" or "Inverter Total yield", as it does not use momentary values of the production (=W), but the summed values over longer times (=kWh). It therefore correctly suggests those two entities.

bdkacz, 2022-08-07:
I recommend total yield - in my experience it causes fewer problems than daily

closed 2022-08-16 by wlcrs