wlcrs/huawei_solar #107: [Bug]: Reactive power wrong device class

yousaf465, 2022-08-05:
### System Health details

<html>
<body>

version | core-2022.8.1
-- | --
Installation Type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
User | root
Virtual Environment | false
Python Version | 3.10.5
Operating System Family | Linux
Operating System Version | 5.15.32-v8
CPU Architecture | aarch64
Timezone | Asia/Karachi
Configuration Directory | /config

</body>
</html>

### Huawei Solar Setup

10KTL M1 with SDongle and power meter. 3Phase.

### Describe the issue

The device class for reactive power is wrong, which results in this error 
```
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:396
Integration: Sensor (documentation, issues)
First occurred: 10:30:10 PM (2 occurrences)
Last logged: 10:30:10 PM

sensor.grid_reactive_power has unit var which is unsupported for device_class power
sensor.grid_reactive_power_2 has unit kVarh which is unsupported for device_class energy
```
The solution is to change it to reactive_power: Reactive power in var
```
device_class: reactive_power
```
https://www.home-assistant.io/integrations/sensor#device-class 

### Relevant debug logs

```text
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:396
Integration: Sensor (documentation, issues)
First occurred: 10:30:10 PM (2 occurrences)
Last logged: 10:30:10 PM

sensor.grid_reactive_power has unit var which is unsupported for device_class power
sensor.grid_reactive_power_2 has unit kVarh which is unsupported for device_class energy
```

wlcrs, 2022-08-06:
Thank you for spotting and reporting this error.

closed 2022-08-06 by wlcrs

yousaf465, 2022-08-07:
> Thank you for spotting and reporting this error.

Thanks for fixing it, when will it be released?

commit 7a1b60d9 (2022-08-06): Fix #107: correct Reactive Power Device Class