wlcrs/huawei_solar #108: [Feature integration with back up box function off-grid mode]:

emocabras75, 2022-08-09:
### Describe your feature request

when the system enters off grid mode without the grid operator's power grid, power supply is supplied by battery via the backup line of the Backup Box-B0. in this condition the integration does not work and shows "config entry 'SUN2000-6KTL-L1' for huawei_solar integration not ready yet: Could not update 102170045850 values".
 when the grid voltage is switched on again by the operator., the inverter re-assign on grid mode and integration work perfect.
be possible to make integration work with inveter off grid status (island mode)

wlcrs, 2022-08-09:
You'll need to gather extensive logs showing errors with the full stack trace if you want me to do anything about this. See end of the Readme on how to do this

emocabras75, 2022-08-10:
022-08-11 01:41:08.070 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting to 192.168.178.38:6607.
2022-08-11 01:41:08.071 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-08-11 01:41:08.083 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server
2022-08-11 01:41:08.084 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-08-11 01:41:08.088 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.178.38:6607.
2022-08-11 01:41:09.091 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2022-08-11 01:41:09.092 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0xa7 0xfe 0x0 0x1
2022-08-11 01:41:09.094 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2022-08-11 01:41:09.109 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x1 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x3c
2022-08-11 01:41:09.109 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x1 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x3c
2022-08-11 01:41:09.109 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-08-11 01:41:09.110 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1
2022-08-11 01:41:09.164 DEBUG (MainThread) [pymodbus.payload] [60]
2022-08-11 01:41:09.166 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000
2022-08-11 01:41:09.167 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x2 0x0 0x0 0x0 0x6 0x0 0x3 0xb7 0x98 0x0 0x1
2022-08-11 01:41:09.170 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 2
2022-08-11 01:41:09.220 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x2 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x2
2022-08-11 01:41:09.221 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x2 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x2
2022-08-11 01:41:09.221 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-08-11 01:41:09.222 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 2
2022-08-11 01:41:09.276 DEBUG (MainThread) [pymodbus.payload] [2]
2022-08-11 01:41:09.278 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000
2022-08-11 01:41:09.280 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x3 0x0 0x0 0x0 0x6 0x0 0x3 0x75 0x30 0x0 0x19
2022-08-11 01:41:09.282 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 3
2022-08-11 01:41:09.366 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x3 0x0 0x0 0x0 0x35 0x0 0x3 0x32 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x36 0x4b 0x54 0x4c 0x2d 0x4c 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x31 0x30 0x32 0x31 0x37 0x30 0x30 0x34 0x35 0x38 0x35 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2022-08-11 01:41:09.367 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x3 0x0 0x0 0x0 0x35 0x0 0x3 0x32 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x36 0x4b 0x54 0x4c 0x2d 0x4c 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x31 0x30 0x32 0x31 0x37 0x30 0x30 0x34 0x35 0x38 0x35 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2022-08-11 01:41:09.368 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-08-11 01:41:09.368 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 3
2022-08-11 01:41:09.421 DEBUG (MainThread) [pymodbus.payload] [21333, 20018, 12336, 12333, 13899, 21580, 11596, 12544, 0, 0, 0, 0, 0, 0, 0, 12592, 12849, 14128, 12340, 13624, 13616, 0, 0, 0, 0]
2022-08-11 01:41:09.422 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30071
2022-08-11 01:41:09.423 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x4 0x0 0x0 0x0 0x6 0x0 0x3 0x75 0x77 0x0 0x1
2022-08-11 01:41:09.424 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 4
2022-08-11 01:41:09.455 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x4 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x2
2022-08-11 01:41:09.456 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x4 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x2
2022-08-11 01:41:09.456 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-08-11 01:41:09.457 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 4
2022-08-11 01:41:09.516 DEBUG (MainThread) [pymodbus.payload] [2]
2022-08-11 01:41:09.517 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37200
2022-08-11 01:41:09.518 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x5 0x0 0x0 0x0 0x6 0x0 0x3 0x91 0x50 0x0 0x1
2022-08-11 01:41:09.525 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 5
2022-08-11 01:41:09.608 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x5 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x10
2022-08-11 01:41:09.608 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x5 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x10
2022-08-11 01:41:09.609 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-08-11 01:41:09.609 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 5
2022-08-11 01:41:09.675 DEBUG (MainThread) [pymodbus.payload] [16]
2022-08-11 01:41:09.676 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100
[... cut ...]
2022-08-11 01:41:12.550 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x06\xdc']
2022-08-11 01:41:12.551 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x03\xc3']
2022-08-11 01:41:12.551 DEBUG (MainThread) [pymodbus.payload] [b'\xff\xff', b'\xfe\x9a']
2022-08-11 01:41:12.555 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None
2022-08-11 01:41:12.556 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.

emocabras75, 2022-08-10:
this happens in a continuous cycle when the inverter is off-grid

emocabras75, 2022-08-10:
[https://imgur.com/TPKI2TK](url)

emocabras75, 2022-08-11:
this i can see when it works in on grid mode
############################################

2022-08-11 02:43:25.296 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x33 0x0 0x0 0x0 0x7 0x0 0x3 0x4 0x0 0x0 0x0 0x0
2022-08-11 02:43:25.296 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-08-11 02:43:25.297 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 51
2022-08-11 02:43:25.354 DEBUG (MainThread) [pymodbus.payload] [0, 0]
2022-08-11 02:43:25.355 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x00\x00']
2022-08-11 02:43:34.196 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-08-11 02:43:34.197 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x34 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x40 0x0 0x34
2022-08-11 02:43:34.199 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 52
2022-08-11 02:43:34.357 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x34 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x0 0x0 0x9 0x1 0x0 0x0 0x0 0x0 0x9 0x4f 0x0 0x3 0x0 0x0 0x0 0x0 0x5 0xf5 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x6 0xdc 0x0 0x0 0x1 0x5a 0xff 0xff 0xff 0xff 0x3 0xe8 0x13 0x88 0x27 0x10 0x1 0x6d 0xb 0xb8 0x2 0x0 0x0 0x0 0x62 0xf4 0x65 0x8f 0xff 0xff 0xff 0xff 0x0 0x0 0x1 0x5b 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x6d 0x0 0x6 0xbc 0xb3 0x0 0x7 0x6c 0xb5 0x62 0xf4 0x6c 0xc4 0x0 0x0 0x0 0x16 0x0 0x0 0x0 0xa9
2022-08-11 02:43:34.359 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x34 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x0 0x0 0x9 0x1 0x0 0x0 0x0 0x0 0x9 0x4f 0x0 0x3 0x0 0x0 0x0 0x0 0x5 0xf5 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x6 0xdc 0x0 0x0 0x1 0x5a 0xff 0xff 0xff 0xff 0x3 0xe8 0x13 0x88 0x27 0x10 0x1 0x6d 0xb 0xb8 0x2 0x0 0x0 0x0 0x62 0xf4 0x65 0x8f 0xff 0xff 0xff 0xff 0x0 0x0 0x1 0x5b 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x6d 0x0 0x6 0xbc 0xb3 0x0 0x7 0x6c 0xb5 0x62 0xf4 0x6c 0xc4 0x0 0x0 0x0 0x16 0x0 0x0 0x0 0xa9
2022-08-11 02:43:34.359 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-08-11 02:43:34.360 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 52
2022-08-11 02:43:34.414 DEBUG (MainThread) [pymodbus.payload] [0, 0, 2305, 0, 0, 2383, 3, 0, 0, 1525, 0, 0, 0, 0, 0, 1756, 0, 346, 65535, 65535, 1000, 5000, 10000, 365, 3000, 512, 0, 25332, 25999, 65535, 65535, 0, 347, 0, 0, 0, 0, 0, 0, 0, 0, 109, 6, 48307, 7, 27829, 25332, 27844, 0, 22, 0, 169]
2022-08-11 02:43:34.415 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x00\x00']
2022-08-11 02:43:34.416 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x05\xf5']
2022-08-11 02:43:34.417 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x00\x00']
2022-08-11 02:43:34.418 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x06\xdc']
2022-08-11 02:43:34.419 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x01Z']
2022-08-11 02:43:34.419 DEBUG (MainThread) [pymodbus.payload] [b'\xff\xff', b'\xff\xff']
2022-08-11 02:43:34.422 DEBUG (MainThread) [pymodbus.payload] [b'b\xf4', b'e\x8f']
2022-08-11 02:43:34.424 DEBUG (MainThread) [pymodbus.payload] [b'\xff\xff', b'\xff\xff']
2022-08-11 02:43:34.424 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x06', b'\xbc\xb3']
2022-08-11 02:43:34.425 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x00\xa9']
2022-08-11 02:43:34.426 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-08-11 02:43:34.426 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x35 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x10 0x0 0x4
2022-08-11 02:43:34.427 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 53
2022-08-11 02:43:34.491 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x35 0x0 0x0 0x0 0xb 0x0 0x3 0x8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2022-08-11 02:43:34.492 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x35 0x0 0x0 0x0 0xb 0x0 0x3 0x8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2022-08-11 02:43:34.493 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-08-11 02:43:34.494 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 53
2022-08-11 02:43:34.554 DEBUG (MainThread) [pymodbus.payload] [0, 0, 0, 0]
2022-08-11 02:43:34.556 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37201
2022-08-11 02:43:34.557 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x36 0x0 0x0 0x0 0x6 0x0 0x3 0x91 0x51 0x0 0x1
2022-08-11 02:43:34.559 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 54
2022-08-11 02:43:34.575 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x36 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x0
2022-08-11 02:43:34.576 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x36 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x0
2022-08-11 02:43:34.576 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-08-11 02:43:34.577 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 54
2022-08-11 02:43:34.632 DEBUG (MainThread) [pymodbus.payload] [0]
2022-08-11 02:43:34.634 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-08-11 02:43:34.635 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x37 0x0 0x0 0x0 0x6 0x0 0x3 0x90 0xed 0x0 0x25
[... cut ...]
2022-08-11 02:43:50.302 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x3d 0x0 0x0 0x0 0x6 0x0 0x6 0xc3 0x4f 0x0 0x1
2022-08-11 02:43:50.302 DEBUG (MainThread) [pymodbus.factory] Factory Response[WriteSingleRegisterResponse: 6]
2022-08-11 02:43:50.303 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 61
2022-08-11 02:43:50.304 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-08-11 02:44:00.054 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.

emocabras75, 2022-08-11:
[image]
[image]
[image]
[image]
[image]
[image]

emocabras75, 2022-08-11:
i ask myself

because when on-grid i successfully capture data from ip inverter router 192.168.178.38 and when off grid i can't receive data ,.
fusion solar works both ways and the data is always sent from the router coming from ip 192.168.178.38

wlcrs, 2022-08-16:
After reviewing the logs, I can only conclude that the inverter itself actively shuts down the connection when the off-grid mode is enabled. Maybe this is a "feature" to conserve power?

I'm closing this issue as there is nothing I can do from this side.

closed 2022-08-16 by wlcrs

emocabras75, 2022-08-16:
ok sorry to disturb. inverter does not close connection, fusionsolar app server receives data, it seems that integration does not accept inverter values if sensors related to export / import are not active.

wlcrs, 2022-08-16:
That is not what the logs are telling me.

emocabras75, 2022-08-16:
I don't know how to understand a little.
just know that inverter sends data to fusionsolar through my router at ip 192.168.178.38 and same ip address is used for hassio integration.