wlcrs/huawei_solar #109: Receiving error when attempting to add integration with elevated permissions

BertisAU, 2022-08-15:
[image]

BertisAU, 2022-08-15:
[image]

wlcrs, 2022-08-15:
Please describe your system: what inverter, what firmware version? Do you have optimizers? If so: which?

BertisAU, 2022-08-15:
I have two Huawei inverters but they are not connected to each other. I have tried to add the integration to both of them and receive the same error.

SUN2000-10KTL-M2 running on version V100R001C00SPC141
SUN2000-5KTL-M1 running on version V100R001C00SPC141

Optimisers model is SUN2000-450W-P

There are (26) optimisers connected to the 10kW inverter and (13) optimisers  connected to the 5kW inverter . 

All optimisers are on version V100R002C10SPC002.

wlcrs, 2022-08-16:
What version of the integration are you using? The line numbers in your error message are of, and it's getting me confused.

When using the latest version, this error should not occur as I check the size of the package before trying to process it:

```
        # Check if we have an empty file
        if len(file_data) < struct.calcsize(OptimizerRealTimeDataFile.HEADER):
            return

        self.file_version = struct.unpack_from(
            OptimizerRealTimeDataFile.HEADER, file_data, offset
        )

```

BertisAU, 2022-08-16:
I just tried to add the integration again and it's installed correctly for the 10kW. I'm really not sure what has changed but I'm happy it's working now.
[image]

closed 2022-08-16 by BertisAU