wlcrs/huawei_solar #110: [Feature Request]:  Document how to use Force Charge/Discharge Mode

dklemm, 2022-08-15:
### Describe your feature request

Setting the "Force Charge Power" does not change the force charge mode.

Please can you describe how "Force Charge/Discharge Mode" can be using from this integration.

[image]

[image]

wlcrs, 2022-08-16:
Is this clear enough? 

https://github.com/wlcrs/huawei_solar/wiki/Force-charge-discharge-battery

dklemm, 2022-08-16:
Thats perfect, I didn't even know about the services.

I think there is some overlap with config, I wonder if it make sense to make these readonly entites/attrs.  

It would also be good to to try and report the force charge/discharge state and remaning time/soc in the entities.

FWIW my system allowed me to set the power to 5000w (2x2500w battries) but there is some other limit of 3000w that prevented it from charging.  I'll give the services a try to see if they report a fail in the same setup.  I think the config should be limited to the lowest but I'm still trying to figure out where this limit comes from.

I will do some more testing and raise and Issues/Feature Req if I find anything

closed 2022-08-16 by dklemm

sadiq81, 2023-01-15:
@dklemm how did you make that card.