wlcrs/huawei_solar #115: [Bug]: Battery Forcible charge power UX and functionality.

JohnMcLear, 2022-08-21:
### System Health details

## System Information

version | core-2022.8.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.55
arch | aarch64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4778
Installed Version | 1.26.2
Stage | running
Available Repositories | 1097
Downloaded Repositories | 10

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.08.3
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 113.9 GB
disk_used | 78.3 GB
healthy | true
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), MariaDB (2.5.1), Let's Encrypt (4.12.6), Check Home Assistant configuration (3.11.0), Terminal & SSH (9.6.0), Zigbee2MQTT (1.27.0-1), Log Viewer (0.14.0), Mosquitto broker (6.1.2), ESPHome (2022.8.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 1
views | 7
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 12 July 2022 at 12:41
-- | --
current_recorder_run | 21 August 2022 at 19:01
estimated_db_size | 7885.02 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

<details><summary>Sonoff</summary>

version | 3.1.0 (3871c0b)
-- | --
cloud_online | 43 / 183
local_online | 1 / 1

</details>

### Huawei Solar Setup

SDongle, Network @ 192

### Describe the issue

First of, having the option to modify battery charging is amazing, thank you!

1. When Battery Forcible charge power is changed the Logbook isn't updated.  Is this expected UX?  It makes it look as if the inputs aren't working.  Numeric increments also do not update the Logbook.  Selects and Checkboxes of other values do update the Logbook.  
2. When Battery Forcible charge power is changed the value doesn't appear to have any impact on actual charging of the battery.
3. Thanks again!

### Relevant debug logs

```text
Logger: pymodbus.client.asynchronous.async_io
Source: /usr/local/lib/python3.10/site-packages/pymodbus/client/asynchronous/async_io/__init__.py:307
First occurred: 19:04:16 (70 occurrences)
Last logged: 19:16:44

    Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
    Failed to connect: [Errno 111] Connect call failed ('10.0.0.106', 6607)
    Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 6607, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 6607)
    Failed to connect: [Errno 111] Connect call failed ('10.0.0.182', 6607)
    Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('10.0.0.134', 6607), [Errno 111] Connect call failed ('fdd2:b8b:5577::9bb', 6607, 0, 0)
```

wlcrs, 2022-08-21:
> config_dir /config Home Assistant Community Store
> Downloaded Repositories 10 Home Assistant Cloud
> can_reach_cloud ok Home Assistant Supervisor
> (6.1.2), ESPHome (2022.8.0) Dashboards
> mode storage Recorder
> database_version 3.38.5 Sonoff
> local_online 1 / 1 Huawei Solar Setup

I'm thinking to remove the number entities which allow you to set the
forcible charge and forcible discharge power apart from calling the
services, as they do not have any actual function. Same goes for the
"Charge from grid" toggle. I would only keep them as parameters of the
service calls.

This way, things cannot get out of sync, and the numbers are only set on
the inverter when they are actionable.

Would that make more sense to you?

JohnMcLear, 2022-08-21:
Yea, but I'm not sure it would improve things.  I like having the values visible so I can see their entity IDs 📦 

Interestingly though calling from service was exactly what I did as a debug step to ensure the values are updated and while doing the call from service did work and the number did update, it was not reflected in Logbook or the Inverter hardware.

dklemm, 2022-08-24:
I just wanted to let you know something I observered today related to this change.

I force charged my battery a few days ago, I think I used time based.  Tonight my battery went to 0%,  at this point it started to pull 1kw from the grid.   Checking the settings it wa still mode: "Max self consumption", "Charge from grid" : true, and Forcible charge power : 1000w.

Toggling "Charge from grid" off then on seam to resolve. I was just a supprised this happened,  I wonder if froce charge mode was still active and calling stop charge would have also had the same effect.

wlcrs, 2022-08-25:
@dklemm I'm not intimately familiar with what really happens when you start a force charge/discharge. I observed the behavior for a couple of minutes and documented that.

I haven't been actively using it myself, even though I started developing this integration in the first place because I wanted to actively manipulate my battery. However, preliminary results showed that the constant term of the inverter losses are just too big to bother with it, leading to round trip losses of >40% 😞

dklemm, 2022-08-25:
>40% thats really dissapointing.  Its still benifical for me in the winter with my peak tarrif being 6 times more expensive.  

At the moment I'm just using force charge to avoid using the battery off-peak , so I just trickle charge it at 100w.   I do wonder if the losses are are greater using fast charge.  My intention is to dynamily set the charge wattage and time to match my off-peak so the battery is charging at the lowest for the fill off-peak period rather than maxing it out.

JohnMcLear, 2022-08-26:
One thing we should document is that Battery Charging is a positive value and Discharging is a negative.  It's obvious for the most part but worth documenting.

I gave up with trying to manage the settings using the integration and switched to using the smart meter provided by Huawei but I'm happy to keep trying to figure things out if it's useful to @wlcrs

wlcrs, 2022-08-26:
Thank you for your feedback. 

What solution from Huawei did you switch to? I'm not sure by what you're referring to with "smart meter provided by Huawei"?

JohnMcLear, 2022-08-26:
Sorry I wasn't thorough.  I'm using the DSU6664-H: https://support.huawei.com/enterprise/en/doc/EDOC1100020896

This "Smart Power Sensor" sends the usage of my load and what I'm sending to the grid and it's supposed to ensure the battery is filled up prior to sending back to the grid.  It doesn't seem to be working properly though which is not ideal.  In general I'm finding the comms to my battery being very unreliable, I doubt it's related to this integration though.  I think it makes sense for me to make the "official method" stable then look to see if I can drop the official method and used to something more integration/software based.

wlcrs, 2022-08-26:
Ok, I assumed that every installation of a SUN2000+battery would always have such a "Smart Power Sensor" installed by default. That was why I thought you were referring to something else.

I initially had problems with my installation because:
- for one of my phases the blue/white wires were connected to my DTSU-666H the wrong way;
- for another of my phases the clamp was installed in the wrong direction

(Did the installer notice that he made an error installing the clamp and try to correct this by switching the wiring, alas doing it that on the wrong phase? I'll never know).

Also: the battery does not react instantaneously to spiky energy usage. I frequently see it lagging behind for a few seconds, for instance when I'm preparing a coffee with my bean machine.

JohnMcLear, 2022-08-26:
> Ok, I assumed that every installation of a SUN2000+battery would always have such a "Smart Power Sensor" installed by default. That was why I thought you were referring to something else.

Yea, that makes sense.  The problem is that the lead time on these is 6 months now so we might see some people installing other CT Clamps/energy usage sensors that tie into HA (IE Emporia) which could show "consumption" and more importantly, if feeding energy to the grid and battery is < 100% then you could have an automation to change the inverter to feed into the battery instead of the grid.

I only have a single phase and I triple checked the CT Clamp.  This is the output of "Smart Meter > Active power".

## Active Power
[image]

You can see I get a negative value over night when the solar is not providing any power.  You can see I get a positive value in the day.  This is expected.

You can see below that I get a large amount of energy exported to the grid despite the inverter being on "maximum-self-consumption" and the battery having an SOC of 2%.

## Export
[image]

## SOC
[image]

1. Can you confirm that "Power Meter > Active Power" shows positive for you if you are exporting to the grid?
2. Any thoughts where to check RE SOC / Lack of charging?  I'm doing a force charge now just to see if keeping a higher minimum SOC will help.

wlcrs, 2022-08-26:
Yesterday my battery was depleted around 06:40 and fully charged again around 13:45:

[image]

[image]
[image]

Not a lot of suggestions I can make: check if you didn't change the regular "Battery maximum charging power" and "Battery maximum discharging power" settings? These are set to 2500W for my 5kWh battery in my case. Otherwise: contact your installer or Huawei for support.

JohnMcLear, 2022-08-26:
I'm the installer 📦, I'll keep grinding at it, I'm sure I'll get somewhere soon!

JohnMcLear, 2022-08-26:
Oh just one final question.  How often is your Battery State of Capacity updating?!  For me it's like once every two hours or something really slow...  It looks like yours is every 5 minutes or so...  Also, are you RS485 or over TCP?

wlcrs, 2022-08-26:
1. The SoC is reported in whole numbers. So the interval between changes depends on how fast the 50Wh has been drawn from/pushed into the battery.

2. I personally use ModbusTCP via the `SUN2000-<serial no>` WiFi broadcasted by the inverter. I don't have a dongle.

JohnMcLear, 2022-08-26:
1. Ack, thanks!
2. I'm using a similar method but **I also have a dongle**.  I assume without the dongle you can't use the official Huawei Fusion Web User Interface?  I'm tempted to remove the dongle and restart the Inverter to see if the behavior changes because it does feel like many hands are interfering and I want HA to be priority..

JohnMcLear, 2022-08-27:
Disconnecting the dongle appears to be working...  I'm going to monitor for 3-5 days and see if the results are correct then I can look to see if the changes here are taking effect.

JohnMcLear, 2022-08-31:
Further update, disconnecting the dongle is working and the battery is function very well..

[image]

I'm happy and confident it's working so my next step will be re-introducing the dongle to see if this issue is related to some sort of Fusion management conflict.

JohnMcLear, 2022-09-05:
100% confirmed fitting the dongle causes an issue.  This issue is resolved by removing the dongle and rebooting the inverter.

[image]

At