wlcrs/huawei_solar #117: [Bug]: Unexpected error while adding SUN2000

dMopp, 2022-08-23:
### System Health details

## System Information

version | core-2022.8.6
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.10.0-16-amd64
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4999
Installed Version | 1.26.2
Stage | running
Available Repositories | 1163
Downloaded Repositories | 40

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Debian GNU/Linux 11 (bullseye)
-- | --
update_channel | stable
supervisor_version | supervisor-2022.08.3
agent_version | 1.2.2
docker_version | 20.10.17
disk_total | 96.8 GB
disk_used | 22.4 GB
healthy | true
supported | true
supervisor_api | ok
version_api | ok
installed_addons | Grafana (7.6.0), InfluxDB (4.5.0), Mosquitto broker (6.1.2), AppDaemon (0.9.3), MariaDB (2.5.1), phpMyAdmin (0.8.1), Glances (0.16.0), Let's Encrypt (4.12.6), Eufy Security Add-on (0.9.4), RTSP Simple Server Add-on (v0.17.6)

</details>

<details><summary>Dashboards</summary>

dashboards | 9
-- | --
resources | 19
views | 12
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | August 13, 2022 at 07:02
-- | --
current_recorder_run | August 23, 2022 at 16:10
database_engine | mysql
database_version | 10.6.8

</details>

### Huawei Solar Setup

 SUN2000-5KTL-M1
SDongleA
3-Phase Power Meter

### Describe the issue

When i try to setup the Inverter using the serial mode, i got an unexpected Error.

I hope i added all required informations, sorry for my last report !

### Relevant debug logs

```text
2022-08-23 17:49:49.470 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-08-23 17:49:49.483 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server
2022-08-23 17:49:49.483 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-08-23 17:49:49.483 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
2022-08-23 17:49:49.483 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2022-08-23 17:49:49.483 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x1 0x3 0xa7 0xfe 0x0 0x1 0xc6 0x8e
2022-08-23 17:49:49.483 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2022-08-23 17:49:49.492 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x3f 0xac 0x2 0xff 0xff 0xf2 0xfe 0xcf 0xf7 0xff
2022-08-23 17:49:49.492 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Frame check failed, ignoring!!
2022-08-23 17:49:49.492 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Resetting frame - Current Frame in buffer - 0x3f 0xac 0x2 0xff 0xff 0xf2 0xfe 0xcf 0xf7 0xff
2022-08-23 17:49:49.514 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0xfe 0x1d 0x0 0x0 0x8c 0xff 0xff 0xff 0xde 0x0 0x1 0x0 0x0 0x3 0xb 0x23 0x23
2022-08-23 17:49:49.514 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Frame check failed, ignoring!!
2022-08-23 17:49:49.514 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Resetting frame - Current Frame in buffer - 0xfe 0x1d 0x0 0x0 0x8c 0xff 0xff 0xff 0xde 0x0 0x1 0x0 0x0 0x3 0xb 0x23 0x23
2022-08-23 17:49:49.610 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x1 0x41 0x33 0x11 0x0 0x5 0xb7 0xef 0x1 0xb7 0xfc 0x1 0xb7 0xfe 0x1 0xb8 0x8a 0x2 0xba 0xa5 0x1 0xf1 0xd0
2022-08-23 17:49:49.610 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Getting Frame - 0x41 0x33 0x11 0x0 0x5 0xb7 0xef 0x1 0xb7 0xfc 0x1 0xb7 0xfe 0x1 0xb8 0x8a 0x2 0xba 0xa5 0x1
2022-08-23 17:49:49.610 DEBUG (MainThread) [pymodbus.factory] Factory Response[PrivateHuaweiModbusResponse: 65]
2022-08-23 17:49:49.610 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Frame advanced, resetting header!!
2022-08-23 17:49:49.610 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1
2022-08-23 17:49:49.657 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x1 0x41 0x33 0x1d 0x0 0x5 0xb7 0xef 0x1 0x0 0x0 0xb7 0xfc 0x1 0x0 0x0 0xb7 0xfe 0x1 0x0 0x0 0xb8 0x8a 0x2 0x0 0x0 0x15 0x7c 0xba 0xa5 0x1 0x0
2022-08-23 17:49:49.657 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Frame - [b'\x01A3\x1d\x00\x05\xb7\xef\x01\x00\x00\xb7\xfc\x01\x00\x00\xb7\xfe\x01\x00\x00\xb8\x8a\x02\x00\x00\x15|\xba\xa5\x01\x00'] not ready
2022-08-23 17:49:49.661 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 180, in create_rtu
await huawei_solar._initialize()
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 118, in _initialize
self.time_zone = (await self.get(rn.TIME_ZONE)).value
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 248, in get
return (await self.get_multiple([name], slave))[0]
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 295, in get_multiple
response.registers, byteorder=Endian.Big, wordorder=Endian.Big
AttributeError: 'PrivateHuaweiModbusResponse' object has no attribute 'registers'
2022-08-23 17:49:49.662 ERROR (MainThread) [custom_components.huawei_solar.config_flow] 'PrivateHuaweiModbusResponse' object has no attribute 'registers'
Traceback (most recent call last):
File "/config/custom_components/huawei_solar/config_flow.py", line 225, in async_step_setup_serial
info = await validate_serial_setup(user_input)
File "/config/custom_components/huawei_solar/config_flow.py", line 53, in validate_serial_setup
bridge = await HuaweiSolarBridge.create_rtu(
File "/usr/local/lib/python3.10/site-packages/huawei_solar/bridge.py", line 91, in create_rtu
client = await AsyncHuaweiSolar.create_rtu(port, slave_id, loop=loop)
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 190, in create_rtu
raise err
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 180, in create_rtu
await huawei_solar._initialize()
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 118, in _initialize
self.time_zone = (await self.get(rn.TIME_ZONE)).value
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 248, in get
return (await self.get_multiple([name], slave))[0]
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 295, in get_multiple
response.registers, byteorder=Endian.Big, wordorder=Endian.Big
AttributeError: 'PrivateHuaweiModbusResponse' object has no attribute 'registers'
2022-08-23 17:49:49.664 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None
2022-08-23 17:49:49.664 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
```

wlcrs, 2022-08-23:
I checked the raw data being received, and it's almost gibberish compared to what it should be: a lot of errors and unexpected packages.

Please test your serial connection as described here: https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter#testing-the-serial-connection . 

Apart from changing the wiring, can you also test what happens if you remove de SDongleA? It's possible that this dongle interferes with the serial connection, but I have been unable to test it myself because my installation came without one.

dMopp, 2022-08-23:
I have round about 5M of Cat6 Cable. I will try another Twisted Pair first. Removing the Dongle would end in loosing cloud connection and is not an option for me  :( If this does not help, i will have to find an MacOS compatible tools to debug serial connection

BTW: I have a different connector, see image:
[image]

dMopp, 2022-08-24:
Ok, other Twisted Pair, same issue., see:

`Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:23.905 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.#033[0m
Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:23.921 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server#033[0m
Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[32m2022-08-24 08:39:23.921 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.#033[0m
Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[32m2022-08-24 08:39:23.922 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0#033[0m
Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:23.923 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006#033[0m
Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:23.923 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x1 0x3 0xa7 0xfe 0x0 0x1 0xc6 0x8e#033[0m
Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:23.923 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1#033[0m
Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:23.955 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x1 0x3 0x2 0x0 0x3c 0xb8 0x55#033[0m
Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:23.955 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Getting Frame - 0x3 0x2 0x0 0x3c#033[0m
Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:23.955 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]#033[0m
Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:23.955 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Frame advanced, resetting header!!#033[0m
Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:23.956 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1#033[0m
Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:23.995 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x1 0x41 0x33 0x8 0x0 0x2 0x3f 0xac 0x2 0x3f 0xb0 0x9 0x6d 0x69#033[0m
Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:23.995 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Getting Frame - 0x41 0x33 0x8 0x0 0x2 0x3f 0xac 0x2 0x3f 0xb0 0x9#033[0m
Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:23.995 DEBUG (MainThread) [pymodbus.factory] Factory Response[PrivateHuaweiModbusResponse: 65]#033[0m
Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:23.995 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Frame advanced, resetting header!!#033[0m
Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:23.995 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1#033[0m
Aug 24 08:39:23 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:23.995 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Unrequested message: PrivateHuaweiModbusResponse(51)#033[0m
Aug 24 08:39:24 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:24.006 DEBUG (MainThread) [pymodbus.payload] [60]#033[0m
Aug 24 08:39:24 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:24.006 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000#033[0m
Aug 24 08:39:24 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:24.006 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x1 0x3 0xb7 0x98 0x0 0x1 0x22 0x51#033[0m
Aug 24 08:39:24 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:24.007 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1#033[0m
Aug 24 08:39:24 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:24.043 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x1 0x41 0x73 0x7e 0x6c 0xff 0xff 0xbb 0xd7 0x0 0x0 0x29 0x3f 0xb0 0x9 0xff 0xff 0xff 0xd1 0x0 0x0 0x0 0x5d 0xff 0xff 0xff 0xfb 0x0 0x1 0x0 0x0 0x2#033[0m
Aug 24 08:39:24 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:24.043 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Frame - [b'\x01As~l\xff\xff\xbb\xd7\x00\x00)?\xb0\t\xff\xff\xff\xd1\x00\x00\x00]\xff\xff\xff\xfb\x00\x01\x00\x00\x02'] not ready#033[0m
Aug 24 08:39:24 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:24.049 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x25 0x6a 0x6a#033[0m
Aug 24 08:39:24 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:24.049 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Frame - [b'%jj'] not ready#033[0m
Aug 24 08:39:24 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:24.117 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x41 0x34 0x52 0x0 0x4 0x2c 0x24 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4a 0x38 0xc 0x0 0x4 0x42 0x54 0x32 0x32 0x35 0x30 0x30 0x34 0x38 0x37 0x38 0x38#033[0m
Aug 24 08:39:24 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:24.118 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Frame - [b'\x00A4R\x00\x04,$\x03\x00\x00\x00\x00\x00\x00J8\x0c\x00\x04BT2250048788'] not ready#033[0m
Aug 24 08:39:24 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:24.154 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xa 0x4a 0x67 0x12 0x56 0x31 0x30 0x30 0x52 0x30 0x30 0x31 0x43 0x30 0x30 0x53 0x50 0x43 0x31 0x32 0x35 0x0 0x0#033[0m
[... cut ...]
Aug 24 08:39:24 ulli be7606b67c2f[691]:     response.registers, byteorder=Endian.Big, wordorder=Endian.Big
Aug 24 08:39:24 ulli be7606b67c2f[691]: AttributeError: 'PrivateHuaweiModbusResponse' object has no attribute 'registers'#033[0m
Aug 24 08:39:24 ulli be7606b67c2f[691]: #033[36m2022-08-24 08:39:24.527 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None#033[0m
Aug 24 08:39:24 ulli be7606b67c2f[691]: #033[32m2022-08-24 08:39:24.527 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.#033[0m`

dMopp, 2022-08-24:
I was able to setup a testlab.

When iam trying to read register 30000, length 15, i got this raw data:
[test1.txt](https://github.com/wlcrs/huawei_solar/files/9413923/test1.txt)

wlcrs, 2022-08-24:
That `test1.txt` file also shows an error instead of a proper response.

> Removing the Dongle would end in loosing cloud connection and is not an option for me :(

Can you please remove it just for a few minutes to verify if it is the source of the problems? If so, I'll update the installation instructions with a warning that a serial connection cannot be combined with an active SDongleA.

dMopp, 2022-08-24:
I removed it, and it was going trough. Thats weird. Is that a configuration thing? Should i disable TCP Modbus ? (TCP/Network Modbus is working quite ok)

wlcrs, 2022-08-25:
I don't understand your questions.

I did update the Wiki with a warning that the serial connection cannot be used in combination with an SDongle. I suggest that you use the manuals I linked in https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter#network-connection to verify that Modbus TCP is enabled on your dongle.

closed 2022-08-25 by wlcrs