wlcrs/huawei_solar #118: Migrate to API tokens for uploading to PyPI

Emilv2, 2022-08-23:
### System Health details

Hi, I don't have your e-mail, so I'll just make it an issue. I don't know if  you do get them, but I keep getting these e-mails from PyPi:

>What?

>During your recent upload or upload attempt of huawei-solar to PyPI, we noticed you used basic authentication (username & password). However, your account has two-factor authentication (2FA) enabled.

>In the near future, PyPI will begin prohibiting uploads using basic authentication for accounts with two-factor authentication enabled. Instead, we will require API tokens to be used.
What should I do?

>First, generate an API token for your account or project at https://pypi.org/manage/account/token/. Then, use this token when publishing instead of your username and password. See https://pypi.org/help/#apitoken for help using API tokens to publish. 

### Huawei Solar Setup

/

### Describe the issue

/

### Relevant debug logs

```text
/
```

wlcrs, 2022-08-24:
Hi @Emilv2 ,

To solve this I need become an Owner on both the huawei-solar library on pypi.org (to generate the new tokens + bonus: it will allow me to clean up all the rc and beta-releases I did in the past), and your repository over at https://gitlab.com/Emilv2/huawei-solar/ to set the new environment variables (gives me permission to edit environment variables in the CI/CD environment + bonus: permission to push things to your master branch, which I currently cannot )

Relevant urls to make those changes: 

- https://pypi.org/manage/project/huawei-solar/collaboration/
- https://gitlab.com/Emilv2/huawei-solar/-/project_members

Thanks in advance!

Emilv2, 2022-08-25:
Done

wlcrs, 2022-08-26:
Thanks. I updated the secrets for the CI pipeline in Gitlab with the API tokens I generated in PyPI.

closed 2022-08-26 by wlcrs

Emilv2, 2022-08-26:
I guess it's too late now, but please don't delete old versions, there is no need for that. It can break some people's setup.