wlcrs/huawei_solar #123: [Bug]:

johanvh83, 2022-08-28:
### System Health details

Home Assistant 2022.8.7
Supervisor 2022.08.3
Operating System 8.4
Frontend 20220802.0 - latest

### Huawei Solar Setup

Inverter: 
Single phase
Battery type 
Power meter connected

### Describe the issue

Since update from 2022.8.6 to 2022.8.7 I have below error (see logs).
I restored to V2022.8.6 and the problem is resolved.

### Relevant debug logs

```text
Logger: homeassistant.setup
Source: setup.py:184
First occurred: 11:18:51 AM (1 occurrences)
Last logged: 11:18:51 AM

Setup failed for custom integration huawei_solar: Requirements for huawei_solar not found: ['huawei-solar==2.0.0rc5'].

Logger: homeassistant.util.package
Source: util/package.py:98
First occurred: 11:18:44 AM (3 occurrences)
Last logged: 11:18:51 AM

Unable to install package huawei-solar==2.0.0rc5: ERROR: Could not find a version that satisfies the requirement huawei-solar==2.0.0rc5 (from versions: 0.0.1.linux-x86_64, 0.0.2.linux-x86_64, 0.0.3.linux-x86_64, 0.0.4.linux-x86_64, 0.0.5.linux-x86_64, 0.0.6.linux-x86_64, 0.0.7.linux-x86_64, 0.0.8.linux-x86_64, 0.0.9.linux-x86_64, 0.0.10.linux-x86_64, 0.0.11.linux-x86_64, 0.1.0b2.linux-x86_64, 0.1.0b4.linux-x86_64, 0.1.0.linux-x86_64, 0.1.1.linux-x86_64, 1.0.1.linux-x86_64, 1.1.0b1.linux-x86_64, 1.1.0.linux-x86_64, 1.2.0.linux-x86_64, 0.0.1, 0.0.2, 0.0.3, 0.0.4, 0.0.5, 0.0.6, 0.0.7, 0.0.8, 0.0.9, 0.0.10, 0.0.11, 0.1.0b2, 0.1.0b4, 0.1.0, 0.1.1, 1.0.1, 1.1.0b1, 1.1.0, 1.2.0, 2.0.0, 2.0.1, 2.0.2, 2.1.0, 2.1.1, 2.1.2, 2.1.3) ERROR: No matching distribution found for huawei-solar==2.0.0rc5
```

wlcrs, 2022-08-28:
Please upgrade your integration to the latest version. You are running an outdated and unsupported version.

johanvh83, 2022-08-28:
Thanks for your comment. Even when updateing to beta V1.1.3b it's not resolved.

johanvh83, 2022-08-28:
I don't believe the problem is related to HA versions or your integrations versions.
Any idea where this comes from: Unable to install package huawei-solar==2.0.0rc5

johanvh83, 2022-08-28:
Hi wlcrs, The problem is resolved, finally, after many hours.
1 year ago I started after the solar installation with the EmilV2 integration. Since I use your integration I just had removed the EmilV2 YAML configuration and renamed the EmilV2 folder in the custom_components folder. For some reason this occured the problem with your integration.
I removed the EmilV2 folder and restarted HA. Your integration starts now without any problem!

wlcrs, 2022-08-28:
Glad to hear you got it resolved

closed 2022-08-28 by wlcrs