wlcrs/huawei_solar #128: [Bug]: Inverter changed IP

marithpl, 2022-09-03:
### System Health details

Connection problem

### Huawei Solar Setup

SDongle 

### Describe the issue

I've today problem with router. I had to reset factory settings.
Now Dongle have different IP, how to change it without loosing data.

### Relevant debug logs

```text
n/d
```

wlcrs, 2022-09-03:
This is not a bug, and it's bad practice to allow dynamic IP assignment for something like this. You need to set a fixed IP for your dongle instead.

closed 2022-09-03 by wlcrs

marithpl, 2022-09-03:
I always use IP reservation.
I didn't found option to set static ip for inverter.

marithpl, 2022-09-03:
But I found a solution.
In HA config folder there is hidden folder .storage with core.config_entries.

Everything seems to work

kaciker, 2023-08-10:
> But I found a solution. In HA config folder there is hidden folder .storage with core.config_entries.
> 
> Everything seems to work

Thanks ¡ you saved my live..