wlcrs/huawei_solar #130: [Bug]: Battery controls not created when using RS485 method

be04376, 2022-09-07:
### System Health details

## System Information

version | core-2022.8.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.60
arch | aarch64
timezone | Europe/Brussels
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4379
Installed Version | 1.27.1
Stage | running
Available Repositories | 1106
Downloaded Repositories | 17

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.5
-- | --
update_channel | stable
supervisor_version | supervisor-2022.08.5
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 113.9 GB
disk_used | 21.9 GB
healthy | true
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | Samba share (10.0.0), File editor (5.3.3), Let's Encrypt (4.12.7), Mosquitto broker (6.1.3), Portainer (2.0.0), SSH & Web Terminal (12.0.2), Grafana (7.6.0), Home Assistant Google Drive Backup (0.108.4), Z-Wave JS (0.1.70), InfluxDB (4.5.0), MariaDB (2.5.1), Studio Code Server (5.3.0), ESPHome (2022.3.1), Zigbee2MQTT (1.27.2-1), Eufy Security Add-on (0.9.4), RTSP Simple Server Add-on (v0.17.6)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 10
views | 11
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | August 28, 2022 at 6:20 AM
-- | --
current_recorder_run | September 7, 2022 at 8:22 PM
estimated_db_size | 2921.70 MiB
database_engine | mysql
database_version | 10.6.8

</details>

<details><summary>Solcast PV Forecast</summary>

can_reach_server | ok
-- | --
remaining_requests | 26
rooftop_site_count | 1
polling_hours | 6 -> 21 and once at midnight

</details>

### Huawei Solar Setup

inverter: SUN2000-4.6KTL-L1 firmware:V200R001C00SPC124
power meter single phase
battery: LUNA2000-5KW-C0 + 2x LUNA2000-5-E0 firmware:V100R002C00SPC111
No dongle
No optimizers

### Describe the issue

When I connect to the inverter AP I have battery controls ( charge from grid, select operating mode)
I switched to the preferred RS485 method, and I can see power, battery SOC etc, but i lost the battery controls.
They are not visible in the battery device section of the integration
[home-assistant (2).log](https://github.com/wlcrs/huawei_solar/files/9509031/home-assistant.2.log)

### Relevant debug logs

```text
Uploaded in the description field
```

wlcrs, 2022-09-08:
Can you test again with beta-release 1.1.3 beta 2 ?

be04376, 2022-09-08:
beta-release 1.1.3 beta 2 resolves the problem.
I have all configuration items of the battery, and the service to forcible charge is working.
I noticed that during the setup flow the battery now popups before you enter the area of the inverter. In 1.1.3 b1 this was after you entered the area.

Thx for the quick support and super addon.

closed 2022-09-08 by be04376