wlcrs/huawei_solar #135: [Bug]: Getting "Unexpected error"

Joikast, 2022-09-13:
### System Health details

## System Information

version | core-2022.9.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.55
arch | x86_64
timezone | Europe/Stockholm
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4887
Installed Version | 1.27.1
Stage | running
Available Repositories | 1110
Downloaded Repositories | 5

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | October 9, 2022 at 2:00 AM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-west-2-1.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.08.6
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 43.6 GB
disk_used | 16.7 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Grafana (7.6.0), InfluxDB (4.5.0), Terminal & SSH (9.6.1), deCONZ (6.15.0), Mosquitto broker (6.1.3), Node-RED (13.3.2), Z-Wave JS (0.1.70)

</details>

<details><summary>Dashboards</summary>
[... 227 lines cut ...]
2022-09-13 15:23:57.059 INFO (MainThread) [homeassistant.components.websocket_api.http.connection] [139652262119216] Connection closed by client
2022-09-13 15:26:00.924 INFO (MainThread) [homeassistant.components.websocket_api.http.connection] [139652262463472] Connection closed by client
2022-09-13 15:26:49.925 INFO (MainThread) [homeassistant.components.websocket_api.http.connection] [139652263009232] Connection closed by client
2022-09-13 15:27:38.928 INFO (MainThread) [homeassistant.components.websocket_api.http.connection] [139652279820416] Connection closed by client
2022-09-13 15:28:00.527 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-09-13 15:28:00.527 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
2022-09-13 15:28:00.528 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2022-09-13 15:28:00.777 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000
2022-09-13 15:28:00.862 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000
2022-09-13 15:28:01.530 ERROR (MainThread) [custom_components.huawei_solar.config_flow] 'PrivateHuaweiModbusResponse' object has no attribute 'registers'
Traceback (most recent call last):
File "/config/custom_components/huawei_solar/config_flow.py", line 225, in async_step_setup_serial
info = await validate_serial_setup(user_input)
File "/config/custom_components/huawei_solar/config_flow.py", line 53, in validate_serial_setup
bridge = await HuaweiSolarBridge.create_rtu(
File "/usr/local/lib/python3.10/site-packages/huawei_solar/bridge.py", line 94, in create_rtu
await HuaweiSolarBridge.__populate_fields(bridge)
File "/usr/local/lib/python3.10/site-packages/huawei_solar/bridge.py", line 118, in __populate_fields
model_name_result, serial_number_result = await bridge.client.get_multiple(
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 292, in get_multiple
response.registers, byteorder=Endian.Big, wordorder=Endian.Big
AttributeError: 'PrivateHuaweiModbusResponse' object has no attribute 'registers'
2022-09-13 15:28:27.926 INFO (MainThread) [homeassistant.components.websocket_api.http.connection] [139652262807104] Connection closed by client
2022-09-13 15:28:37.957 INFO (MainThread) [homeassistant.components.websocket_api.http.connection] [139652262817952] Connection closed by client
```

wlcrs, 2022-09-13:
Does your inverter have an SDongle attached on its USB port? I've seen similar bug reports in the past because the SDongle and serial connection seem to interfere with each other.

Joikast, 2022-09-13:
Yes, i am using an SDongle.

[image]

So what would be the recommended way to use the integration?

Joikast, 2022-09-13:
Ok, just need to get the access to enable unrestricted modbus-TCP (I guess that the default value is disabled?)

Or, an alternative would perhaps be to use a wired connection for the inverter and then remove the dongle and run through serial?

Joikast, 2022-09-13:
By the way, I can confirm that serial connection works when I unplug the SDongleA.

closed 2022-09-13 by wlcrs

wlcrs, 2022-09-13:
> Ok, just need to get the access to enable unrestricted modbus-TCP (I guess that the default value is disabled?)
> 
> Or, an alternative would perhaps be to use a wired connection (RJ45 port, ethernet) for the inverter and then remove the dongle and run through serial?

Almost missed this reply. Yes, when connecting to the inverter via the SDongleA I think you need to enable unrestricted modbus tcp via the FusionSolar device commissioning app. 

I don't fully understand what you try to say in your 2nd paragraph, so I can't answer on your suggestion there.

wlcrs, 2022-09-13:
> Thanks a lot!
> 
> Do you have any clue why this is happening? could it be that the wifi signal is interfering with the serial communication?

My assumption is that the SDongleA is internally connecting to the same serial interface exposed by the inverter mcu on the 485a1 and 485b1 pins. So signals get mixed on a very low level. 
> 
> One last thing, do you happen to know if the inverter uses the SDongleA to upload its data to the FusionSolar, or is it using the built in AP?

The SDongleA seems to be only necessary in scenarios where the inverter is too far away from the home wifi to connect to the internet, and a wired connection to the internet router is needed instead. If that's not a problem, the SDongleA has no real advantage afaik. I suspect that in earlier models of the Huawei inverters this was not the case (and the dongle was thus needed for the monitoring via FusionSolar), but not all installers got that memo and they just keep installing them on the current generation of inverters. Maybe a sneaky way to bill some extra components?

Joikast, 2022-09-13:
> My assumption is that the SDongleA is internally connecting to the same serial interface exposed by the inverter mcu on the 485a1 and 485b1 pins. So signals get mixed on a very low level.
>
>
> The [...]

So you are basically saying that the inverter has a built in wifi-adapter that it uses to connect to internet/FusionSolar, and the SDongleA is only used if you want to run by cable (RJ45) instead? If that is the case I have really been mixing up things..

wlcrs, 2022-09-13:
Yes. My installation has no SDongleA and reporting to FusionSolar works just fine by directly connecting to my home WiFi.

Joikast, 2022-09-13:
> Yes. My installation has no SDongleA and reporting to FusionSolar works just fine by directly connecting to my home WiFi.

wow, this is actually really great news, then I will just unplug that dongle and run through serial communication instead.

Again, thanks a lot and thanks for a really nice integration.