wlcrs/huawei_solar #140: [Bug]: All sensors become unavailable

themaninspain, 2022-09-19:
### System Health details

## System Information

version | core-2022.9.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.67
arch | x86_64
timezone | Europe/Madrid
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4901
Installed Version | 1.27.2
Stage | running
Available Repositories | 1109
Downloaded Repositories | 14

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | September 14, 2023 at 2:00 AM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-central-1-4.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.0
-- | --
update_channel | stable
supervisor_version | supervisor-2022.08.6
agent_version | 1.3.0
docker_version | 20.10.17
disk_total | 109.3 GB
disk_used | 15.4 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Check Home Assistant configuration (3.11.0), Network UPS Tools (0.11.0), SSH & Web Terminal (12.0.2), Home Assistant Google Drive Backup (0.108.4), AdGuard Home (4.7.4), SQLite Web (3.6.1), Z-Wave JS (0.1.71)

</details>

<details><summary>Dashboards</summary>
[... 336 lines cut ...]
2022-09-19 13:40:02.005 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x5 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x40 0x0 0x34
2022-09-19 13:40:02.007 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 5
2022-09-19 13:40:07.010 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x6 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x40 0x0 0x34
2022-09-19 13:40:07.012 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 6
2022-09-19 13:40:12.023 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x7 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x40 0x0 0x34
2022-09-19 13:40:12.024 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 7
2022-09-19 13:40:32.131 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x5 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x3 0x76 0x8 0xf3 0x0 0x0 0x0 0x0 0x7 0xec 0x0 0x3 0x0 0x0 0x0 0x0 0xe 0xdf 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x17 0x70 0x0 0x0 0x3 0x63 0x0 0x0 0x0 0x0 0x3 0xe8 0x13 0x89 0x26 0x35 0x1 0x75 0xb 0xb8 0x2 0x0 0x0 0x0 0x63 0x28 0x21 0x26 0x63 0x27 0x82 0x36 0x0 0x0 0x3 0x67 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xd 0x18 0x23 0x0 0xd 0x6e 0xc7 0x63 0x28 0x72 0x30 0x0 0x0 0x0 0x42 0x0 0x0 0x3 0x61
2022-09-19 13:40:32.132 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x5 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x3 0x76 0x8 0xf3 0x0 0x0 0x0 0x0 0x7 0xec 0x0 0x3 0x0 0x0 0x0 0x0 0xe 0xdf 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x17 0x70 0x0 0x0 0x3 0x63 0x0 0x0 0x0 0x0 0x3 0xe8 0x13 0x89 0x26 0x35 0x1 0x75 0xb 0xb8 0x2 0x0 0x0 0x0 0x63 0x28 0x21 0x26 0x63 0x27 0x82 0x36 0x0 0x0 0x3 0x67 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xd 0x18 0x23 0x0 0xd 0x6e 0xc7 0x63 0x28 0x72 0x30 0x0 0x0 0x0 0x42 0x0 0x0 0x3 0x61
2022-09-19 13:40:32.132 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-09-19 13:40:32.132 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 5
2022-09-19 13:40:32.173 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x6 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x3 0x76 0x8 0xf3 0x0 0x0 0x0 0x0 0x7 0xec 0x0 0x3 0x0 0x0 0x0 0x0 0xe 0xdf 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x17 0x70 0x0 0x0 0x3 0x63 0x0 0x0 0x0 0x0 0x3 0xe8 0x13 0x89 0x26 0x35 0x1 0x75 0xb 0xb8 0x2 0x0 0x0 0x0 0x63 0x28 0x21 0x26 0x63 0x27 0x82 0x36 0x0 0x0 0x3 0x67 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xd 0x18 0x23 0x0 0xd 0x6e 0xc7 0x63 0x28 0x72 0x30 0x0 0x0 0x0 0x42 0x0 0x0 0x3 0x61 0x0 0x7 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x3 0x76 0x8 0xf3 0x0 0x0 0x0 0x0 0x7 0xec 0x0 0x3 0x0 0x0 0x0 0x0 0xe 0xdf 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x17 0x70 0x0 0x0 0x3 0x63 0x0 0x0 0x0 0x0 0x3 0xe8 0x13 0x89 0x26 0x35 0x1 0x75 0xb 0xb8 0x2 0x0 0x0 0x0 0x63 0x28 0x21 0x26 0x63 0x27 0x82 0x36 0x0 0x0 0x3 0x67 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xd 0x18 0x23 0x0 0xd 0x6e 0xc7 0x63 0x28 0x72 0x30 0x0 0x0 0x0 0x42 0x0 0x0 0x3 0x61
2022-09-19 13:40:32.174 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x6 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x3 0x76 0x8 0xf3 0x0 0x0 0x0 0x0 0x7 0xec 0x0 0x3 0x0 0x0 0x0 0x0 0xe 0xdf 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x17 0x70 0x0 0x0 0x3 0x63 0x0 0x0 0x0 0x0 0x3 0xe8 0x13 0x89 0x26 0x35 0x1 0x75 0xb 0xb8 0x2 0x0 0x0 0x0 0x63 0x28 0x21 0x26 0x63 0x27 0x82 0x36 0x0 0x0 0x3 0x67 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xd 0x18 0x23 0x0 0xd 0x6e 0xc7 0x63 0x28 0x72 0x30 0x0 0x0 0x0 0x42 0x0 0x0 0x3 0x61 0x0 0x7 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x3 0x76 0x8 0xf3 0x0 0x0 0x0 0x0 0x7 0xec 0x0 0x3 0x0 0x0 0x0 0x0 0xe 0xdf 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x17 0x70 0x0 0x0 0x3 0x63 0x0 0x0 0x0 0x0 0x3 0xe8 0x13 0x89 0x26 0x35 0x1 0x75 0xb 0xb8 0x2 0x0 0x0 0x0 0x63 0x28 0x21 0x26 0x63 0x27 0x82 0x36 0x0 0x0 0x3 0x67 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xd 0x18 0x23 0x0 0xd 0x6e 0xc7 0x63 0x28 0x72 0x30 0x0 0x0 0x0 0x42 0x0 0x0 0x3 0x61
2022-09-19 13:40:32.174 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-09-19 13:40:32.174 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 6
2022-09-19 13:40:32.174 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-09-19 13:40:32.175 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 7
2022-09-19 13:40:47.002 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x8 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x40 0x0 0x34
2022-09-19 13:40:47.003 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 8
2022-09-19 13:40:52.015 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x9 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x40 0x0 0x34
2022-09-19 13:40:52.016 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 9
2022-09-19 13:40:57.020 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0xa 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x40 0x0 0x34
2022-09-19 13:40:57.021 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 10
2022-09-19 13:41:02.025 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0xb 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x40 0x0 0x34
2022-09-19 13:41:02.026 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 11
```

wlcrs, 2022-09-20:
I don't see any problems with the integration, but I do see disconnects/timeouts. Please check your setup: is the connection to the inverter stable enough?

closed 2022-09-20 by wlcrs

themaninspain, 2022-09-20:
The inverter connection was stable and has been for the last 10 months. The problem only started this month.
I will get tech support to re-check the inverter.

tomderidder, 2022-09-20:
Same issue happend with my installation. It was caused by unstable wifi connection, and it is solved by moving my ha machine next to the inverter.

themaninspain, 2022-09-20:
> Same issue happend with my installation. It was caused by unstable wifi connection, and it is solved by moving my ha machine next to the inverter.

I haven't changed the location of my AP's (I use unifi) but I will do some tests to see if it helps.

themaninspain, 2022-09-22:
> > Same issue happend with my installation. It was caused by unstable wifi connection, and it is solved by moving my ha machine next to the inverter.
> 
> I haven't changed the location of my AP's (I use unifi) but I will do some tests to see if it helps.

That did it! I have 3 AP's and for some reason the inverter had started connecting to the one furthest away - I forced it to connect to the nearest AP and I am now getting consistent readings.