wlcrs/huawei_solar #141: [Bug]: Forced charging

kristianolofsson, 2022-09-26:
### System Health details

-

### Huawei Solar Setup

-

### Describe the issue

In the configuration card, values like forced power charge and discharge can be set. But if eg the mode is set to Max Self Consumption the forced charging can never be start despite changing the power. This requires that parameter 471000 (Forced charging mode) is set for charging/discharging or nothing will happen.

### Relevant debug logs

```text
-
```

wlcrs, 2022-09-26:
no logs = no feedback

closed 2022-09-26 by wlcrs

kristianolofsson, 2022-09-26:
What log is needed? How is the forced charging suppose to work? Please advise so I can provide suitable log.

wlcrs, 2022-09-26:
I provided instructions in the bug report template: perform the actions you think that do not work, and submit the logs of those actions.

You also need to specify all details on your installation. It is impossible to debug otherwise. If you don't do any effort on this bug report, I won't neither.

Please note that I only provide support on this integration correctly setting parameters on the inverter (based on documentation provided by Huawei). I do not necessarily know how these parameter changes affect how the inverter operates.

kristianolofsson, 2022-09-26:
Sorry for being to quick in my analysis. The parameters provided in the configuration card are working and you can provide eg. the forced charging/discharging value. Even the switch to feed from grid or not. But what i would like to adress is that the forced charging value makes no sense if not parameter 47100 is provided. Since this switched starts the forced charging. If i do change this switch manually in the inverter its working and the configuration values are used. So i guess it should either be part of that you change the values for the forced power or that it should be a separate parameter. 
Im using a Huawei inverter with a 15kwh battery.

wlcrs, 2022-09-26:
In the 1.1.3 beta release, I've already removed the separate forced
charge/discharge entities from the integration as they are indeed confusing.

You should only use the HA services instead, as documented on
https://github.com/wlcrs/huawei_solar/wiki/Force-charge-discharge-battery

kristianolofsson, 2022-09-26:
Thanks! This will make it more clear.
Randomly I do receive errors in the service call for forced start/stopped charging. Is this a response delay issue? It happens even though I que the messages.

"
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 200, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1738, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1775, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/config/custom_components/huawei_solar/services.py", line 168, in stop_forcible_charge
    await bridge.set(rn.STORAGE_FORCIBLE_DISCHARGE_POWER, 0)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/bridge.py", line 359, in set
    return await self.client.set(name, value, slave=self.slave_id)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 496, in set
    response = await self.write_registers(
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 520, in write_registers
    response = await self._client.protocol.write_registers(
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
"

kristianolofsson, 2022-09-27:
Ok, so i did some more test. I also asked a few friends of mine to do the same tests and they all got the same issue.
We are all running HA connected with Nodered over websocket. We are all using your integration and services to force the charging.
Upon manual test it seems to work fine, the inverter is triggered either with forced charging or discharging.

But the issue that occurs seems to be a time out issue. Either by the websocket, the modbus integration or a combination of both.

Is there anyway to locate this?

I really like your job and would be very happy if this can be managed using Nodered.

Kristian

https://github.com/wlcrs/huawei_solar/wiki/Force-charge-discharge-battery