wlcrs/huawei_solar #143: [Bug]: Huawei-solar always connecting to 127.0.0.1

stevengoossensB, 2022-09-30:
### System Health details

/

### Huawei Solar Setup

I'm connecting over the network to port 502. 

### Describe the issue

Whatever IP I fill out in the UI configuration, it seems the integration attempts to use 127.0.0.1 instead of the IP I fill out. When I change the port, I can see that change reflected in the logs. Obviously, I'm not adding 127.0.0.1 as an IP and I have tried to put the IP between quotes as well.

### Relevant debug logs

```text
Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py:344
First occurred: 17:30:33 (1 occurrences)
Last logged: 17:30:33

Modbus client is not connected to the inverter.
NoneType: None

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 150, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 115, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 245, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 287, in get_multiple
    response = await self._read_registers(
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 378, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 133, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 345, in _do_read
    raise ConnectionException(message)
huawei_solar.exceptions.ConnectionException: Modbus client is not connected to the inverter.

Logger: pymodbus.client.asynchronous.async_io
Source: /usr/local/lib/python3.10/site-packages/pymodbus/client/asynchronous/async_io/__init__.py:307
First occurred: 17:30:33 (6 occurrences)
Last logged: 17:30:40

Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
```

wlcrs, 2022-09-30:
Hi Steven, please post the full logs instead of just this one stacktrace. Also describe your installation in full with firmware version + do you have a dongle?

Have you confirmed with other tools that you can connect to that modbus tcp interface?

stevengoossensB, 2022-10-02:
Hi,

I have a script using the huawei-solar library to connect over TCP 502, which works and gives me all data back, so connectivity is not the issue. 

I'm running Home Assistant in the 'latest' container published on dockerhub. It seems the issue is the entry of the parameters, which is not capturing the input I pass for the IP.

There's not much more relevant information in the logs apart from the small stacktrace and I prefer not to share the full log, containing all of the other integration logs.

This happens when I fill out the config in the UI:
2022-10-02 11:37:29.072 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2022-10-02 11:37:29.073 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
2022-10-02 11:37:29.073 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 150, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 115, in _initialize
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 245, in get
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 287, in get_multiple
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 378, in _read_registers
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 345, in _do_read
huawei_solar.exceptions.ConnectionException: Modbus client is not connected to the inverter.

This happens right after:
2022-10-02 11:37:35.330 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-10-02 11:37:41.736 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-10-02 11:37:54.542 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-10-02 11:38:20.148 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-10-02 11:39:11.354 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-10-02 11:39:57.124 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-10-02 11:40:53.760 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-10-02 11:44:18.565 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-10-02 11:44:57.129 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)

==> It seems for some reason a pymodbus client with an empty IP address was created, defaulting to localhost.

I'll do some further investigation today and see whether I can (temporarily) fix it by hardcoding the IP address of my inverter.

wlcrs, 2022-10-02:
I also suspect a pymodbus problem. I've seen that error before in other bugreports, but the real problem was always something else: bad connection to the inverter, etc

dperez-sct, 2022-10-02:
I have the same problem:

```Logger: pymodbus.client.asynchronous.async_io
Source: /usr/local/lib/python3.10/site-packages/pymodbus/client/asynchronous/async_io/__init__.py:307
First occurred: 22:06:34 (10 occurrences)
Last logged: 22:08:16

Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)```

And, of course, I'm putting the IP of the dongle on the IP parameter.

stevengoossensB, 2022-10-03:
Hi,

After some further debugging, I can see the connection attempt through tcpdump, but it looks like the script (or pymodbus) isn't waiting long enough for the connection to be established. The reconnection attempts to 127.0.0.1 is then probably because the CONF_HOST is set to 'None' since the initial connection failed and the input wasn't persisted.

My inverter software is **V100R001C00SPC335** and it's a **SUN2000 4KTL**. Important detail is that I actually had the Home Assistant integration work 1 week ago, so there must be something that introduced this bug.

I tried manually upgrading pymodbus to 3.0.0.dev5, but the problem remains there as well.

stevengoossensB, 2022-10-03:
Bug can be closed. Apparently the behaviour is caused by having 2 simultaneous modbus connections to the inverter. I stopped the daemon with I was using in the past and will migrate that monitoring to the HomeAssistant integration.