wlcrs/huawei_solar #145: [Bug]: Fatal errors show up in log after HA 2022.10.1 upgrade

bengtner, 2022-10-08:
### System Health details

## System Information

version | core-2022.10.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.61-v8
arch | aarch64
timezone | Europe/Stockholm
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4946
Installed Version | 1.26.2
Stage | running
Available Repositories | 1118
Downloaded Repositories | 8

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | October 20, 2022 at 02:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | false
remote_server | eu-west-2-4.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.0
-- | --
update_channel | stable
supervisor_version | supervisor-2022.09.1
agent_version | 1.3.0
docker_version | 20.10.17
disk_total | 116.7 GB
disk_used | 5.9 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.6.1), Studio Code Server (5.4.0), Mosquitto broker (6.1.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 9
-- | --
resources | 3
views | 10
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | September 28, 2022 at 15:02
-- | --
current_recorder_run | October 7, 2022 at 21:42
estimated_db_size | 219.76 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

<details><summary>Sonoff</summary>

version | 3.0.5 (200f243)
-- | --
cloud_online | 0 / 1
local_online | 1 / 1
debug | failed to load: 

</details>

### Huawei Solar Setup

SUN2000-10KTL inverter and LUNAR2000 battery.
Connected through sDongle

### Describe the issue

After upgrading HA to 2022.10.1 yesterday, the following error shows up in the log. It seems not to have any clear impact on operation fortunately. This error showed up one moret time this morning. Have not seen this before the upgrade to 2022.10.1.

*****************************
This error originated from a custom integration.

Logger: custom_components.huawei_solar
Source: custom_components/huawei_solar/__init__.py:250
Integration: Huawei Solar
First occurred: 19:42:22 (1 occurrences)
Last logged: 19:42:22

Unexpected error fetching HV2240546633_data_update_coordinator data: unpack requires a buffer of 2 bytes
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 205, in _async_refresh
    [self.data](http://self.data/) = await self._async_update_data()
  File "/config/custom_components/huawei_solar/__init__.py", line 250, in _async_update_data
    return await self.bridge.update()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/bridge.py", line 186, in update
    result.update(await _get_multiple_to_dict(self._pv_registers))
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/bridge.py", line 179, in _get_multiple_to_dict
    zip(names, await self.client.get_multiple(names, self.slave_id))
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 303, in get_multiple
    result.append(await self._decode_response(registers[idx], decoder))
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 237, in _decode_response
    result = reg.decode(decoder, self)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/registers.py", line 70, in decode
    result = getattr(decoder, self._decode_function_name)()
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 439, in decode_16bit_int
    return unpack(fstring, handle)[0]
struct.error: unpack requires a buffer of 2 bytes

### Relevant debug logs

```text
Se above
```

wlcrs, 2022-10-08:
Not a bug, just some corrupt packages that were received.

closed 2022-10-08 by wlcrs

bengtner, 2022-10-08:
Ok, I'm fine with that. But it is strange that this shows up after the 2022.10.1 upgrade