wlcrs/huawei_solar #146: [Bug]: Deprecated method and attributes

BenoitDefraigne, 2022-10-08:
### System Health details

## System Information

version | core-2022.9.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.61-v8
arch | aarch64
timezone | Europe/Brussels
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4964
Installed Version | 1.27.2
Stage | running
Available Repositories | 1119
Downloaded Repositories | 2

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.0
-- | --
update_channel | stable
supervisor_version | supervisor-2022.09.1
agent_version | 1.3.0
docker_version | 20.10.17
disk_total | 27.7 GB
disk_used | 7.6 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.4.1), Terminal & SSH (9.6.1), Log Viewer (0.14.0), Samba share (10.0.0), Mosquitto broker (6.1.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 0
views | 2
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | September 28, 2022 at 7:39 PM
-- | --
current_recorder_run | October 8, 2022 at 12:15 PM
estimated_db_size | 1415.39 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Huawei Solar Setup

SUN2000L-3.68KTL
WiFi Network connection (no RS485 available)

### Describe the issue

logs indicate an deprecated method that won't be supported as of HA2022.10 - and 2022.10.1 is out there...

log contents:
Logger: homeassistant.components.number
Source: components/number/__init__.py:119
Integration: Number (documentation, issues)
First occurred: 12:15:28 PM (7 occurrences)
Last logged: 12:15:28 PM

custom_components.huawei_solar.number is setting deprecated attributes on an instance of NumberEntityDescription, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom integration author.
custom_components.huawei_solar.number::HuaweiSolarNumberEntity is overriding deprecated methods on an instance of NumberEntity, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom integration author.

### Relevant debug logs

```text
2022-10-08 21:46:38.459 WARNING (MainThread) [homeassistant.components.number] custom_components.huawei_solar.number is setting deprecated attributes on an instance of NumberEntityDescription, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom integration author.
2022-10-08 21:46:38.467 WARNING (MainThread) [homeassistant.components.number] custom_components.huawei_solar.number is setting deprecated attributes on an instance of NumberEntityDescription, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom integration author.
2022-10-08 21:46:38.471 WARNING (MainThread) [homeassistant.components.number] custom_components.huawei_solar.number is setting deprecated attributes on an instance of NumberEntityDescription, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom integration author.
2022-10-08 21:46:38.473 WARNING (MainThread) [homeassistant.components.number] custom_components.huawei_solar.number is setting deprecated attributes on an instance of NumberEntityDescription, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom integration author.
2022-10-08 21:46:38.476 WARNING (MainThread) [homeassistant.components.number] custom_components.huawei_solar.number is setting deprecated attributes on an instance of NumberEntityDescription, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom integration author.
2022-10-08 21:46:38.479 WARNING (MainThread) [homeassistant.components.number] custom_components.huawei_solar.number is setting deprecated attributes on an instance of NumberEntityDescription, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom integration author.
2022-10-08 21:46:38.482 WARNING (MainThread) [homeassistant.components.number] custom_components.huawei_solar.number::HuaweiSolarNumberEntity is overriding deprecated methods on an instance of NumberEntity, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom integration author.
```

wlcrs, 2022-10-08:
This was fixed in v1.1.1 . Upgrade your integration please

BenoitDefraigne, 2022-10-08:
Thank you for the answer.
Could indicate how to upgrade, as I would did'nt see a notification in HA nor did I find  instructions,  so I am afraid of doing it wrong and harming the integration functioning :(

Thanks again!

wlcrs, 2022-10-08:
https://hacs.xyz/docs/basic/getting_started

closed 2022-10-08 by wlcrs