wlcrs/huawei_solar #150: [Bug]: Reading from Inverter failed when trying to add inverter 2

antez88, 2022-10-12:
### System Health details

## System Information

version | core-2022.9.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.61-v7
arch | armv7l
timezone | Europe/Stockholm
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4947
Installed Version | 1.25.5
Stage | running
Available Repositories | 1119
Downloaded Repositories | 5

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 1 november 2022 01:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | eu-west-2-3.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.0
-- | --
update_channel | stable
supervisor_version | supervisor-2022.10.0
agent_version | 1.3.0
docker_version | 20.10.17
disk_total | 56.6 GB
disk_used | 6.3 GB
healthy | true
supported | true
board | rpi3
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Terminal & SSH (9.4.0), Home Assistant Google Drive Backup (0.108.2), Node-RED (13.4.0), MariaDB (2.4.0), TellStick (2.1.0), InfluxDB (4.5.0), Grafana (8.0.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 1
mode | storage

</details>

<details><summary>Nibe Uplink Heat Pumps</summary>

api_endpoint_reachable | ok
-- | --

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 3 oktober 2022 18:23
-- | --
current_recorder_run | 12 oktober 2022 20:42
estimated_db_size | 42.71 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Huawei Solar Setup

See attached images.

2 Inverters: 
SUN2000-10TKL
SUN2000-8TKL

Smart Meter 3phase

Huawei Luna 10 kWh battery

SDongle

[image]
[image]
[image]
[image]
[image]

### Describe the issue

When trying to connect through Network via SDongle (double checked IP in my TP Link Deco M5) I get an Error for one of my inverters (the master inverter where my battery and SDongle is connected).
When trying to connect just my 8TKL inverter the integration works fine.
But as soon as I try with my 10TKL, either isolated or both of them with comma separated slave IDs I get an Error.

Have double checked the slave IDs in each Inverter as seen in the snaps above.
SlaveID no 1 = 8TKL Inverter
SlaveID no 2 = 10TKL Inverter
Have also tried both ports 502, 6607 and also with on/off in all the variations with setting “elevate permissions”.
Also triple checked in both Fusion Solar and app in Inverters and Dongle that Modbus is set to Enable Unrestricted.

Lost for ideas now so I`m reaching out for help here :) 

Snap with error below:

[image]

### Relevant debug logs

```text
Huawei Solar Log:

Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py:156 
First occurred: 20:49:23 (1 occurrences) 
Last logged: 20:49:23

Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 133, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 347, in _do_read
    response = await self._client.protocol.read_holding_registers(
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 150, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 115, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 245, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 287, in get_multiple
    response = await self._read_registers(
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 378, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 141, in retry
    await _call_handlers(on_giveup, *details)
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 34, in _call_handlers
    await hdlr(details)
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 16, in f
    return coro_or_func(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 323, in backoff_giveup
    raise ReadException(
huawei_solar.exceptions.ReadException: Failed to read register 43006 after 5 tries

Felet härrör från en anpassad integration.

Logger: backoff
Source: custom_components/huawei_solar/config_flow.py:101 
Integration: Huawei Solar 
First occurred: 20:49:23 (1 occurrences) 
Last logged: 20:49:23
```

wlcrs, 2022-10-13:
This is a well done bug report. Thank you for that.

So when you setup the integration with the combination: 192.168.68.125, port 502 and slave id 1 it correctly sets up the integration for your 8TKL inverter?

In that case I would expect that "1,2" is the correct value to put in the slave id's fields.

If you're still experiencing issues. Please post the full logs (including pymodbus logs) where you first setup the integration with the 8KTL alone, then remove the integration again and try to setup again with the 10KTL included.

Note: I don't have an SDongleA or second inverter in my own installation. I've implemented support just by reading the technical manual, but without being able to test it in practice. I need proper logs of actual owners to understand how the system behaves in all possible configurations.

antez88, 2022-10-13:
BIG thanks! Opposite order works, i.e —> “1,2”.
Think it´s the only variation I haven`t tried and should have been the most logical to try :)

So at least in my case the Master Inverter (Slave 2) wasn `t the that should be first in order.

THANKS!

closed 2022-10-13 by wlcrs