wlcrs/huawei_solar #154: [Bug]: No Battery in Home Assistant

novaccr, 2022-10-20:
### System Health details

## System Information

version | core-2022.10.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.72
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.28.2
Stage | running
Available Repositories | 1124
Downloaded Repositories | 13

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 23. Oktober 2022, 02:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | false
remote_server | eu-west-2-3.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.2
-- | --
update_channel | stable
supervisor_version | supervisor-2022.10.0
agent_version | 1.4.1
docker_version | 20.10.17
disk_total | 30.8 GB
disk_used | 6.5 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Samba share (10.0.0), ESPHome (2022.10.0), File editor (5.4.1), Terminal & SSH (9.6.1), evcc (nightly) (nightly)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 1
views | 4
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 10. Oktober 2022, 13:43
-- | --
current_recorder_run | 20. Oktober 2022, 08:26
estimated_db_size | 1983.56 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Huawei Solar Setup

2 x Huawei KTL-M1 
Sdongle
10 KwH Luna200

### Describe the issue

Hello, everyone,

During the integration, only my two inverters and my power meter are displayed.
The battery is always missing. It was there once and after a restart it was gone.
repeated installation of the plugin does not help and the last beta is recorded.

### Relevant debug logs

```text
2022-10-20 08:27:21.094 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2022-10-20 08:27:21.095 ERROR (MainThread) [custom_components.huawei_solar] Error fetching TA2210031196_data_update_coordinator data: Could not update TA2210031196 values: Modbus client is not connected to the inverter.
2022-10-20 08:27:22.785 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2022-10-20 08:27:22.785 ERROR (MainThread) [custom_components.huawei_solar] Error fetching TA2210031230_data_update_coordinator data: Could not update TA2210031230 values: Modbus client is not connected to the inverter.
2022-10-20 08:27:29.458 WARNING (MainThread) [homeassistant.bootstrap] Waiting on integrations to complete setup: media_player.pioneer
2022-10-20 08:27:34.726 ERROR (MainThread) [homeassistant.components.media_player] Setup of platform pioneer is taking longer than 60 seconds. Startup will proceed without waiting any longer.
2022-10-20 08:27:35.197 ERROR (SyncWorker_3) [homeassistant] Error doing job: Unclosed client session
2022-10-20 08:27:35.265 ERROR (SyncWorker_3) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
```

wlcrs, 2022-10-20:
How are you connecting to the inverter? What happens if you unplug the SDongle and restart HA?

novaccr, 2022-10-20:
connect it via wifi using the sdongle

wlcrs, 2022-10-20:
Your logs show a very poor connection to the inverter, with very frequent disconnects... Please investigate what causes those and try other connection methods if necessary.

As this is not a bug, I'll be closing the issue.

closed 2022-10-20 by wlcrs

novaccr, 2022-10-20:
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.huawei_solar
Source: custom_components/huawei_solar/__init__.py:250 
Integration: Huawei Solar 
First occurred: 14:26:24 (1 occurrences) 
Last logged: 14:26:24

Unexpected error fetching TA2210031196_data_update_coordinator data: unpack requires a buffer of 4 bytes
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 205, in _async_refresh
    self.data = await self._async_update_data()
  File "/config/custom_components/huawei_solar/__init__.py", line 250, in _async_update_data
    return await self.bridge.update()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/bridge.py", line 184, in update
    result = await _get_multiple_to_dict(INVERTER_REGISTERS)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/bridge.py", line 179, in _get_multiple_to_dict
    zip(names, await self.client.get_multiple(names, self.slave_id))
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 298, in get_multiple
    result = [await self._decode_response(registers[0], decoder)]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 240, in _decode_response
    result = reg.decode(decoder, self)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/registers.py", line 70, in decode
    result = getattr(decoder, self._decode_function_name)()
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 447, in decode_32bit_int
    handle = self._unpack_words(fstring, handle)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 361, in _unpack_words
    handle = unpack(up, handle)
struct.error: unpack requires a buffer of 4 bytes