wlcrs/huawei_solar #156: [Bug]: Can't connect to Inverter: "Unexpected Error"

Andurilll, 2022-10-21:
### System Health details

## System Information

version | core-2022.10.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.61-v8
arch | aarch64
timezone | Europe/-------
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4862
Installed Version | 1.28.2
Stage | running
Available Repositories | 1123
Downloaded Repositories | 2

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.2
-- | --
update_channel | stable
supervisor_version | supervisor-2022.10.0
agent_version | 1.4.1
docker_version | 20.10.17
disk_total | 109.3 GB
disk_used | 3.4 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | SSH & Web Terminal (12.0.2), Grafana (8.0.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 21. Oktober 2022 um 05:47
-- | --
current_recorder_run | 21. Oktober 2022 um 09:36
estimated_db_size | 0.41 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Huawei Solar Setup

Inverter Type:  Sun2000-3KTL-M1
Inverter Firmware version: ...146
SDongle present: yes / no (What is this? I have the Inverter connected over Ethernet)
Power meter present: three phase 
Battery: (LUNA2000 5kWh - not connected yet)
Battery Firmware version: ---

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Describe the issue

If i try to connect the RS485 USB over modbusTCP to the Inverter, i get a "Unexpected error"
I Activated the RS485_1 with the Parameters on the Inverter 

Baud rate: 9600
Protocol type: MODBUS
Com address: 1
[RS485 Bus Frame capture: 1] This does not exist in the Menu!

### Relevant debug logs

```text
2022-10-21 09:41:39.262 ERROR (MainThread) [custom_components.huawei_solar.config_flow] 'PrivateHuaweiModbusResponse' object has no attribute 'registers'
Traceback (most recent call last):
File "/config/custom_components/huawei_solar/config_flow.py", line 225, in async_step_setup_serial
info = await validate_serial_setup(user_input)
File "/config/custom_components/huawei_solar/config_flow.py", line 53, in validate_serial_setup
bridge = await HuaweiSolarBridge.create_rtu(
File "/usr/local/lib/python3.10/site-packages/huawei_solar/bridge.py", line 91, in create_rtu
client = await AsyncHuaweiSolar.create_rtu(port, slave_id, loop=loop)
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 187, in create_rtu
raise err
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 177, in create_rtu
await huawei_solar._initialize()
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 115, in _initialize
self.time_zone = (await self.get(rn.TIME_ZONE)).value
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 245, in get
return (await self.get_multiple([name], slave))[0]
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 292, in get_multiple
response.registers, byteorder=Endian.Big, wordorder=Endian.Big
AttributeError: 'PrivateHuaweiModbusResponse' object has no attribute 'registers'
```

wlcrs, 2022-10-21:
As stated on https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter#overview: 

⚠Warning⚠: The SDongleA-05 dongle interferes with devices connected to the RS485A1 and RS485B1 pins on the inverter. If your inverter has such a dongle installed, we recommend connecting to the Modbus-TCP port exposed by that dongle instead.

As you are saying that you have a wired connection to your inverter, this implies that you have an [SDongle](https://support.huawei.com/enterprise/en/doc/EDOC1100102027) attached to your inverter

closed 2022-10-21 by wlcrs

Andurilll, 2022-10-21:
Thank you for the fast reply🙂

Okay i get it. Didn't know this is the wrong dongle. Have never Seen a Picture and it is not displayed as SDongleA-5. Instead it is just called Dongle-1 with the Model Name: SDongleA

If i use the ModbusTCP Port exposed from this dongle, can i connect my RBPi to the inverter and to my Home Network?

Is there no other Way to connect with rs485? Do i need the dongle or can i connect the inverter without to the Network?

wlcrs, 2022-10-21:
> If i use the ModbusTCP Port exposed from this dongle, can i connect my RBPi to the inverter and to my Home Network?

Sorry, but I don't understand your question as your explanation is too limited. Further more, I cannot possibly provide free personal support for every setup. Please read the Wiki, and ask questions if things are unclear via the Discussions tab so that other members in the community can chime in.

> Is there no other Way to connect with rs485? Do i need the dongle or can i connect the inverter without to the Network?

The SDongle is only needed for wired setups. My inverter has no dongle and just directly connects to the home WiFi.