wlcrs/huawei_solar #163: [Bug]:

bekovan, 2022-10-27:
### System Health details

Unable to install package huawei-solar==2.1.2: ERROR: Cannot install huawei-solar==2.1.2 because these package versions have conflicting dependencies. ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts

### Huawei Solar Setup

Inverter Type: sun2000
Inverter Firmware version:
SDongle present: no
Power meter present: three phase / single phase / no : three phase
Battery: LG RESU xxkWh / LUNA2000 5kWh / None
Battery Firmware version:

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Describe the issue

After installing Home assistant 2022.11.0b0, the huawei_solar was not loading, noticed following error message 
Unable to install package huawei-solar==2.1.2: ERROR: Cannot install huawei-solar==2.1.2 because these package versions have conflicting dependencies. ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts

### Relevant debug logs

```text
I estimate this will be generic as it it not between inverter & home assistnt but home assistant to integration, if you need them please let me know
```

gieljnssns, 2022-10-27:
Also just installed `2022.11.0b0` and seeing the same error.

wlcrs, 2022-10-27:
@gieljnssns https://pypi.org/project/huawei-solar/#history shows that huawei-solar 2.1.2 is available. So I would expect a network problem, or PyPI outage?

@bekovan Do you have other integrations that use Modbus? Maybe there is a conflict in pyModbus versions that are required?

bekovan, 2022-10-27:
As far as I know this integration is the only one using modbus

gieljnssns, 2022-10-27:
@wlcrs 
I also have the same error as @bekovan 
When reverting to 2022.10.5 the integration is back.

bekovan, 2022-10-27:
"When reverting to 2022.10.5 the integration is back." --> same here

wlcrs, 2022-10-27:
The `backoff` dependency was causing an issue. I've upgraded the `huawei-solar` library to fix this. Can you please test beta release 1.1.4b1 ? I'd like to get a few reports that everything works as expected before doing a stable release.

bekovan, 2022-10-27:
I still have an issue :
Setup failed for custom integration huawei_solar: Requirements for huawei_solar not found: ['huawei-solar==2.1.3'].

jeveli, 2022-10-27:
> The `backoff` dependency was causing an issue. I've upgraded the `huawei-solar` library to fix this. Can you please test beta release 1.1.4b1 ? I'd like to get a few reports that everything works as expected before doing a stable release.

Can confirm it works, have now updated to latest core version and 1.1.4b1

wlcrs, 2022-10-27:
> I still have an issue : Setup failed for custom integration huawei_solar: Requirements for huawei_solar not found: ['huawei-solar==2.1.3'].

Make sure to install 1.1.4b1, not 1.1.3b2 or earlier.

bekovan, 2022-10-27:
Possible I downloaded 1.1.3b2 initially. Now it is working correct with 2022.11.0b0.

dMopp, 2022-10-27:
Logger: homeassistant.setup
Source: setup.py:184
First occurred: 12:56:12 (1 occurrences)
Last logged: 12:56:12

Setup failed for custom integration huawei_solar: Requirements for huawei_solar not found: ['huawei-solar==2.1.4b2'].

:(

thtro, 2022-10-27:
Same here, using latest 114b1 (Home Assistant 2022.10.4), I get this message:
Setup failed for custom integration huawei_solar: Requirements for huawei_solar not found: ['huawei-solar==2.1.4b2']

thtro, 2022-10-27:
Updated HASS to 2022.10.5 and restarted a few times, still same error as above and additionally this appeared:
Unable to install package huawei-solar==2.1.4b2: ERROR: Cannot install huawei-solar==2.1.4b2 because these package versions have conflicting dependencies. ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts

wlcrs, 2022-10-27:
Ok, I understand now what's going on. This is caused by the backoff-dependency that was forced in home assistant to be <2.0 before 2022.11.0b0, but is now required to be >=2.0 in the new beta version.

So for now:
- 2022.10 or lower are compatible with all versions of this integration except the new beta
- 2022.11.0b0 is only compatible with the new beta.

wlcrs, 2022-10-27:
I've created the issue upstream https://github.com/home-assistant/core/issues/81085, and have released a new beta-version that is compatible with both 2022.11.x and 2022.10.x (and below).

I've retracted version 1.1.4b1 and replaced it with version 1.1.4b2. The only difference is that I don't set any restriction on the version of `backoff` anymore.

dMopp, 2022-10-27:
Works fine with current stable, good job:)

gieljnssns, 2022-10-27:
Works with current beta. 
thanks

bekovan, 2022-10-27:
thank you for the quick support !

closed 2022-10-27 by wlcrs