wlcrs/huawei_solar #164: [Bug]:

jeveli, 2022-10-27:
### System Health details

## System Information

version | core-2022.10.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.61-v7
arch | armv7l
timezone | Europe/Helsinki
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.28.3
Stage | running
Available Repositories | 1202
Downloaded Repositories | 27

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | November 2, 2022 at 2:00 AM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | eu-west-2-4.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.3
-- | --
update_channel | beta
supervisor_version | supervisor-2022.10.1
agent_version | 1.4.1
docker_version | 20.10.18
disk_total | 13.5 GB
disk_used | 7.4 GB
healthy | true
supported | true
board | rpi3
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.15.0), File editor (5.4.1), SSH & Web Terminal (12.0.2), Home Assistant Google Drive Backup (0.108.4), Mosquitto broker (6.1.3), FTP (4.6.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 22
views | 9
mode | storage

</details>

<details><summary>Nibe Uplink Heat Pumps</summary>

api_endpoint_reachable | ok
-- | --

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | October 17, 2022 at 5:26 AM
-- | --
current_recorder_run | October 27, 2022 at 10:09 AM
estimated_db_size | 1205.72 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Huawei Solar Setup

Inverter Type:
Inverter Firmware version: 
SDongle present: yes
Power meter present: three phase 
Battery: LG RESU xxkWh / LUNA2000 10kWh

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Describe the issue

When updating to Home Assistant Core to version 2022.11.0b0 this morning, the Huawei integration stopped working.
Tried to restart it, restarted home assistant but nothing worked, when reverted to the previous version 2022.10.5 it started to woek again.
Screenshots of the errors: 
[image]

[image]

[image]

[image]

### Relevant debug logs

```text
I did not copy the log.
```

wlcrs, 2022-10-27:
Duplicate of #163

closed 2022-10-27 by wlcrs

stomko11, 2022-11-03:
Updated to 2022.11 final  and the issue is still there. Following logs can be observed:
'''
Logger: homeassistant.util.package
Source: util/package.py:98
First occurred: 07:06:59 (3 occurrences)
Last logged: 07:07:03

Unable to install package huawei-solar==2.0.2: ERROR: Cannot install huawei-solar==2.0.2 because these package versions have conflicting dependencies. ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts [notice] A new release of pip available: 22.2.1 -> 22.3 [notice] To update, run: pip install --upgrade pi
'''
And
'''
Logger: homeassistant.setup
Source: setup.py:184
First occurred: 07:07:03 (1 occurrences)
Last logged: 07:07:03

Setup failed for custom integration huawei_solar: Requirements for huawei_solar not found: ['huawei-solar==2.0.2'].
'''

wlcrs, 2022-11-03:
Please update to the latest version of the integration.

stomko11, 2022-11-03:
Right. I noticed too late I am posting to closed topic. Thanks.