wlcrs/huawei_solar #165: [Bug]: Integrations keeps on being disconnected from inverter/battery several times a day ( TimeOut?)

cmd85, 2022-10-28:
### System Health details

## System Information

version | core-2022.10.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.74
arch | x86_64
timezone | Europe/Stockholm
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4868
Installed Version | 1.28.3
Stage | running
Available Repositories | 1133
Downloaded Repositories | 21

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | November 22, 2022 at 01:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-west-2-0.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Easee EV Charger</summary>

component_version | 0.9.46
-- | --
reach_easee_cloud | ok
connected2stream | true

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.3
-- | --
update_channel | stable
supervisor_version | supervisor-2022.10.0
agent_version | 1.4.1
docker_version | 20.10.18
disk_total | 62.3 GB
disk_used | 13.6 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.1.3), File editor (5.4.1), Terminal & SSH (9.6.1), Z-Wave JS (0.1.74), Z-Wave JS UI (1.2.0), Zigbee2MQTT (1.28.0-1), Studio Code Server (5.4.0), Node-RED (13.5.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 13
views | 14
mode | storage

</details>

<details><summary>Nibe Uplink Heat Pumps</summary>

api_endpoint_reachable | ok
-- | --

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | October 19, 2022 at 05:03
-- | --
current_recorder_run | October 28, 2022 at 07:28
estimated_db_size | 505.40 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Huawei Solar Setup

Inverter Type: Sun20000
Inverter Firmware version: latest
SDongle present: yes
Power meter present: three phase 
Battery: LUNA2000 10kWh
Battery Firmware version:

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Describe the issue

The integration is getting disconnected and I have to manually reload it.
It fixes it 50% of the time or I have to wait and try a bit later. 

Now its stuck on Initializing after trying to reload it.

### Relevant debug logs

```text
Logger: homeassistant.config_entries
Source: custom_components/huawei_solar/__init__.py:71
Integration: Huawei Solar
First occurred: 10:58:19 (4 occurrences)
Last logged: 14:58:11

Error setting up entry SUN2000-10KTL-M1SUN2000-12 for huawei_solar
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 365, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/huawei_solar/__init__.py", line 158, in async_setup_entry
    raise err
  File "/config/custom_components/huawei_solar/__init__.py", line 71, in async_setup_entry
    await primary_bridge.login(
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/bridge.py", line 318, in login
    if not await self.client.login(username, password, self.slave_id):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 590, in login
    login_response = await self._client.protocol.execute(login_request)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```

wlcrs, 2022-10-28:
A TimeoutError occurs when the inverter does not reply to the request within a reasonable amount of time. This is typically caused by a poor connection to the inverter. I've adapted the integration setup-logic so that Home Assistant will now recognize it as a retryable error. (cfr. [Handling Setup Failures](https://developers.home-assistant.io/docs/integration_setup_failures/) )

The integration already has a very thorough retry-policy: all failing requests are retried up to five times, each time with longer pauses in between. Apart from that, the data_update_coordinator will try to fetch data from the inverter every 30 seconds, and restart the connection if necessary.

I've released 1.1.4 beta 3 which includes this fix.

I would advice you to check the stability of the connection between your SDongle and home network. Maybe use the [ping-component](https://www.home-assistant.io/integrations/ping/) from Home Assistant to get some data on its stability?

cmd85, 2022-10-28:
Ok, yes it's ot the strongest connection but it's a bit weird it would fail occasionally like this, signal shouldn't change drastically. 

I have to have a repeater wifi network between hour and inverter position since I can't connect via normal one. 

You don't happen to know a IP over power which works as a repeater ? :)

cmd85, 2022-10-28:
Notice after a reboot it now says this, then tries to initialize again and repeat.
Same source you think ? 
"
Retrying setup: Could not read register value, has another device interrupted the connection?
"

wlcrs, 2022-10-28:
> You don't happen to know a IP over power which works as a repeater ? :)
These seem widely available? https://www.coolblue.be/en/powerline-adapters/wifi

> Notice after a reboot it now says this, then tries to initialize again and repeat.
> Same source you think ?
> "
> Retrying setup: Could not read register value, has another device interrupted the connection?
> "

You typically get this message if someone/something else connects to the Modbus interface: all previous connections get closed, leading to this error. This can of course also be caused by poor stability of the connection.

closed 2022-10-29 by wlcrs

cmd85, 2022-10-30:
Low prio fyi message:

Upgraded my repeater to a proper router one, got a powerline on the way also if needed. 

But finally after some repeater networking with the new one it's bow working again, let's see if the powerline is needed later :);

wlcrs, 2022-10-30:
I revisited the underlying library logic today, and made some improvements along the way: I now immediately retry the login when I get a Timeout. I already did this for other (read) requests, and have copied this logic to the login sequence too.

I've also prepared an upgrade to the just-released pyModbus 3.0, but encountered a bug in 3.0.0. This has been fixed, and next week I expect 3.0.1 to be released, after which I can upgrade that library as well. The 3.0 version is a very welcome cleanup of that library, and I hope that the stability will improve a bit as well.

That's as much as I can do from my side. Hopefully these improvements + your new hardware create a stable setup for you.