wlcrs/huawei_solar #166: [Bug]: Optimizers are offline results that the integration doesn't load

Petta71, 2022-10-28:
### System Health details

## System Information

version | core-2022.10.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.61-v8
arch | aarch64
timezone | Europe/Stockholm
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4993
Installed Version | 1.27.2
Stage | running
Available Repositories | 1143
Downloaded Repositories | 19

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 3 november 2022 01:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-central-1-4.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Easee EV Charger</summary>

component_version | 0.9.45
-- | --
reach_easee_cloud | ok
connected2stream | true

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.3
-- | --
update_channel | stable
supervisor_version | supervisor-2022.10.0
agent_version | 1.4.1
docker_version | 20.10.18
disk_total | 219.4 GB
disk_used | 18.1 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.4.1), AppDaemon 3 (5.0.1), AirCast (3.4.1), MariaDB (2.5.1), InfluxDB (4.5.0), Grafana (8.0.2), Samba share (10.0.0), FTP (4.6.1), SSH & Web Terminal (12.0.2), phpMyAdmin (0.8.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 15
views | 75
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 27 oktober 2022 16:16
-- | --
current_recorder_run | 28 oktober 2022 19:34
estimated_db_size | 491.53 MiB
database_engine | mysql
database_version | 10.6.8

</details>

### Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1
Inverter Firmware version: V100R001C00SPC146
SDongle present: yes 
Power meter present: three phase 
Battery: LUNA2000 10 kWh 
Battery Firmware version:  V100R002C00SPC114 

### How do you connect to the inverter?

Via the SDongle, wired connection

### Describe the issue

I have optimizers installed in my system and there is a problem with these. They are found by the inverter but they are offline. As a result of that the integration can't start. The optimizer problem is with with Huawei but in meantime it would be great to have option to set optimizers as fault in the integration. 

### Relevant debug logs

```text
Logger: homeassistant.config_entries
Source: custom_components/huawei_solar/__init__.py:100 
Integration: Huawei Solar 
First occurred: 20:13:03 (1 occurrences) 
Last logged: 20:13:03

Error setting up entry SUN2000-10KTL-M1SUN2000-12 for huawei_solar
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 365, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/huawei_solar/__init__.py", line 142, in async_setup_entry
    raise err
  File "/config/custom_components/huawei_solar/__init__.py", line 100, in async_setup_entry
    optimizer_system_infos = await bridge.get_optimizer_system_information_data()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/bridge.py", line 264, in get_optimizer_system_information_data
    system_information_data = OptimizerSystemInformationDataFile(file_data)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/files.py", line 250, in __init__
    sn.decode("ascii").rstrip("\x00"),
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 0: ordinal not in range(128)
```

wlcrs, 2022-10-28:
I need more context. Can you please post your full logs with the debug logging for the integration and pymodbus enabled? Cfr. the instructions given in the bug report form or at the end of the Readme.

What optimizers do you have? Is the problem only present when the optimizers are offline in the evening, or also during the day?

Petta71, 2022-10-28:
I have the 29 SUN2000-450W-P. No, the main problem is that the inverter find the optimizers when you conduct a search but they remain offline and gives no readings. So I think the integration find the optimizers but since they are offline something happens. Can you do something with the logs? I am kind of new to fetching logs. Is it possible to not let the integration look för optimizers? Like setting the optimizers=false

["Error setting up entry SUN2000-10KTL-M1\u0000\u0000\u0000\u0000SUN2000-12 for huawei_solar"],"level":"ERROR","source":["custom_components/huawei_solar/__init__.py",100],"timestamp":1666985740.477016,"exception":"Traceback (most recent call last):\n File \"/usr/src/homeassistant/homeassistant/config_entries.py\", line 365, in async_setup\n result = await component.async_setup_entry(hass, self)\n File \"/config/custom_components/huawei_solar/__init__.py\", line 142, in async_setup_entry\n raise err\n File \"/config/custom_components/huawei_solar/__init__.py\", line 100, in async_setup_entry\n optimizer_system_infos = await bridge.get_optimizer_system_information_data()\n File \"/usr/local/lib/python3.10/site-packages/huawei_solar/bridge.py\", line 264, in get_optimizer_system_information_data\n system_information_data = OptimizerSystemInformationDataFile(file_data)\n File \"/usr/local/lib/python3.10/site-packages/huawei_solar/files.py\", line 250, in __init__\n sn.decode(\"ascii\").rstrip(\"\\x00\"),\nUnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 0: ordinal not in 

2022-10-28 21:44:06.261 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [547102128208] Sending {"id":241,"type":"result","success":true,"result":{"domain":"huawei_solar","name":"Huawei Solar","config_flow":true,"documentation":"https://www.home-assistant.io/integrations/huawei_solar","issue_tracker":"https://github.com/wlcrs/huawei_solar/issues","requirements":["pyserial==3.5","pyserial-asyncio==0.6","huawei-solar==2.1.2"],"codeowners":["@wlcrs"],"iot_class":"local_polling","version":"1.1.2","loggers":["huawei_solar","pymodbus"],"is_built_in":false}}"

wlcrs, 2022-10-29:
No, It is not possible to configure the integration to skip scanning for the optimizers.

You can retrieve the full logs by clicking the "Load full logs" button on the bottom of  Settings > System > Logs . Make sure to add the extra logger-configuration to your `configuration.yaml`  and do a restart before retieving those full logs.

Petta71, 2022-10-31:
Ok. I see. The best way is to fix my case with the optimizers and get these online again. You can close this issue now. 

Thanx for a great job with the integration!

closed 2022-10-31 by Petta71