wlcrs/huawei_solar #167: [Bug]:  Problem with 1.1.4 and direct connection

bdkacz, 2022-10-29:
### Diagnostics

Hello. After update to 1.1.4   Integration cannot connect (cannot be loaded) if we have a sticking connection:

<inverter> --wifi sun2000-xxxx --- <ap in client mode> - our LAN --- <Home assistant> - in wiki method descrtibed as Using DD-WRT

In logs error  

### Huawei Solar Setup

Inverter Type: SUN2000-6-KTL-M0 also SUN2000-30-KTL-M3
Inverter Firmware version: latest
SDongle present: yes 
Power meter  no
Battery:  None
Battery Firmware version:

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Describe the issue

As in Diagnostics section. Tesdteo on  two different HA installatioon - one with Huawei SUN200-6KTL-M0 and second with Huawei SUN2000-30KTL-M3. BTW  in place where I have m 30KTL-M3 I have also SUN200-8-KLT-M1 integrated using sdongle (it support modbus over IP of dongle in LAN)  and  I can get data from 8KTL-M1

### Relevant debug logs

```text
2022-10-29 16:05:26.089 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (huawei_solar.exceptions.SlaveBusyException)
2022-10-29 16:05:27.327 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.SlaveBusyException)
2022-10-29 16:05:29.487 INFO (MainThread) [backoff] Backing off _do_read(...) for 4.0s (huawei_solar.exceptions.SlaveBusyException)
2022-10-29 16:05:33.528 INFO (MainThread) [backoff] Backing off _do_read(...) for 8.0s (huawei_solar.exceptions.SlaveBusyException)
```

wlcrs, 2022-10-29:
Thanks for adding the extended logging. This is really helpful for me.

Can you please add the same logs from your working setup on 1.1.2? This will allow me to compare and find the root cause.

bdkacz, 2022-10-29:
`2022-10-29 17:00:30.747 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting to 192.168.8.1:502.
2022-10-29 17:00:30.747 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-10-29 17:00:31.540 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server
2022-10-29 17:00:31.541 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-10-29 17:00:31.541 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.8.1:502.
2022-10-29 17:00:34.794 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2022-10-29 17:00:34.795 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0xa7 0xfe 0x0 0x1
2022-10-29 17:00:34.795 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2022-10-29 17:00:34.860 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x1 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x3c
2022-10-29 17:00:34.860 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x1 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x3c
2022-10-29 17:00:34.860 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-10-29 17:00:34.861 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1
2022-10-29 17:00:35.149 DEBUG (MainThread) [pymodbus.payload] [60]
2022-10-29 17:00:35.150 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000
2022-10-29 17:00:35.150 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x2 0x0 0x0 0x0 0x6 0x0 0x3 0xb7 0x98 0x0 0x1
2022-10-29 17:00:35.150 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 2
2022-10-29 17:00:35.298 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x2 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x0
2022-10-29 17:00:35.299 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x2 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x0
2022-10-29 17:00:35.299 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-10-29 17:00:35.299 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 2
2022-10-29 17:00:35.807 DEBUG (MainThread) [pymodbus.payload] [0]
2022-10-29 17:00:35.807 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000
2022-10-29 17:00:35.807 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x3 0x0 0x0 0x0 0x6 0x0 0x3 0x75 0x30 0x0 0x19
2022-10-29 17:00:35.808 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 3
2022-10-29 17:00:35.985 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x3 0x0 0x0 0x0 0x35 0x0 0x3 0x32 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x36 0x4b 0x54 0x4c 0x2d 0x4d 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x48 0x56 0x31 0x39 0x39 0x30 0x30 0x32 0x36 0x38 0x30 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2022-10-29 17:00:35.985 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x3 0x0 0x0 0x0 0x35 0x0 0x3 0x32 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x36 0x4b 0x54 0x4c 0x2d 0x4d 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x48 0x56 0x31 0x39 0x39 0x30 0x30 0x32 0x36 0x38 0x30 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2022-10-29 17:00:35.985 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-10-29 17:00:35.985 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 3
2022-10-29 17:00:42.510 DEBUG (MainThread) [pymodbus.payload] [21333, 20018, 12336, 12333, 13899, 21580, 11597, 12288, 0, 0, 0, 0, 0, 0, 0, 18518, 12601, 14640, 12338, 13880, 12337, 0, 0, 0, 0]
2022-10-29 17:00:42.510 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30071
2022-10-29 17:00:42.510 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x4 0x0 0x0 0x0 0x6 0x0 0x3 0x75 0x77 0x0 0x1
2022-10-29 17:00:42.511 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 4
2022-10-29 17:00:42.638 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x4 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x2
2022-10-29 17:00:42.639 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x4 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x2
2022-10-29 17:00:42.639 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-10-29 17:00:42.639 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 4
2022-10-29 17:00:42.750 DEBUG (MainThread) [pymodbus.payload] [2]
2022-10-29 17:00:42.751 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37200
2022-10-29 17:00:42.751 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x5 0x0 0x0 0x0 0x6 0x0 0x3 0x91 0x50 0x0 0x1
2022-10-29 17:00:42.751 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 5
2022-10-29 17:00:42.789 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x5 0x0 0x0 0x0 0x3 0x0 0x83 0x2
2022-10-29 17:00:42.789 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x5 0x0 0x0 0x0 0x3 0x0 0x83 0x2
2022-10-29 17:00:42.789 DEBUG (MainThread) [pymodbus.factory] Factory Response[131]
2022-10-29 17:00:42.789 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 5
2022-10-29 17:00:42.790 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100
2022-10-29 17:00:42.791 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x6 0x0 0x0 0x0 0x6 0x0 0x3 0x90 0xec 0x0 0x1
[... cut ...]
2022-10-29 17:00:55.223 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-10-29 17:00:55.223 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 14
2022-10-29 17:00:55.273 DEBUG (MainThread) [pymodbus.payload] [2706, 9, 2670, 9]
2022-10-29 17:00:55.273 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching HV1990026801_data_update_coordinator data in 0.144 seconds (success: True)
2022-10-29 17:01:05.766 ERROR (SyncWorker_6) [miio.miioprotocol] Got error when receiving: timed out
`

wlcrs, 2022-10-29:
Thank you. I'll work on a fix in the coming days

MEKadan, 2022-10-29:
> After rooling back to 1.1.2 all working again.

How do you roll back an HACS integration in general? And especially this?

bdkacz, 2022-10-29:
> > After rooling back to 1.1.2 all working again.
> 
> How do you roll back an HACS integration in general? And especially this?

Go yto hacks-integrations- huawei solar, three dots on the top right and select "download again", select from the list  verison 1.1.12, then after install restart HA

MEKadan, 2022-10-29:
> > > After rooling back to 1.1.2 all working again.
> > 
> > 
> > How do you roll back an HACS integration in general? And especially this?
> 
> Go to hacks-integrations- huawei solar, then three dots on the top right and select "download again", select from the list version 1.1.2, then after install restart HA

Thanks, this was very useful!

wlcrs, 2022-10-30:
Fixed in the latest release

closed 2022-10-30 by wlcrs

bdkacz, 2022-10-30:
> Fixed in the latest release

I can confirm -looks working fine  in both locations