wlcrs/huawei_solar #168: [Bug]: Connection lost during request

hkusulja, 2022-11-01:
### Describe the issue

Hello,
I have new Huawei SUN2000-6KTL-1 , I have used Huawei FusionSolar android app to setup it.
The integrated WiFi is connected to my house WiFi. Firmware is updated (software version V200R001C00SPC124 and MBUS firmware is to V100R001C00SPC320). Modbus tcp is enabled unrestricted. Installater password is set.
I can open connection from computer to port 6607.
After HACS and wlcrs/huawei_solar installation. I am unable to connect - Reading from the inverter failed.

In logs i have:  
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Connection lost during request

What other devices / connections is possible to have / how to disable them?
Should I disable Huawei cloud connection inside FusionSolar app ("Management System" connection to EU / global servers)?
How to disable "device commissioning" if enabled at all.
Integrated WiFi is enabled also as AP for initial setup / configuration still, but non clients are connected.

### Bescribe your Huawei Solar Setup

I have a new Huawei SUN2000-6KTL-1, I am using Huawei FusionSolar Android App to setup and configure inverter. The inverter integrated WiFi is configured and connected to my house WiFi.

Firmware is updated (software version V200R001C00SPC124 and MBUS firmware is to V100R001C00SPC320). Modbus TCP is enabled unrestricted. The installer password is set.

Additional huawei power meter is single phase.

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

first time setup, no integration added yet due to not configuring initialy correctly.

### Upload your relevant debug logs

```text
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Connection lost during request
```

### Please confirm the following:

- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I increased the logging configuration, restarted HA, and have copied all relevant log lines from the 'Full Logs' into the textarea.

wlcrs, 2022-11-01:
You do not need to disable the connection to the Huawei cloud servers for this integration to work.

Make sure that the integration is the only thing connecting to the inverter. This error only occurs when the modbus connection is stolen from the integration. I have nothing else to add/suggest tbh.

hkusulja, 2022-11-02:
@wlcrs thank you, this is a new device / inverter in existing network, except android app for configuration nothing was ever connected. Does somebody have more experience with the mentioned inverter and what else could be? maybe some additional setting? newer firmware issue/limitation, or something else.

xm4rcell0x, 2022-11-06:
have you already tried on port 502 ?

hkusulja, 2022-11-06:
> have you already tried on port 502 ?

this port was never opened on my device, and is still not.

I tried for days, and seems something inside device is interrupting communication so i can not get modbus tcp to get working (still the same / original error as described above). Maybe something from Huawei, but i do not know what.

xm4rcell0x, 2022-11-06:
For me the exact opposite...port 502 open and 6607 closed.