wlcrs/huawei_solar #169: [Feature Request]: Changing modbus IP

stevevg, 2022-11-01:
### Describe your feature request

Hi all,
I have installed this integration (currently using v1.1.6), and I’ve set it up to connect to the inverter via a wifi router with openwrt.
Now, I’ve been given the installer credentials and I was able to configure the inverter to send modbus tcp over my wifi/FE dongle, so that I could connect to it “directly” (same lan) instead of requiring a secondary router, near the inverter.

Is there a way to reconfigure the integration without reinstalling the integration itself (I suppose I would have to reconfigure all my sensors, in the latter case)? What I need to do, actually, is simply change the IP the integration connects to, from 192.168.200.1:6607 to 192.168.0.170:502 (I can connect via Modbus Poll).
In other words, where does this integration store its configuration? I've found something in core.config_entities, as explained [here](https://github.com/wlcrs/huawei_solar/issues/128), but changing it by hand results in the integration being unable to connect to my inverter.
I've also tried to [ask to the community](https://community.home-assistant.io/t/huawei-solar-how-to-change-modbus-ip/481235), but no luck so far.

Thanks for any suggestion

### Proper usage

- [X] I confirm that this is not a bug report

wlcrs, 2022-11-01:
No. This is not supported.

closed 2022-11-01 by wlcrs

paulelsner, 2023-11-29:
Hi, I have the same issue. Where is the IP stored, can it be overwritten manually somewhere in the Database?

mucki12, 2024-02-29:
You can change the IP and port in /homeassistant/.storage/core.config_entries.
But be careful :-)