wlcrs/huawei_solar #174: [Feature Request]: What is the "Advanded: elevate permissions" used for?

Fonzy007, 2022-11-06:
### Describe your feature request

Hi,
First off, thanks a lot for this great integration.

As part of the setup, there is that option "Advanced: elevate permissions". While browsing the documentation, I could not find any description of what it is used for and what additional functinalities it brings.

Could you please explain its use? Happy to create a PR based on your explanation to further improve the documentation.

Thanks,
Francois

### Proper usage

- [X] I confirm that this is not a bug report

wlcrs, 2022-11-06:
By default, it is only possible to read Modbus registers from the inverter when connecting over Modbus-TCP. 
This means that it's not possible to:
- write registers: for example to set battery parameters
- use some Huawei-specific Modbus extensions: necessary to read optimizer data

Elevating permissions performs the same login sequence as done by the "Device commissioning" part of the FusionSolar App. This gives you unlimited access to everything that can be configured/requested from the inverter.

Note that when connecting via the RS485 pins on the inverter, it is not necessary to perform this login sequence: you already have unlimited access.

Note that I already explained this concisely in the README: 
> You should only check the `Advanced: elevate permissions` checkbox if you intend to dynamically change your battery settings or want access to the optimizer data.

I don't know if it would be a good idea to expand on this somewhere else? I'm rather fond of keeping the setup instructions as short and 'to the point' as possible.

Fonzy007, 2022-11-06:
sorry I missed the part you highlighted. 

if you don't feel it to add the info to the doc, no problem. anyway, this issue is now logged with the topic discussed - hence people can find it back. I gonna try the connection using RS485 in the future if I find myself limited with the TCP option.

closed 2022-11-06 by wlcrs