wlcrs/huawei_solar #176: Documentation of sensors for "Energy Distribution" and "power-flow-card", and "Load"

SensorsIot, 2022-11-09:
### Describe your feature request

I am in the process of integrating my SUN2000 inverter into HA and would like to use the "Energy Distribution" and the "power-flow-card". I also would like to document the important sensors in Grafana to get data for optimization.

It would be very time-saving if we would get your proposal for which sensors to use for these two diagrams.
For example, I had to create a template sensor to adjust the flow direction for the battery and the grid. Maybe I missed something.

I also would be interested in your proposal on how to calculate the "load" (AFAIK there is no register in the SUN2000 providing this data). For me, this is a very important number to optimize my home and I would like to have it as a sensor to automatically document its change over time.

### Proper usage

- [X] I confirm that this is not a bug report

wlcrs, 2022-11-09:
I already shared the configuration of the graphs I personally use here: https://github.com/wlcrs/huawei_solar/wiki/Visualisations. 

Feel free to create an entry in the Wiki with the setup you end up with.

closed 2022-11-09 by wlcrs

iggnator, 2022-12-19:
For lack of your email address, I'm attaching the excel file for the AWS-301L power meter MODBUS protocol.
[English-Communication protocol MODBUS RTU for KWS-301L.ods](https://github.com/wlcrs/huawei_solar/files/10260634/English-Communication.protocol.MODBUS.RTU.for.KWS-301L.ods)
Forgive my very poor software design for Arduino, I used the MEGA2560 as I wanted the dual serial bus for debugging. I had to save it as a .txt file as github did not accept a .ino for attaching.
[MODBUS_POWER_METER.txt](https://github.com/wlcrs/huawei_solar/files/10260739/MODBUS_POWER_METER.txt)

SensorsIot, 2022-12-19:
I created a repo with your content: https://github.com/SensorsIot/KWS-AC301 
Thank you for your contribution.