wlcrs/huawei_solar #192: [Bug]: Integration fail to start with 1.2.0b1

xm4rcell0x, 2022-11-27:
### Describe the issue

`## System Information

version | core-2022.11.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.74
arch | x86_64
timezone | Europe/Rome
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 3659
Installed Version | 1.28.3
Stage | running
Available Repositories | 1222
Downloaded Repositories | 41

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 4

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.3
-- | --
update_channel | stable
supervisor_version | supervisor-2022.11.2
agent_version | 1.4.1
docker_version | 20.10.18
disk_total | 30.8 GB
disk_used | 10.6 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Network UPS Tools (0.11.0), TasmoAdmin (0.21.0), SSH & Web Terminal (12.1.3), Samba share (10.0.0), Studio Code Server (5.4.1), File editor (5.4.2), Mosquitto broker (6.1.3), Bentel Absoluta (1.1), MariaDB (2.5.1), Home Assistant Google Drive Backup (0.109.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 24
views | 33
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 17 novembre 2022 22:24
-- | --
current_recorder_run | 27 novembre 2022 13:31
database_engine | mysql
database_version | 10.6.8

</details>

<details><summary>Xiaomi Miot Auto</summary>

component_version | 0.7.2
-- | --
can_reach_server | pending
can_reach_spec | ok
logged_accounts | 0
total_devices | 0

</details>`

The integration fails to start after the upgrade from 1.1.6 to 1.2.0b1 , HA report this on every reboot:
`Logger: homeassistant.setup
Source: setup.py:192
First occurred: 13:31:05 (1 occurrences)
Last logged: 13:31:05

Setup failed for custom integration huawei_solar: Unable to import component: cannot import name 'AsyncModbusSerialClient' from 'pymodbus.client' (/usr/local/lib/python3.10/site-packages/pymodbus/client/__init__.py)`

### Bescribe your Huawei Solar Setup

Inverter Type: 10KTL-M1
Inverter Firmware version: 147
SDongle present: no
Power meter present: three phase 
Battery: LUNA2000 20kWh
Battery Firmware version: 116

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
logs: https://gist.github.com/xm4rcell0x/1b09cd9cd82cc36c0302b85d7d58e222
```

### Please confirm the following:

- [ ] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I increased the logging configuration, restarted HA, and have copied all relevant log lines from the 'Full Logs' into the textarea.

wlcrs, 2022-11-28:
This error seems to indicate that the new version of pyModbus was not installed properly. Do you have any other integrations that also use modbus? This can cause a version conflict :-/ 

Can you upload the diagnostics file of another integration? This includes a list of all the installed Python dependencies, which will allow me to verify the cause.

wlcrs, 2022-11-28:
Ok, that explains the conflict. Welcome in the Python dependency hell.

An update to pyModbus to v3 is in the works afaik: https://github.com/riptideio/pymodbus/issues/1146#issuecomment-1296237506 . 

For now, you'll have to use the previous version of this integration until the modbus integration is upgraded to use v3 of pyModbus.

closed 2022-11-28 by wlcrs

FredericMa, 2023-01-19:
I noticed that pymodbus V3.1 has been released a few days ago and it has also been updated in the HA repo so it will probably be included in the 2023.2 release. Any chance to bump the version in this integration so I can use the latest and greatest of this integration in the next HA release 😄

xm4rcell0x, 2023-01-20:
that would be great!

wlcrs, 2023-01-20:
v1.2.0 has been released, but there still seems to be an issue with serial setups. To be investigated...

martinarva, 2023-01-25:
Upgraded to 1.2 and i'm on Home Assistant 2023.1.7, Frontend 20230110.0 - latest.

After restart there is an error:
#######
Logger: homeassistant.setup
Source: setup.py:210
First occurred: 2:14:18 PM (1 occurrences)
Last logged: 2:14:18 PM

Setup failed for custom integration huawei_solar: Unable to import component: cannot import name 'AsyncModbusSerialClient' from 'pymodbus.client' (/usr/local/lib/python3.10/site-packages/pymodbus/client/init.py)
########

Tried to downgrade to 1.16, but error remains.

Is it related to same issue here?

xm4rcell0x, 2023-02-02:
Try now to upgrade to huawei Solar 1.2.3 on HA Core 2023.2
For me works like charm