wlcrs/huawei_solar #195: [Feature Request]: add

VederJay, 2022-11-29:
### Describe your feature request

Allow changing the minimal power consumption at which the battery starts compensating for it, allowing a dynamic peak shaving.

This setting is available in the FusionSolar app under "Power Adjustment" --> "Battery Control" and is called "Maximum grid power during battery discharge".

[image]

Context (i.e. why this has become important): in Belgium, the cost of power distribution (pre 2023) is divided over consumers according to how much energy they consume. Starting January 1st 2023, with the introduction of the "capaciteitstarief", this will change. The amount of peak power will become a major factor. It is calculated based on your average consumption per quarter, with the highest such average during a month taken as the "peak" of that month. The average of the last 12 months, using the monthly "peaks", determines the cost you'll pay. (So, a single high consumption for 15 minutes is enough to drive up your costs for that month and for the entire year thereafter.)

The consequence of this new way of cost distribution is that it becomes important to "shave the peaks". When you can discharge e.g. 2 kWh from the battery, it is beneficial to not use that for the _first_ 2 kWh of consumption but to instead save it up and only discharge for any consumption which exceeds 2.5 kWh. For an equal discharge of 2 kWh, the latter approach will result in lower billing.

Of course, this becomes a balancing act. On the one hand, you want to use whatever you could save in the battery to shave only the highest peaks, but on the other hand you don't want to have _not_ used part of your stored energy because you were saving it for peaks that never came. So the "height" of the peak shaving will be determined based on your expected excess PV and your expected peaks (e.g. laundry day). 
This means changing the "height" of the peak shaving will be a frequent occurrence, be it manually or (later on) through intelligent prediction and automation.

### Proper usage

- [X] I confirm that this is not a bug report

wlcrs, 2022-11-29:
Hello fellow Belgian 👋

These registers are not (yet) documented by Huawei in their Modbus Definitions PDF. I've even contacted Huawei to ask for them (and the other new battery peak compensation-related register information), but they could not help me.

I've not been able to reverse engineer/Wireshark the necessary information, as the latest Android-version seems to interfere with the packet capture software 😕 I don't have my old phone anymore where that did work properly...

If anyone would be able to supply me with the correct modbus register documentation, I'll gladly implement it.

VederJay, 2022-11-29:
Thanks for your answer! I though this would be a straightforward feature but the setting I indicated is not doing what I thought it would do. Apparently there's a new functionality concerning capacity control which I don't have on my system yet. 

I did notice in the documentation of this new functionality that they limit the range of peak power shaving between 0W-1000W, which is annoying with the Belgian "minimum" of 2500W... But it's still better than nothing of course. 

If I find more info somewhere, I'll let you know.

wlcrs, 2022-11-29:
Afaik it's 1000kW not 1000W. You need to set it to 2.5kW in our case. Works nice in practice: 
[image]

wlcrs, 2022-12-25:
Available starting from 1.2.0 beta 3

closed 2022-12-25 by wlcrs