wlcrs/huawei_solar #199: [Bug]: Failed to execute forcible charge service

jires, 2022-12-03:
### Describe the issue

I have an automation, which triggers huawei_solar.forcible_charge service. After last update, I encounter that script fails most of the time (maybe 8 times out of 10).
I figured out, that is happening only when script is started at full minute, but when is delayed fx. 2 seconds, so is starting ( I've tried only 4-5 times, and it has started every time).
I've tried both, wifi connection to dongle and ethernet connection, same issue.

Code of automation, which fails (automation starts at full minute):

`alias: Luna_charge_duration
description: ""
trigger:
  - platform: time
    at: input_datetime.force_charge_starttime
condition:
  - condition: state
    entity_id: automation.luna_charge_soc
    state: "off"
action:
  - service: huawei_solar.forcible_charge
    data:
      device_id: 71192e030e255f1be1707d55d8bd7755
      power: "5000"
      duration: "{{states('input_number.force_charge_minuntes')}}"
  - delay:
      hours: 0
      minutes: 0
      seconds: 2
      milliseconds: 0
  - service: automation.turn_off
    data:
      stop_actions: false
    target:
      entity_id: automation.luna_charge_duration
mode: single
`
After delaying from full minute with 2 seconds, it works:

`alias: Luna_charge_duration
description: ""
trigger:
  - platform: time
    at: input_datetime.force_charge_starttime
condition:
  - condition: state
    entity_id: automation.luna_charge_soc
    state: "off"
action:
  - delay:
      hours: 0
      minutes: 0
      seconds: 2
      milliseconds: 0
  - service: huawei_solar.forcible_charge
    data:
      device_id: 71192e030e255f1be1707d55d8bd7755
      power: "5000"
      duration: "{{states('input_number.force_charge_minuntes')}}"
  - delay:
      hours: 0
      minutes: 0
      seconds: 2
      milliseconds: 0
  - service: automation.turn_off
    data:
      stop_actions: false
    target:
      entity_id: automation.luna_charge_duration
mode: single
`

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1
Inverter Firmware version: V100R001C00SPC146
SDongle present: yes 
Power meter present: three phase
Battery: LUNA2000 15kWh
Battery Firmware version: V100R002C00SPC114

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-cfc256bc12e768c9e9f3379cb7f0f2f3.json (1).txt](https://github.com/wlcrs/huawei_solar/files/10146255/config_entry-huawei_solar-cfc256bc12e768c9e9f3379cb7f0f2f3.json.1.txt)

### Upload your relevant debug logs

```text
LOG WHEN FAILED:

2022-12-03 12:09:58.399 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-12-03 12:10:00.019 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37046
2022-12-03 12:10:01.708 DEBUG (MainThread) [huawei_solar.huawei_solar] Writing to 47247: [0, 5000]
2022-12-03 12:10:04.727 ERROR (MainThread) [homeassistant.components.automation.luna_charge_duration] Luna_charge_duration: Error executing script. Unexpected error for call_service at pos 1:
Traceback (most recent call last):
File "/usr/local/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
return fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 451, in _async_step
await getattr(self, handler)()
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 684, in _async_call_service_step
await service_task
File "/usr/src/homeassistant/homeassistant/core.py", line 1744, in async_call
task.result()
File "/usr/src/homeassistant/homeassistant/core.py", line 1781, in _execute_service
await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
File "/config/custom_components/huawei_solar/services.py", line 162, in forcible_charge
await bridge.set(rn.STORAGE_FORCIBLE_CHARGE_POWER, power)
File "/usr/local/lib/python3.10/site-packages/huawei_solar/bridge.py", line 372, in set
return await self.client.set(name, value, slave=self.slave_id)
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 508, in set
response = await self.write_registers(
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 527, in write_registers
response = await self._client.protocol.write_registers(
File "/usr/local/lib/python3.10/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
resp = await asyncio.wait_for(req, timeout=self._timeout)
File "/usr/local/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
2022-12-03 12:10:04.750 ERROR (MainThread) [homeassistant.components.automation.luna_charge_duration] While executing automation automation.luna_charge_duration
Traceback (most recent call last):
File "/usr/local/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
return fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/automation/__init__.py", line 551, in async_trigger
await self.action_script.async_run(
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1524, in async_run
await asyncio.shield(run.async_run())
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 409, in async_run
await self._async_step(log_exceptions=False)
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 453, in _async_step
self._handle_exception(
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 476, in _handle_exception
raise exception
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 451, in _async_step
await getattr(self, handler)()
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 684, in _async_call_service_step
await service_task
File "/usr/src/homeassistant/homeassistant/core.py", line 1744, in async_call
task.result()
File "/usr/src/homeassistant/homeassistant/core.py", line 1781, in _execute_service
await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
File "/config/custom_components/huawei_solar/services.py", line 162, in forcible_charge
await bridge.set(rn.STORAGE_FORCIBLE_CHARGE_POWER, power)
File "/usr/local/lib/python3.10/site-packages/huawei_solar/bridge.py", line 372, in set
return await self.client.set(name, value, slave=self.slave_id)
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 508, in set
response = await self.write_registers(
File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 527, in write_registers
response = await self._client.protocol.write_registers(
File "/usr/local/lib/python3.10/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
resp = await asyncio.wait_for(req, timeout=self._timeout)
File "/usr/local/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
2022-12-03 12:10:09.002 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-12-03 12:10:09.548 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2022-12-03 12:10:10.482 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37201
2022-12-03 12:10:11.338 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-12-03 12:10:11.876 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760

LOG WHEN DELAYED 2 SECONDS:

2022-12-03 12:07:04.046 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-12-03 12:07:04.669 DEBUG (MainThread) [huawei_solar.huawei_solar] Writing to 47247: [0, 5000]
2022-12-03 12:07:05.662 DEBUG (MainThread) [huawei_solar.huawei_solar] Writing to 47083: [1]
2022-12-03 12:07:06.446 DEBUG (MainThread) [huawei_solar.huawei_solar] Writing to 47246: [0]
2022-12-03 12:07:07.323 DEBUG (MainThread) [huawei_solar.huawei_solar] Writing to 47100: [1]
2022-12-03 12:07:14.008 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I increased the logging configuration, restarted HA, and have copied all relevant log lines from the 'Full Logs' into the textarea.

wlcrs, 2022-12-03:
I'll have to add a retry-logic for writes too. I have it already in place for read-requests, but didn't want to add them to the more "dangerous" writes if not strictly necessary.

Have you also tried working with a delay of 1 second? If so, what were the results with that?

jires, 2022-12-04:
No, I didn't tried with 1 sec. delay. I will change my automations and let you know.

janmolemans, 2022-12-14:
I have  a similar automation and the same issue. I will try adding the delay for now

jires, 2022-12-15:
I have changed delay to 1 sec. Automation has been triggered about 10 times since,  and no errors so far. 👍👍

wlcrs, 2022-12-27:
Fixed in 1.2.0 beta 5

closed 2022-12-27 by wlcrs