wlcrs/huawei_solar #204: [Bug]: Power production units changed?

inigoml, 2022-12-08:
### Describe the issue

Hi.

Since November 19th, energy reporting units appears to have changed. Now I got an extremely low values.
I've tried to fix them by modifiying unit prefix in my sensor configuration without no luck.
I've battery, in order to prevent battery energy to be added, I've created a new platform with input_power as source. Has been working for nearly a year.
Not sure if has been a HA core update, a huawei_solar update or a firmware update.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-4KTL-L1
Inverter Firmware version: SPC125
SDongle present: no
Power meter present: single phase
Battery: LUNA2000 10kWh 
Battery Firmware version: SPC114

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

Energy dashboard prior November 20th
[image]

Energy dashboard after November 20th.
[image]

Sensor configuration. Take into consideration that I removed units (k) in order to get proper values. But no changes
[image]

### Upload your relevant debug logs

```text
Not yet. It's working but units appear to have changed.
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I increased the logging configuration, restarted HA, and have copied all relevant log lines from the 'Full Logs' into the textarea.

wlcrs, 2022-12-09:
What did you change on 19 November to your installation? Update to the inverter FW, update of HA, update of the integration?

Do you see the same anomalies in FusionSolar?

Please upload the diagnostics file, and a debug file containing the pymodbus traffic if you want me to diagnose this problem further.

inigoml, 2022-12-09:
Inverter was updated on December 3rd. So, it's discarded.
[image]

Fusionsolar is working properly.
[image]
[image]

So, problem must be located at integration or at HA (several upgrades in last days). Take into consideration that besides Huawei inverter, I've two microinverters connected to a smart plug. Both are reporting properly.

I do not discard that I've modified some configuration. Will activate traces.

inigoml, 2022-12-09:
I've reverted to 1.1.6 and now appears to be working properly. Will test tomorrow when sun comes up. But switching from input_power to active_power, is giving units in kWh, so appear to be fixed.
Anyway, not sure if is a bug present in 1.2.0b1 or an error from my side. I modify const.py in order to change polling period from 60 seconds to 300 seconds in order to avoid locking FusionSolar. It could be a manipulation error in const.py from my side. Not sure. Will test anyway tomorrow with both versions in order to clarify.

inigoml, 2022-12-10:
Well... I think I located the real cause of this error. It's not related to inverter or integration.
Basically, if you disable polling at integration level, energy stats are not properly calculated. 
[image]

This is my configuration for updating values. But working this way breaks Energy Dashboard.
[image]

wlcrs, 2022-12-10:
Thank you for the update.

closed 2022-12-10 by wlcrs