wlcrs/huawei_solar #209: [Bug]: "Grid charge maximum power" is capped to a value 10 times lower than reported by the inverter

llak0, 2022-12-20:
### Describe the issue

After reinstalling the integration to "elevate privileges" I observed that the inverter was capping grid battery charge during Time Of Use to 3kW, whilst FusionSolar reports a range of 0.000 to 30.000 kW for the same setting.

Before installing, I was able to set-up the inverter to charge at higher-than-3kW rates via FusionSolar manual set-up.

### Bescribe your Huawei Solar Setup

Inverter Type: 2x (two) SUN2000-4KTL-L1 set up in cascade
Inverter Firmware version: V200R001C00SPC125
SDongle present: yes
Power meter present:  single phase
Battery: LUNA2000 10kWh
Battery Firmware version: V100R002C00SPC116

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

[config_entry-huawei_solar-46433133c659b9e9fd67c933c0b810a1.json.txt](https://github.com/wlcrs/huawei_solar/files/10271954/config_entry-huawei_solar-46433133c659b9e9fd67c933c0b810a1.json.txt)

### Upload your relevant debug logs

```text
--will upload--
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I increased the logging configuration, restarted HA, and have copied all relevant log lines from the 'Full Logs' into the textarea.

wlcrs, 2022-12-20:
It's not clear what your problem is. Please provide steps to reproduce and add screenshots to show the issue

llak0, 2022-12-21:
Thanks for getting back so fast.

The steps to reproduce are:
- Set the inverter to charge from grid on Time of Use, with a maxed out grid charge power value (in my case, 30kW)
- Modify the value of "Grid charge maximum power" on Home Assistant, on an elevated permissions install of huawei_solar. It will be limited to a lower value (in my case, 3kW).
- Observe the charging active power on the battery is reduced to the capped value (3kW), even after modifying it to a higher value on FusionSolar.

This is the FusionSolar battery max grid charge set-up, showing a range reaching 30kW (ignore the overall max charge power on the top, as it's dynamically set by an automation - same behavior happens when that value is maxed out as well):
[image]

This is the setting on HomeAssistant, being capped by a max of 3kW:
[image]

wlcrs, 2022-12-21:
Thank you for explaining. I'm not sure that what you are configuring in FusionSolar and HA are the same configuration parameter.

I think I should change the name of that entity to 'Battery Grid charge power' instead of 'Battery Grid charge maximum power'.

Some notes from my investigation:

The 'Battery Grid charge maximum power' sets register 47242 '[Energy storage] Power of charge from grid', and abides to the limits set out in the documentation from Huawei:

[image]

cfr. code here:

https://github.com/wlcrs/huawei_solar/blob/936d28d8646d81938a76256a2bf775e81c71915c/number.py#L64-L72

The integration has no code that changes the '[Energy storage] Maximum power of charge from grid'.

I double verified that the unit conversion is correct for both registers, and that is the case. Note that the documentation says the unit is 'kW', but it needs to be divided by 1000 (=gain) to get to that value. Thus the actual contents of the register is in watts instead of kilo-watts.

https://github.com/wlcrs/huawei-solar-lib/blob/f1455a9f88214525ec5a2476137ed286ce351a19/src/huawei_solar/registers.py#L710-L713

Sidenote: is a 10kW LUNA2000 battery not limited to 5kW in terms of discharge (and thus also charge) power? This is neither the 30kW or 3kW of your issue description, so I don't fully understand what is going on.

llak0, 2022-12-21:
> Sidenote: is a 10kW LUNA2000 battery not limited to 5kW in terms of discharge (and thus also charge) power? This is neither the 30kW or 3kW of your issue description, so I don't fully understand what is going on.
It is a 10kW LUNA2000 battery indeed, with a 5kW charge/discharge limit - the setting on FusionSolar has a wider range, but it doesn't increase that 5kW limit, only limits it if lower.

I can attempt to manually override the max setting on huawei-solar, set it to a higher limit and try it tonight to see if it has any effect - I assume I would have to modify the line 67 of `huawei_solar/number.py` from:

```
      maximum_key=rn.STORAGE_MAXIMUM_POWER_OF_CHARGE_FROM_GRID, 
```
 to:
```
      native_max_value=30000, 
```

wlcrs, 2022-12-21:
Yes. You can override the maximum like that.

An additional observation: in my installation with a 5kWh battery (same firmware versions), I'm also limited to the same 3000W limit as the one you are experiencing. However, in my case this is consistent with what I'm seeing in FusionSolar.

[image]

Can you update your FusionSolar app to  the latest version? I have v6.23.00.161 installed.

llak0, 2022-12-21:
> Can you update your FusionSolar app to the latest version? I have v6.23.00.161 installed.

I have the same version.

> Yes. You can override the maximum like that.

Thanks!

> An additional observation: in my installation with a 5kWh battery (same firmware versions), I'm also limited to the same 3000W limit as the one you are experiencing. However, in my case this is consistent with what I'm seeing in FusionSolar.

That's interesting! I wonder how that max value is set - since it's an RW register, it's not necessarily tied to the hardware, and it may be modified by FusionSolar during device commissioning (or an old firmware may have set a different value to that register that persists after updates).

I'm checking [Huawei's docs](https://support.huawei.com/enterprise/en/doc/EDOC1100167258/ff9f37a6/setting-the-grid-tied-ess-mode) and there's [a snapshot that shows the same 30kW limit](https://download.huawei.com/mdl/image/download?uuid=33d0c0354baf4093ac1cec652cfdfb2c) that appears in my installation.

I'll go ahead, backup my HA instance, fork the component, and test if a hardcoded higher value can increase that 3kW cap to 5kW later tonight (with favorable electricity tariffs).

Crossing fingers!

llak0, 2022-12-21:
Reporting back: the battery doesn't charge with more than 3kW, despite having hardcoded a 30kW max, and setting it to 4kW.

This is weird, since this battery is supposed to be able to take up to 5kW, but it means it's not a problem with the integration, but with the inverter/battery.

closed 2022-12-25 by wlcrs

cmatte, 2022-12-25:
Hey, I'm also having issues with this, specifically "Battery Grid charge maximum power" was set at 3000 and, while attempting to set it to 2000 W, an error is thrown that the valid range to set this parameter is 0-100.0.