wlcrs/huawei_solar #210: [Feature Request]: Upstream Integration into Home Assistant

herostrat, 2022-12-22:
### Describe your feature request

Hey, thanks for this awesone integration!

I have no deep understandin into the why of costum vs internal integration (afaik often this is because of interfaces like GPIO, SPI) but as this library uses TCP and/or USB I am wondering why not try to integrate it into home assistant.
This would make it easier to not diverge from upstream.

Take this more as a question and less of a feature request, as I know this would be a lot of work and is probably a intentional decision.

Thanks again!

### Proper usage

- [X] I confirm that this is not a bug report

wlcrs, 2022-12-22:
The pull request for this has been created on January 26: https://github.com/home-assistant/core/pull/65000 . I've been maintaining that PR for nearly a year now, with no reaction from the core developers.

There seems to be no interest by any of them to move this PR forward. I've thus decided to stop putting any effort in this until I get any reaction from a core developer.

closed 2022-12-22 by wlcrs

herostrat, 2022-12-22:
Oh sorry, I did not find the PR!
Yeah I see, this is frustrating. There are a number of very interesting PRs that a in the backlog and unfortunately don't seem to get much traction.

Nonetheless, at least I am very grateful for your work (I know, this does no help much).
Again - thanks a lot