wlcrs/huawei_solar #212: [Bug]: After latest upgrade integration doesn't work

johanvh83, 2022-12-23:
### Describe the issue

Upgraded huawei_solar integration to latest version by HACS
Restarted HA but all entities are 'unavailable'.
Checked HA log but no relevant error about the integration.
Re-loaded the integration but no progress.
Enabled debug logging and restarted HA. Problem not resolved as suspected but where to find the debug log?

Anything else to check?

### Bescribe your Huawei Solar Setup

Inverter Type:  SUN2000-4.6KTL-L1
Inverter Firmware version: Unknown
SDongle present: no
Power meter present: single phase
Battery: LUNA2000 10 kWh
Battery Firmware version: Unknown

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-b67bf3ea1f0b71e75077c75ebcbdc0cc.json.txt](https://github.com/wlcrs/huawei_solar/files/10295293/config_entry-huawei_solar-b67bf3ea1f0b71e75077c75ebcbdc0cc.json.txt)

### Upload your relevant debug logs

```text
None
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I increased the logging configuration, restarted HA, and have copied all relevant log lines from the 'Full Logs' into the textarea.

peros550, 2022-12-23:
Same here . Downgraded and worked fine again

johanvh83, 2022-12-23:
Thanks to confirm so I'm not the only one. Downgrade is in progress. Hopefully wlcrs can resolve the problem

wlcrs, 2022-12-23:
I can only resolve something if I have an error message that tells me what is going wrong. 

Go to Settings -> System -> Logs, and use the 'Load full logs' button on the bottom of the page. You should see the debug information from the integration appear on there.

johanvh83, 2022-12-23:
Hi,

The only thing the full log shows is many items like:
2022-12-23 14:23:25.517 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
with different value at the end...
No error message at all.
I've uploaded the diagnosis file. It seems this diagnosis file contains all data from the inverter but all entities are 'unavailable'

johanvh83, 2022-12-23:
I've put back my latest backup from this night and all works well again.
Problem started after upgrade from 1.2.0b1 to 1.2.0b2

dMopp, 2022-12-23:
Same here, stops working with current beta

wlcrs, 2022-12-23:
instead of 'me too' comments, I would appreciate anyone giving more input on the actual issue, ie. an upload of the full logs with debugging enabled. (cfr. [manual on how to enable them](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics)) 

For the moment I have not enough info to work with.

peros550, 2022-12-23:
@wlcrs apologies for the "me too" message. I am newbie with HA, and didn't know how to get logs, so your link above helped me a lot. Here are logs from my case. The HA is connected to Inverter's wifi dongle which is connected to my router. 

So here is my log

```
2022-12-23 23:06:44.477 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-12-23 23:06:44.480 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-12-23 23:06:44.482 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration climate_ip which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-12-23 23:06:58.023 DEBUG (MainThread) [pymodbus.client.tcp] Connecting to 192.168.177.76:502.
2022-12-23 23:06:58.023 DEBUG (MainThread) [pymodbus.client.tcp] Connecting.
2022-12-23 23:06:59.188 DEBUG (MainThread) [pymodbus.client.base] Client connected to modbus server
2022-12-23 23:06:59.188 INFO (MainThread) [pymodbus.client.tcp] Protocol made connection.
2022-12-23 23:06:59.320 INFO (MainThread) [pymodbus.client.tcp] Connected to 192.168.177.76:502.
2022-12-23 23:07:00.483 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2022-12-23 23:07:00.485 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0xa7 0xfe 0x0 0x1
2022-12-23 23:07:00.487 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2022-12-23 23:07:02.786 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x78
2022-12-23 23:07:02.786 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x78
2022-12-23 23:07:02.787 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2022-12-23 23:07:02.787 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1
2022-12-23 23:07:02.858 WARNING (MainThread) [homeassistant.config_entries] Config entry '192.168.177.71' for daikin integration not ready yet: Server disconnected; Retrying in background
2022-12-23 23:07:03.140 DEBUG (MainThread) [pymodbus.payload] [120]
2022-12-23 23:07:03.143 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000
2022-12-23 23:07:03.144 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0xb7 0x98 0x0 0x1
2022-12-23 23:07:03.154 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 2
2022-12-23 23:07:03.374 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x2 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0
2022-12-23 23:07:03.375 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x2 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0
2022-12-23 23:07:03.375 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2022-12-23 23:07:03.375 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 2
2022-12-23 23:07:03.623 DEBUG (MainThread) [pymodbus.payload] [0]
2022-12-23 23:07:03.623 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000
2022-12-23 23:07:03.624 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x3 0x75 0x30 0x0 0x23
2022-12-23 23:07:03.624 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 3
2022-12-23 23:07:03.917 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x3 0x0 0x0 0x0 0x49 0x1 0x3 0x46 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x35 0x4b 0x54 0x4c 0x2d 0x4c 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x48 0x56 0x32 0x32 0x36 0x33 0x34 0x32 0x31 0x39 0x30 0x36 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x35 0x33 0x34 0x36 0x2d 0x30 0x31 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2022-12-23 23:07:03.918 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x3 0x0 0x0 0x0 0x49 0x1 0x3 0x46 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x35 0x4b 0x54 0x4c 0x2d 0x4c 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x48 0x56 0x32 0x32 0x36 0x33 0x34 0x32 0x31 0x39 0x30 0x36 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x35 0x33 0x34 0x36 0x2d 0x30 0x31 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2022-12-23 23:07:03.918 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2022-12-23 23:07:03.918 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 3
2022-12-23 23:07:03.971 DEBUG (MainThread) [pymodbus.payload] [21333, 20018, 12336, 12333, 13643, 21580, 11596, 12544, 0, 0, 0, 0, 0, 0, 0, 18518, 12850, 13875, 13362, 12601, 12342, 0, 0, 0, 0, 12337, 12343, 13619, 13366, 11568, 12597, 0, 0, 0, 0]
2022-12-23 23:07:03.971 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30071
2022-12-23 23:07:03.972 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x4 0x0 0x0 0x0 0x6 0x1 0x3 0x75 0x77 0x0 0x1
2022-12-23 23:07:03.974 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 4
2022-12-23 23:07:04.600 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x4 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2
[... cut ...]
2022-12-23 23:10:05.733 INFO (MainThread) [pymodbus.client.tcp] Protocol lost connection.
2022-12-23 23:10:05.742 DEBUG (MainThread) [pymodbus.client.tcp] Waiting 1000 ms before next connection attempt.
2022-12-23 23:10:06.744 DEBUG (MainThread) [pymodbus.client.tcp] Connecting.
2022-12-23 23:10:06.756 DEBUG (MainThread) [pymodbus.client.base] Client connected to modbus server
2022-12-23 23:10:06.756 INFO (MainThread) [pymodbus.client.tcp] Protocol made connection.
2022-12-23 23:10:06.759 INFO (MainThread) [pymodbus.client.tcp] Connected to 192.168.177.76:502.
```

clunit, 2022-12-23:
For context, I was running 1.2.0b4 without any issues since Monday, but I have not used any previous versions as I just got my SUN/LUNA installed last weekend.

yousaf465, 2022-12-24:
Me too logs added as below
```
2022-12-24 12:32:21.365 DEBUG (MainThread) [pymodbus.payload] [300]
2022-12-24 12:32:21.367 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000
2022-12-24 12:32:21.368 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x2 0x0 0x0 0x0 0x6 0x0 0x3 0xb7 0x98 0x0 0x1
2022-12-24 12:32:21.369 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 2
2022-12-24 12:32:21.420 INFO (MainThread) [homeassistant.setup] Setup of domain automation took 26.0 seconds
2022-12-24 12:32:21.649 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x2 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x0
2022-12-24 12:32:21.649 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x2 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x0
2022-12-24 12:32:21.649 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2022-12-24 12:32:21.649 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 2
2022-12-24 12:32:21.652 INFO (MainThread) [homeassistant.setup] Setup of domain tts took 0.7 seconds
2022-12-24 12:32:21.677 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.esphome
2022-12-24 12:32:21.678 INFO (MainThread) [homeassistant.components.switch] Setting up switch.esphome
2022-12-24 12:32:21.679 INFO (MainThread) [homeassistant.components.light] Setting up light.esphome
2022-12-24 12:32:21.680 INFO (MainThread) [homeassistant.components.button] Setting up button.esphome
2022-12-24 12:32:21.684 INFO (MainThread) [homeassistant.components.weather] Setting up weather.met
2022-12-24 12:32:21.712 INFO (MainThread) [homeassistant.components.islamic_prayer_times] Next update scheduled for: 2022-12-25 00:03:00
2022-12-24 12:32:21.746 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.islamic_prayer_times
2022-12-24 12:32:21.788 ERROR (MainThread) [homeassistant.components.sensor] Platform esphome does not generate unique IDs. ID 30c6f72f836c-uptime already exists - ignoring sensor.10_gal_aquarium_uptime
2022-12-24 12:32:21.852 DEBUG (MainThread) [pymodbus.payload] [0]
2022-12-24 12:32:21.853 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000
2022-12-24 12:32:21.855 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x3 0x0 0x0 0x0 0x6 0x0 0x3 0x75 0x30 0x0 0x23
2022-12-24 12:32:21.856 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 3
2022-12-24 12:32:21.900 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for camera @ 192.168.18.144: Error connecting to ('192.168.18.144', 6053): [Errno 113] Connect call failed ('192.168.18.144', 6053)
2022-12-24 12:32:21.911 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x3 0x0 0x0 0x0 0x49 0x0 0x3 0x46 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x31 0x30 0x4b 0x54 0x4c 0x2d 0x4d 0x31 0x0 0x0 0x0 0x0 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x31 0x32 0x48 0x56 0x32 0x31 0x35 0x30 0x30 0x36 0x37 0x35 0x39 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x34 0x33 0x31 0x31 0x2d 0x30 0x30 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2022-12-24 12:32:21.912 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x3 0x0 0x0 0x0 0x49 0x0 0x3 0x46 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x31 0x30 0x4b 0x54 0x4c 0x2d 0x4d 0x31 0x0 0x0 0x0 0x0 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x31 0x32 0x48 0x56 0x32 0x31 0x35 0x30 0x30 0x36 0x37 0x35 0x39 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x34 0x33 0x31 0x31 0x2d 0x30 0x30 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2022-12-24 12:32:21.912 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2022-12-24 12:32:21.912 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 3
2022-12-24 12:32:21.923 INFO (MainThread) [aioesphomeapi.reconnect_logic] Trying to reconnect to camera @ 192.168.18.144 in the background
2022-12-24 12:32:22.180 INFO (MainThread) [homeassistant.setup] Setting up cast
2022-12-24 12:32:22.181 INFO (MainThread) [homeassistant.setup] Setup of domain cast took 0.0 seconds
2022-12-24 12:32:22.217 DEBUG (MainThread) [pymodbus.payload] [21333, 20018, 12336, 12333, 12592, 19284, 19501, 19761, 0, 0, 21333, 20018, 12336, 12333, 12594, 18518, 12849, 13616, 12342, 14133, 14645, 0, 0, 0, 0, 12337, 12343, 13363, 12593, 11568, 12337, 0, 0, 0, 0]
2022-12-24 12:32:22.220 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30071
2022-12-24 12:32:22.221 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x4 0x0 0x0 0x0 0x6 0x0 0x3 0x75 0x77 0x0 0x1
2022-12-24 12:32:22.222 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 4
2022-12-24 12:32:22.303 INFO (MainThread) [homeassistant.setup] Setting up media_player
2022-12-24 12:32:22.331 INFO (MainThread) [homeassistant.setup] Setup of domain media_player took 0.0 seconds
2022-12-24 12:32:22.339 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x4 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x2
2022-12-24 12:32:22.340 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x4 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x2
2022-12-24 12:32:22.340 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2022-12-24 12:32:22.340 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 4
2022-12-24 12:32:23.877 INFO (MainThread) [homeassistant.setup] Setting up default_config
2022-12-24 12:32:23.884 INFO (MainThread) [homeassistant.components.media_player] Setting up media_player.cast
[... cut ...] set days = (uptime / 86400) | int %} {%- if days > 0 -%}
  {{ days }} days, {{ (uptime - (days * 86400)) | int | timestamp_custom('%H:%M:%S', false) }}
{%- else -%}
  {{ uptime | int | timestamp_custom('%H:%M:%S', false) }}
{%- endif -%}")' for attribute '_attr_native_value' in entity 'sensor.openwrt_uptime'
2022-12-24 12:32:25.794 INFO (MainThread) [custom_components.hacs] Stage changed: startup
2022-12-24 12:32:25.996 INFO (MainThread) [custom_components.hacs] Loading removed repositories

```

yousaf465, 2022-12-24:
tried deleting and readding using dongle's ip 
[image]

wlcrs, 2022-12-24:
Thank you for the logs, this helped me eliminate a lot of potential problems.

I've found and fixed a potential issue. Please upgrade to 1.2.0 beta 3 and test again.

(@peros550 I didn't want to sound harsh. I'm sorry if you felt personally attacked. Communicating in written text is always hard to express intent in that regard.)

dMopp, 2022-12-24:
Ah thats Fine.

the Classic +1/metoo comment is more like a „make it visible“ thing for the ppl why having logs is crucial for the devs :) GitHub is bad in allowing to distinguish between „rare bug“ and „common issue“)

yousaf465, 2022-12-24:
> Thank you for the logs, this helped me eliminate a lot of potential problems.
>
> I've found and fixed a potential issue. Please upgrade to 1.2.0 beta 3 and test again.
>
> (@peros550 I didn't want to [...]

[image]

Entities named changed which messed up the sensors.

peros550, 2022-12-24:
@wlcrs 👍new version works. Many thanks!

 By the way, couldn't find the Inverter ON/OFF switch. I was expecting it to be added in the room's view together with the switches that are already present.

yousaf465, 2022-12-24:
> @wlcrs 👍new version works. Many thanks!
> 
> By the way, couldn't find the Inverter ON/OFF switch. I was expecting it to be added in the room's view together with the switches that are already present.

it is under  inverter > configuration > 
[image]

I haven't got the update yet waiting for the 1.2.0 beta 3 update.

peros550, 2022-12-24:
For me it’s not there. Maybe it requires elevated permissions during the setup ?

wlcrs, 2022-12-24:
> For me it’s not there. Maybe it requires elevated permissions during the setup ?

Yes

yousaf465, 2022-12-24:
> > For me it’s not there. Maybe it requires elevated permissions during the setup ?
> 
> Yes

to fix the new name scheme issue I tried connecting via inverter but this happened 
[image]

Ip 192.168.200.1
Port 6607
slave id 0
advanced checked

closed 2022-12-24 by wlcrs