wlcrs/huawei_solar #214: [Bug]: missing entities

yousaf465, 2022-12-25:
### Describe the issue

Previously I was able to read all line voltages along with per phase power etc with the new voltages and current entities are missing 
[image]

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1SUN2000-12
Inverter Firmware version:
SDongle present: yes 
Power meter present: three phase 
Battery:  None

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-d88209df91d02b23abf7eb0cdff9535a.json.txt](https://github.com/wlcrs/huawei_solar/files/10299527/config_entry-huawei_solar-d88209df91d02b23abf7eb0cdff9535a.json.txt)

### Upload your relevant debug logs

```text
2022-12-25 11:40:01.041 INFO (SyncWorker_8) [homeassistant.loader] Loaded flick_electric from homeassistant.components.flick_electric
2022-12-25 11:40:18.749 ERROR (SyncWorker_6) [homeassistant.components.influxdb] Cannot connect to InfluxDB due to 'HTTPConnectionPool(host='a0d7b954-influxdb', port=8086): Max retries exceeded with url: /write?db=homeassistant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f56739090>: Failed to establish a new connection: [Errno -2] Name does not resolve'))'. Please check that the provided connection details (host, port, etc.) are correct and that your InfluxDB server is running and accessible. Retrying in 60 seconds.
2022-12-25 11:41:22.497 ERROR (SyncWorker_6) [homeassistant.components.influxdb] Cannot connect to InfluxDB due to 'HTTPConnectionPool(host='a0d7b954-influxdb', port=8086): Max retries exceeded with url: /write?db=homeassistant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f570777c0>: Failed to establish a new connection: [Errno -2] Name does not resolve'))'. Please check that the provided connection details (host, port, etc.) are correct and that your InfluxDB server is running and accessible. Retrying in 60 seconds.
2022-12-25 11:42:26.248 ERROR (SyncWorker_6) [homeassistant.components.influxdb] Cannot connect to InfluxDB due to 'HTTPConnectionPool(host='a0d7b954-influxdb', port=8086): Max retries exceeded with url: /write?db=homeassistant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f56b5ab30>: Failed to establish a new connection: [Errno -2] Name does not resolve'))'. Please check that the provided connection details (host, port, etc.) are correct and that your InfluxDB server is running and accessible. Retrying in 60 seconds.
2022-12-25 11:43:09.603 INFO (MainThread) [homeassistant.components.wled] Connection to the WLED WebSocket on 192.168.18.111 has been closed
2022-12-25 11:43:09.678 INFO (MainThread) [homeassistant.components.wled] Fetching wled data recovered
2022-12-25 11:43:30.017 ERROR (SyncWorker_7) [homeassistant.components.influxdb] Cannot connect to InfluxDB due to 'HTTPConnectionPool(host='a0d7b954-influxdb', port=8086): Max retries exceeded with url: /write?db=homeassistant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f56b5a290>: Failed to establish a new connection: [Errno -2] Name does not resolve'))'. Please check that the provided connection details (host, port, etc.) are correct and that your InfluxDB server is running and accessible. Retrying in 60 seconds.
2022-12-25 11:44:33.741 ERROR (SyncWorker_0) [homeassistant.components.influxdb] Cannot connect to InfluxDB due to 'HTTPConnectionPool(host='a0d7b954-influxdb', port=8086): Max retries exceeded with url: /write?db=homeassistant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f569a3bb0>: Failed to establish a new connection: [Errno -2] Name does not resolve'))'. Please check that the provided connection details (host, port, etc.) are correct and that your InfluxDB server is running and accessible. Retrying in 60 seconds.
2022-12-25 11:45:37.529 ERROR (SyncWorker_2) [homeassistant.components.influxdb] Cannot connect to InfluxDB due to 'HTTPConnectionPool(host='a0d7b954-influxdb', port=8086): Max retries exceeded with url: /write?db=homeassistant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f569e76d0>: Failed to establish a new connection: [Errno -2] Name does not resolve'))'. Please check that the provided connection details (host, port, etc.) are correct and that your InfluxDB server is running and accessible. Retrying in 60 seconds.
2022-12-25 11:46:33.044 DEBUG (MainThread) [custom_components.hacs] Nothing in the queue
2022-12-25 11:46:41.254 ERROR (SyncWorker_2) [homeassistant.components.influxdb] Cannot connect to InfluxDB due to 'HTTPConnectionPool(host='a0d7b954-influxdb', port=8086): Max retries exceeded with url: /write?db=homeassistant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f680af0a0>: Failed to establish a new connection: [Errno -2] Name does not resolve'))'. Please check that the provided connection details (host, port, etc.) are correct and that your InfluxDB server is running and accessible. Retrying in 60 seconds.
2022-12-25 11:47:45.016 ERROR (SyncWorker_8) [homeassistant.components.influxdb] Cannot connect to InfluxDB due to 'HTTPConnectionPool(host='a0d7b954-influxdb', port=8086): Max retries exceeded with url: /write?db=homeassistant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f569a2fe0>: Failed to establish a new connection: [Errno -2] Name does not resolve'))'. Please check that the provided connection details (host, port, etc.) are correct and that your InfluxDB server is running and accessible. Retrying in 60 seconds.
2022-12-25 11:48:48.737 ERROR (SyncWorker_6) [homeassistant.components.influxdb] Cannot connect to InfluxDB due to 'HTTPConnectionPool(host='a0d7b954-influxdb', port=8086): Max retries exceeded with url: /write?db=homeassistant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f681189a0>: Failed to establish a new connection: [Errno -2] Name does not resolve'))'. Please check that the provided connection details (host, port, etc.) are correct and that your InfluxDB server is running and accessible. Retrying in 60 seconds.
2022-12-25 11:49:03.162 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-12-25 11:49:03.163 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x55 0x0 0x0 0x0 0x6 0x1 0x3 0x7d 0x40 0x0 0x34
2022-12-25 11:49:03.167 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 85
2022-12-25 11:49:03.771 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x55 0x0 0x0 0x0 0x6b 0x1 0x3 0x68 0x0 0x0 0xd 0xcf 0xf 0xe3 0xf 0xc7 0xf 0xfd 0x9 0x3e 0x9 0x10 0x9 0x35 0x0 0x0 0x13 0xa 0x0 0x0 0x13 0x35 0x0 0x0 0x13 0x1e 0x0 0x0 0xf 0x8a 0x0 0x0 0xd 0x72 0x0 0x0 0x0 0x8 0x3 0xe8 0x13 0xb0 0x26 0x8 0x1 0x50 0xb 0xb8 0x2 0x0 0x0 0x0 0x63 0xa7 0xfb 0xdd 0x63 0xa7 0x25 0xd5 0x0 0x0 0xd 0x76 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x12 0xb8 0x41 0x0 0x13 0x98 0x47 0x63 0xa8 0x38 0xa8 0x0 0x0 0x0 0xdf 0x0 0x0 0x1 0xb1
2022-12-25 11:49:03.772 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x55 0x0 0x0 0x0 0x6b 0x1 0x3 0x68 0x0 0x0 0xd 0xcf 0xf 0xe3 0xf 0xc7 0xf 0xfd 0x9 0x3e 0x9 0x10 0x9 0x35 0x0 0x0 0x13 0xa 0x0 0x0 0x13 0x35 0x0 0x0 0x13 0x1e 0x0 0x0 0xf 0x8a 0x0 0x0 0xd 0x72 0x0 0x0 0x0 0x8 0x3 0xe8 0x13 0xb0 0x26 0x8 0x1 0x50 0xb 0xb8 0x2 0x0 0x0 0x0 0x63 0xa7 0xfb 0xdd 0x63 0xa7 0x25 0xd5 0x0 0x0 0xd 0x76 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x12 0xb8 0x41 0x0 0x13 0x98 0x47 0x63 0xa8 0x38 0xa8 0x0 0x0 0x0 0xdf 0x0 0x0 0x1 0xb1
2022-12-25 11:49:03.772 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2022-12-25 11:49:03.773 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 85
2022-12-25 11:49:03.831 DEBUG (MainThread) [pymodbus.payload] [0, 3535, 4067, 4039, 4093, 2366, 2320, 2357, 0, 4874, 0, 4917, 0, 4894, 0, 3978, 0, 3442, 0, 8, 1000, 5040, 9736, 336, 3000, 512, 0, 25511, 64477, 25511, 9685, 0, 3446, 0, 0, 0, 0, 0, 0, 0, 0, 0, 18, 47169, 19, 38983, 25512, 14504, 0, 223, 0, 433]
2022-12-25 11:49:03.831 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\r\xcf']
2022-12-25 11:49:03.832 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x13\n']
2022-12-25 11:49:03.832 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x135']
[... cut ...]
2022-12-25 11:49:05.786 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x02T']
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I increased the logging configuration, restarted HA, and have copied all relevant log lines from the 'Full Logs' into the textarea.

wlcrs, 2022-12-25:
Those entities are disabled by default. Have you already clicked on them and enabled them again?

yousaf465, 2022-12-25:
> Those entities are disabled by default. Have you already clicked on them and enabled them again?

that fixed it.

closed 2022-12-25 by yousaf465