wlcrs/huawei_solar #218: [Bug]: Reconnecting to Modbus after using the mobile app

Rhysgh, 2022-12-28:
### Describe the issue

When i open the fusion solar app on my iOS device, i lose the Modbus connection on home assistant, and this never reconnects. 

I cannot fina a way to re configure the connection without uninstalling and reinstalling 

### Bescribe your Huawei Solar Setup

Inverter Type: 5kw
Inverter Firmware version: latest
SDongle present: no
Power meter present: single phase 
Battery: LUNA2000 15kWh 
Battery Firmware version: latest
[image]

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

These instructions down show "Download diagnostis"

### Upload your relevant debug logs

```text
.
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I increased the logging configuration, restarted HA, and have copied all relevant log lines from the 'Full Logs' into the textarea.

wlcrs, 2022-12-28:
Please add debugging logs, the diagnostics file and steps to reproduce the problem (eg. What do you exactly do in the mobile app?)

scrichab, 2022-12-28:
i think i had the same issue - i had a running system and  i´m wondering why i didn´t get anymor data from the setup
i´m very new with home assistant and i tried 3 days to fix that but no success and so i make a comple new installation of home assistant and integration

Just in the moment i had a new system setup running - so i will make a backup of the VM and make a try with Fusion Solar Connect to see if the same happens again

scrichab, 2022-12-28:
@Rhysgh you didn´t have a dongle connected?
so you connect with the app localy to the inverter?

Rhysgh, 2022-12-29:
No dongle; I connect to the inverters AP on 192.168.200.1.

I might have figured it out,. Ill update soon.

scrichab, 2022-12-31:
is it possible to reconfigure the integration or do i have to reinstall
i will try to setup to serial connection - because this i have done during teh first installation

Rhysgh, 2023-01-04:
@wlcrs I think this is an issue on my end. Please close this and allow me time to resolve it myself. Ill come back if I need to

closed 2023-01-04 by wlcrs