wlcrs/huawei_solar #221: [Bug]:

ffm777, 2022-12-29:
### Describe the issue

With the newer releases I'm getting frequent connection errors (see log entry below). My setup is a bit special, because I have two inverters to which I connect via their integrated Wifi access points. I have setup two Wifi bridges which connect to the inverters. Each bridge has a seperate IP address so that I setup the integration twice! One for each IP. This worked perfectly until the newer versions.

### Bescribe your Huawei Solar Setup

Inverter Type: 2 inverters 10KTL-M1
Inverter Firmware version: V147
SDongle present: yes
Power meter present: three phase
Battery: LUNA2000 15kWh
Battery Firmware version: V112

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

[config_entry-huawei_solar-e3bd706cbe00098c14f5b40ce025302f.json.txt](https://github.com/wlcrs/huawei_solar/files/10318925/config_entry-huawei_solar-e3bd706cbe00098c14f5b40ce025302f.json.txt)

### Upload your relevant debug logs

```text
Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py:273
First occurred: 11:02:02 AM (108 occurrences)
Last logged: 11:38:53 AM
Modbus client is not connected to the inverter.

NoneType: None
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I increased the logging configuration, restarted HA, and have copied all relevant log lines from the 'Full Logs' into the textarea.

ffm777, 2022-12-29:
Here is the diagnostic for the second instance of the integration (the master inverter). The one uploaded above is only for the second inverter (which has the connection problems).
[config_entry-huawei_solar-22461ac02a9616b59f349488fa748081.json.txt](https://github.com/wlcrs/huawei_solar/files/10319065/config_entry-huawei_solar-22461ac02a9616b59f349488fa748081.json.txt)

wlcrs, 2022-12-30:
Hi Carlos, only the error is not enough info for me to debug the issue. I need to understand what happens before and after that error is being logged.

Please create a debug log by following the instructions described here: https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics, wait for the issue to manifest itself again, and then upload the resulting file in this issue. Thanks.

ffm777, 2022-12-30:
Hi Thjis,
here is the first debug log which is generated using v1.1.6. So the same errors happen with this older version, however not nearly as often. This is the log only for the second instance of the integration which connects to my second inverter. Hope this helps.
[home-assistant_huawei_solar_2022-12-30T14-35-30.565Z.log](https://github.com/wlcrs/huawei_solar/files/10325161/home-assistant_huawei_solar_2022-12-30T14-35-30.565Z.log)

ffm777, 2023-01-02:
Did some more digging on my end. It seems to be a network problem on my end. Could have been another device trying to connect to the inverter's AP resulting in a connection error from the HA integration. I'm now running version 1.1.6 without problems for 2 days. Will update to 1.2.x on the weekend and see if this also works fine.

closed 2023-01-07 by wlcrs