wlcrs/huawei_solar #225: [Bug]: cant connect to inverter in 1.2.0b5

etamtlosz, 2023-01-01:
### Describe the issue

After recommending to upgrade to beta, to get access to my optimizers, as stated in #222 , I can not connect to the inverter anymore.
Tried deleting the integration from HACS, multiple HA restarts, etc.
The procedure crashes in the moment this line appears in the log ```2023-01-01 19:36:24.510 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off file upload for 10.0 seconds after 3 tries```
[image]

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-5KTL-L1
Inverter Firmware version: V200R001C00SPC125
SDongle present: yes
Power meter present: single phase
Battery: LUNA2000 10kWh
Battery Firmware version :V100R002C00SPC116

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
2023-01-01 19:35:26.727 DEBUG (MainThread) [pymodbus.client.tcp] Connecting to 192.168.2.170:502.
2023-01-01 19:35:26.727 DEBUG (MainThread) [pymodbus.client.tcp] Connecting.
2023-01-01 19:35:26.729 DEBUG (MainThread) [pymodbus.client.base] Client connected to modbus server
2023-01-01 19:35:26.729 INFO (MainThread) [pymodbus.client.tcp] Protocol made connection.
2023-01-01 19:35:26.729 INFO (MainThread) [pymodbus.client.tcp] Connected to 192.168.2.170:502.
2023-01-01 19:35:27.731 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2023-01-01 19:35:27.731 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0xa7 0xfe 0x0 0x1
2023-01-01 19:35:27.731 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2023-01-01 19:35:28.000 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x2 0xd0
2023-01-01 19:35:28.000 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x2 0xd0
2023-01-01 19:35:28.000 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2023-01-01 19:35:28.001 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1
2023-01-01 19:35:28.052 DEBUG (MainThread) [pymodbus.payload] [720]
2023-01-01 19:35:28.052 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000
2023-01-01 19:35:28.053 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0xb7 0x98 0x0 0x1
2023-01-01 19:35:28.053 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 2
2023-01-01 19:35:28.109 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x2 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2
2023-01-01 19:35:28.109 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x2 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2
2023-01-01 19:35:28.109 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2023-01-01 19:35:28.109 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 2
2023-01-01 19:35:28.160 DEBUG (MainThread) [pymodbus.payload] [2]
2023-01-01 19:35:28.160 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000
2023-01-01 19:35:28.160 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x3 0x75 0x30 0x0 0x23
2023-01-01 19:35:28.160 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 3
2023-01-01 19:35:29.329 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x3 0x0 0x0 0x0 0x49 0x1 0x3 0x46 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x35 0x4b 0x54 0x4c 0x2d 0x4c 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x54 0x41 0x32 0x32 0x35 0x30 0x30 0x34 0x31 0x31 0x39 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x35 0x33 0x34 0x36 0x2d 0x30 0x31 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-01-01 19:35:29.329 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x3 0x0 0x0 0x0 0x49 0x1 0x3 0x46 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x35 0x4b 0x54 0x4c 0x2d 0x4c 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x54 0x41 0x32 0x32 0x35 0x30 0x30 0x34 0x31 0x31 0x39 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x35 0x33 0x34 0x36 0x2d 0x30 0x31 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-01-01 19:35:29.329 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2023-01-01 19:35:29.329 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 3
2023-01-01 19:35:29.380 DEBUG (MainThread) [pymodbus.payload] [21333, 20018, 12336, 12333, 13643, 21580, 11596, 12544, 0, 0, 0, 0, 0, 0, 0, 21569, 12850, 13616, 12340, 12593, 14640, 0, 0, 0, 0, 12337, 12343, 13619, 13366, 11568, 12597, 0, 0, 0, 0]
2023-01-01 19:35:29.380 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30071
2023-01-01 19:35:29.380 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x4 0x0 0x0 0x0 0x6 0x1 0x3 0x75 0x77 0x0 0x1
2023-01-01 19:35:29.381 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 4
2023-01-01 19:35:30.365 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x4 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2
2023-01-01 19:35:30.365 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x4 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2
2023-01-01 19:35:30.365 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2023-01-01 19:35:30.365 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 4
2023-01-01 19:35:30.416 DEBUG (MainThread) [pymodbus.payload] [2]
2023-01-01 19:35:30.416 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37200
2023-01-01 19:35:30.416 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x5 0x0 0x0 0x0 0x6 0x1 0x3 0x91 0x50 0x0 0x1
2023-01-01 19:35:30.417 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 5
2023-01-01 19:35:30.815 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x5 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x4
2023-01-01 19:35:30.815 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x5 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x4
2023-01-01 19:35:30.815 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
[... 265 lines cut ...]
2023-01-01 19:35:55.451 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2023-01-01 19:35:55.451 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 19
2023-01-01 19:35:55.502 DEBUG (MainThread) [pymodbus.payload] [0, 500, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2023-01-01 19:35:55.502 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2250041190_configuration_update_coordinator data in 3.394 seconds (success: True)
2023-01-01 19:35:55.502 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading file 0x45
2023-01-01 19:35:55.503 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x14 0x0 0x0 0x0 0x5 0x0 0x41 0x5 0x1 0x45
2023-01-01 19:35:55.503 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 20
2023-01-01 19:35:58.504 INFO (MainThread) [backoff] Backing off _perform_request(...) for 10.0s (asyncio.exceptions.TimeoutError)
2023-01-01 19:35:58.504 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off file upload for 10.0 seconds after 1 tries
2023-01-01 19:36:08.505 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x15 0x0 0x0 0x0 0x5 0x0 0x41 0x5 0x1 0x45
2023-01-01 19:36:08.506 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 21
2023-01-01 19:36:11.507 INFO (MainThread) [backoff] Backing off _perform_request(...) for 10.0s (asyncio.exceptions.TimeoutError)
2023-01-01 19:36:11.507 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off file upload for 10.0 seconds after 2 tries
2023-01-01 19:36:21.509 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x16 0x0 0x0 0x0 0x5 0x0 0x41 0x5 0x1 0x45
2023-01-01 19:36:21.509 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 22
2023-01-01 19:36:24.510 INFO (MainThread) [backoff] Backing off _perform_request(...) for 10.0s (asyncio.exceptions.TimeoutError)
2023-01-01 19:36:24.510 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off file upload for 10.0 seconds after 3 tries
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I increased the logging configuration, restarted HA, and have copied all relevant log lines from the 'Full Logs' into the textarea.

wlcrs, 2023-01-01:
Can you let me know if the problem persists if you retry while the inverters are online (eg. when there is enough sun to let them all generate some power)?

etamtlosz, 2023-01-01:
I'll do that in a few hours, since the Sun is setting right now here in New Zealand.

wlcrs, 2023-01-01:
Ok, thanks for the confirmation. I'll need to look into working around the timeouts you are receiving. Very strange that those are consistent after many retries.

wlcrs, 2023-01-01:
I would expect your logs to contain a stacktrace of the error that is occurring. Can you also post that error here please?

I've gone through all the log lines from your initial bugreport, and those are all valid communication between HA and the inverter. The "Backing off file upload" is something that is retried up to 6 times (as required by the documentation provided by Huawei), so I'm surprised that it's already cut of after 3 retries.

etamtlosz, 2023-01-01:
Good morning. Unfortunately the same is happening.
```2023-01-02 07:17:53.995 DEBUG (MainThread) [pymodbus.client.tcp] Connecting to 192.168.2.170:502.
2023-01-02 07:17:53.995 DEBUG (MainThread) [pymodbus.client.tcp] Connecting.
2023-01-02 07:17:53.996 DEBUG (MainThread) [pymodbus.client.base] Client connected to modbus server
2023-01-02 07:17:53.996 INFO (MainThread) [pymodbus.client.tcp] Protocol made connection.
2023-01-02 07:17:53.997 INFO (MainThread) [pymodbus.client.tcp] Connected to 192.168.2.170:502.
2023-01-02 07:17:54.998 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2023-01-02 07:17:54.998 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0xa7 0xfe 0x0 0x1
2023-01-02 07:17:54.998 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2023-01-02 07:17:55.861 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x2 0xd0
2023-01-02 07:17:55.861 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x2 0xd0
2023-01-02 07:17:55.861 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2023-01-02 07:17:55.861 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1
2023-01-02 07:17:55.912 DEBUG (MainThread) [pymodbus.payload] [720]
2023-01-02 07:17:55.912 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000
2023-01-02 07:17:55.912 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0xb7 0x98 0x0 0x1
2023-01-02 07:17:55.913 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 2
2023-01-02 07:17:56.758 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x2 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2
2023-01-02 07:17:56.758 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x2 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2
2023-01-02 07:17:56.758 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2023-01-02 07:17:56.758 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 2
2023-01-02 07:17:56.809 DEBUG (MainThread) [pymodbus.payload] [2]
2023-01-02 07:17:56.809 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000
2023-01-02 07:17:56.810 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x3 0x75 0x30 0x0 0x23
2023-01-02 07:17:56.810 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 3
2023-01-02 07:17:57.559 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x3 0x0 0x0 0x0 0x49 0x1 0x3 0x46 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x35 0x4b 0x54 0x4c 0x2d 0x4c 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x54 0x41 0x32 0x32 0x35 0x30 0x30 0x34 0x31 0x31 0x39 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x35 0x33 0x34 0x36 0x2d 0x30 0x31 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-01-02 07:17:57.559 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x3 0x0 0x0 0x0 0x49 0x1 0x3 0x46 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x35 0x4b 0x54 0x4c 0x2d 0x4c 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x54 0x41 0x32 0x32 0x35 0x30 0x30 0x34 0x31 0x31 0x39 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x35 0x33 0x34 0x36 0x2d 0x30 0x31 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-01-02 07:17:57.559 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2023-01-02 07:17:57.559 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 3
2023-01-02 07:17:57.610 DEBUG (MainThread) [pymodbus.payload] [21333, 20018, 12336, 12333, 13643, 21580, 11596, 12544, 0, 0, 0, 0, 0, 0, 0, 21569, 12850, 13616, 12340, 12593, 14640, 0, 0, 0, 0, 12337, 12343, 13619, 13366, 11568, 12597, 0, 0, 0, 0]
2023-01-02 07:17:57.610 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30071
2023-01-02 07:17:57.611 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x4 0x0 0x0 0x0 0x6 0x1 0x3 0x75 0x77 0x0 0x1
2023-01-02 07:17:57.611 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 4
2023-01-02 07:17:58.515 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x4 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2
2023-01-02 07:17:58.515 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x4 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2
2023-01-02 07:17:58.515 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2023-01-02 07:17:58.515 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 4
2023-01-02 07:17:58.566 DEBUG (MainThread) [pymodbus.payload] [2]
2023-01-02 07:17:58.567 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37200
2023-01-02 07:17:58.567 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x5 0x0 0x0 0x0 0x6 0x1 0x3 0x91 0x50 0x0 0x1
2023-01-02 07:17:58.567 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 5
2023-01-02 07:17:59.329 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x5 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x4
2023-01-02 07:17:59.329 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x5 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x4
2023-01-02 07:17:59.329 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2023-01-02 07:17:59.329 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 5
2023-01-02 07:17:59.380 DEBUG (MainThread) [pymodbus.payload] [4]
[... cut ...]
2023-01-02 07:18:38.433 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off file upload for 10.0 seconds after 2 tries
2023-01-02 07:18:48.434 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x16 0x0 0x0 0x0 0x5 0x0 0x41 0x5 0x1 0x45
2023-01-02 07:18:48.434 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 22
2023-01-02 07:18:51.436 INFO (MainThread) [backoff] Backing off _perform_request(...) for 10.0s (asyncio.exceptions.TimeoutError)
2023-01-02 07:18:51.436 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off file upload for 10.0 seconds after 3 tries```

wlcrs, 2023-01-01:
I'm afraid that what you are sending me is still incomplete for me to figure out what is going wrong. 

Can you please increase the overall 'default' logging level so that you hopefully capture the real error that is occurring?

Adjust the `logger:` entry in your `configuration.yaml` as follows and restart HA before retrying:
```
logger:
  default: info
  logs:
    pymodbus: debug 
    huawei_solar: debug
    homeassistant.components.huawei_solar: debug
```

etamtlosz, 2023-01-01:
Done. Here's the unfiltered log:
```2023-01-02 07:55:59.977 INFO (SyncWorker_7) [homeassistant.util.package] Attempting install of huawei-solar==2.2.0b6
2023-01-02 07:56:06.140 INFO (SyncWorker_5) [homeassistant.components.proximity] eni phone: proximity calculation complete
2023-01-02 07:56:06.141 INFO (SyncWorker_5) [homeassistant.components.proximity] eni phone: proximity calculation complete
2023-01-02 07:56:06.141 INFO (SyncWorker_5) [homeassistant.components.proximity] eni phone: proximity calculation complete
2023-01-02 07:56:06.142 INFO (SyncWorker_4) [homeassistant.components.proximity] eni phone: proximity calculation complete
2023-01-02 07:56:06.142 INFO (SyncWorker_4) [homeassistant.components.proximity] eni phone: proximity calculation complete
2023-01-02 07:56:23.436 INFO (MainThread) [custom_components.blueiris.api.blue_iris_api] Updating data from BI Server (wonderNVR)
2023-01-02 07:56:23.516 DEBUG (MainThread) [pymodbus.client.tcp] Connecting to 192.168.2.170:502.
2023-01-02 07:56:23.516 DEBUG (MainThread) [pymodbus.client.tcp] Connecting.
2023-01-02 07:56:23.517 DEBUG (MainThread) [pymodbus.client.base] Client connected to modbus server
2023-01-02 07:56:23.517 INFO (MainThread) [pymodbus.client.tcp] Protocol made connection.
2023-01-02 07:56:23.518 INFO (MainThread) [pymodbus.client.tcp] Connected to 192.168.2.170:502.
2023-01-02 07:56:24.519 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2023-01-02 07:56:24.519 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0xa7 0xfe 0x0 0x1
2023-01-02 07:56:24.520 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2023-01-02 07:56:24.754 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x2 0xd0
2023-01-02 07:56:24.754 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x2 0xd0
2023-01-02 07:56:24.754 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2023-01-02 07:56:24.755 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1
2023-01-02 07:56:24.805 DEBUG (MainThread) [pymodbus.payload] [720]
2023-01-02 07:56:24.806 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000
2023-01-02 07:56:24.806 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0xb7 0x98 0x0 0x1
2023-01-02 07:56:24.806 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 2
2023-01-02 07:56:24.992 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x2 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2
2023-01-02 07:56:24.992 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x2 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2
2023-01-02 07:56:24.992 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2023-01-02 07:56:24.993 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 2
2023-01-02 07:56:25.043 DEBUG (MainThread) [pymodbus.payload] [2]
2023-01-02 07:56:25.044 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000
2023-01-02 07:56:25.044 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x3 0x75 0x30 0x0 0x23
2023-01-02 07:56:25.044 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 3
2023-01-02 07:56:25.570 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x3 0x0 0x0 0x0 0x49 0x1 0x3 0x46 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x35 0x4b 0x54 0x4c 0x2d 0x4c 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x54 0x41 0x32 0x32 0x35 0x30 0x30 0x34 0x31 0x31 0x39 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x35 0x33 0x34 0x36 0x2d 0x30 0x31 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-01-02 07:56:25.571 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x3 0x0 0x0 0x0 0x49 0x1 0x3 0x46 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x35 0x4b 0x54 0x4c 0x2d 0x4c 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x54 0x41 0x32 0x32 0x35 0x30 0x30 0x34 0x31 0x31 0x39 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x35 0x33 0x34 0x36 0x2d 0x30 0x31 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-01-02 07:56:25.571 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2023-01-02 07:56:25.571 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 3
2023-01-02 07:56:25.621 DEBUG (MainThread) [pymodbus.payload] [21333, 20018, 12336, 12333, 13643, 21580, 11596, 12544, 0, 0, 0, 0, 0, 0, 0, 21569, 12850, 13616, 12340, 12593, 14640, 0, 0, 0, 0, 12337, 12343, 13619, 13366, 11568, 12597, 0, 0, 0, 0]
2023-01-02 07:56:25.622 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30071
2023-01-02 07:56:25.622 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x4 0x0 0x0 0x0 0x6 0x1 0x3 0x75 0x77 0x0 0x1
2023-01-02 07:56:25.622 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 4
2023-01-02 07:56:26.207 DEBUG (MainThread) [pymodbus.client.base] recv: 0x0 0x4 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2
2023-01-02 07:56:26.208 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x4 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2
2023-01-02 07:56:26.208 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2023-01-02 07:56:26.208 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 4
2023-01-02 07:56:26.261 DEBUG (MainThread) [pymodbus.payload] [2]
[... cut ...]
2023-01-02 07:57:06.181 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off file upload for 10.0 seconds after 2 tries
2023-01-02 07:57:16.183 DEBUG (MainThread) [pymodbus.client.base] send: 0x0 0x16 0x0 0x0 0x0 0x5 0x0 0x41 0x5 0x1 0x45
2023-01-02 07:57:16.183 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 22
2023-01-02 07:57:19.185 INFO (MainThread) [backoff] Backing off _perform_request(...) for 10.0s (asyncio.exceptions.TimeoutError)
2023-01-02 07:57:19.186 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off file upload for 10.0 seconds after 3 tries```

wlcrs, 2023-01-01:
I'm still missing any clue on what error is really occurring. Is there really nothing of potential interest that appears after what you copy&pasting now? If not, can you increase the 'default' logging level to 'debug' and try again? This should create considerable amount of output. 

Please also upload the diagnostics file of another integration, as this will help me to understand how the rest of your HA system looks like (version numbers of all installed dependencies)

etamtlosz, 2023-01-01:
OK, doing it right now.
BTW:
1. the latest production version installs without any issues, except I can not see my optimizers.
2. The beta version remains as initializing after the error. A reboot or delete is needed to get rid of it. In the logs this keeps popping up:```
3. 2023-01-02 08:06:20.940 DEBUG (MainThread) [pymodbus.client.tcp] Connecting.
2023-01-02 08:06:20.942 DEBUG (MainThread) [pymodbus.client.base] Client connected to modbus server
2023-01-02 08:06:20.942 INFO (MainThread) [pymodbus.client.tcp] Protocol made connection.
2023-01-02 08:06:20.943 INFO (MainThread) [pymodbus.client.tcp] Connected to 192.168.2.170:502.
```

Stand by for the `debug` logging level.

wlcrs, 2023-01-01:
You'll probably want to upload the home-assistant.log as a file by dragging&dropping it. This logging level will create a considerable amount of output.

FYI: I've been creating fresh HA installations and have been installing my own installer multiple times already this evening, but am unable to recreate your error for now.

etamtlosz, 2023-01-01:
[config_entry-webostv-d1f82df05b35e29580d97318bfbd7f64.json.txt](https://github.com/wlcrs/huawei_solar/files/10329333/config_entry-webostv-d1f82df05b35e29580d97318bfbd7f64.json.txt)
HA log file is larger than 25MB, so I've uploaded it to GDrive. I hope there's no sensible data in it.
https://drive.google.com/file/d/1XnxkY3bFTgvv1OcSBnhTHzxVDgoQMBF7/view?usp=share_link

wlcrs, 2023-01-01:
I've downloaded it. Feel free to delete it from your drive again.

etamtlosz, 2023-01-01:
Strange thing just happened.
Had to restart HA to change the default logger level, and after a long restart procedure - Huawei solar needed time to initialize, I have the integration up and running, unfortunately without optimizers.
```
Logger: homeassistant.bootstrap
Source: bootstrap.py:486
First occurred: 8:42:39 AM (1 occurrences)
Last logged: 8:42:39 AM

Waiting on integrations to complete setup: huawei_solar
```
Now I can provide a diag file.
[config_entry-huawei_solar-f2a0b70a5f9ecfbd9e900875a68dd88d.json.txt](https://github.com/wlcrs/huawei_solar/files/10329361/config_entry-huawei_solar-f2a0b70a5f9ecfbd9e900875a68dd88d.json.txt)

etamtlosz, 2023-01-01:
Forgot to attach the HA log file to the previous message.
[home-assistant_2023-01-01T19-48-36.882Z.log.txt](https://github.com/wlcrs/huawei_solar/files/10329366/home-assistant_2023-01-01T19-48-36.882Z.log.txt)

wlcrs, 2023-01-01:
Ok, glad to hear that we've progressed a bit so that we can better debug the issue.

```
2023-01-02 08:43:12.143 ERROR (MainThread) [backoff] Giving up _perform_request(...) after 6 tries (asyncio.exceptions.TimeoutError)
```

The integration has to eventually give up fetching the optimizer data: even after retrying 6 times, and waiting for 10 seconds on each try for a response to occur we cannot read the optimizer information file from the inverter. Your inverter is thus not adhering to the spec that Huawei defined.

It is therefore choosing to ignore the optimizers as to not hinder the setup of the rest of the integration:

```
2023-01-02 08:43:12.146 INFO (MainThread) [custom_components.huawei_solar] Cannot create optimizer sensor entities as the integration has insufficient permissions. Consider enabling elevated permissions to get more optimizer data
```
(ignore the part in the message about insufficient permissions, not applicable in your case)

A few things to consider:
- Do a full power cycle of your inverter. Power off your inverter completely: shut down the AC and DC side of your inverter, also power off your battery. Wait for all lights to go off (takes a minute or two). Then start it up back again. This can take up to 10 minutes.
- Scan your inverters again. Use the 'device commissioning' part of your FusionSolar app to connect to your inverter, go to Maintenance -> Subdevice management -> click on the 'auto search' button in the optimizer list. 
(out of curiosity: are you able to correctly see your inverters in the 'device monitoring' tab of the commissioning tool before and after doing that?)

wlcrs, 2023-01-01:
FYI: I did not learn anything from your 85MB log. I think it is incomplete because it had not been fully flushed to disk on the moment you uploaded it. It only included 2 retries instead of the 3 that you uploaded with your previous log excerpts. 

I've thus removed it from my disk again, as I don't want to store any of your -potentially- sensitive data for longer than necessary.

etamtlosz, 2023-01-01:
I did restart the whole system, ran the subdevice discovery and the 4 optimizers I have have been rediscovered, but they didn't show up in HA. So I removed the integration, added again, but Now I can't get the inverter discovered into HA, not even after a restart.

PS
I can see the inverter under device commissioning and the optimizers too, in the app.

wlcrs, 2023-01-02:
I slept on the problem for a night, and found that it probably is a bug introduced in the major version upgrade of pyModbus.

This will be tough to find the root cause, so I'll potentially have to revert that upgrade.

etamtlosz, 2023-01-02:
Good morning. I had to revert to 1.1.6 to be able to run my automations that control the maximum charging power.
If there's any way I can help, let me know. I'll be available in ~9hours.

wlcrs, 2023-01-05:
I found the root cause: the slave id was not passed correctly when retrieving the optimizer data. This caused it to default to slave_id 0 (which works for many people, including me).

Can you try again with 1.2.0b7?

etamtlosz, 2023-01-05:
Updated to b7 and got connected after the restart. All the devices are there AFAIK.
At the moment I'm not at home, to confirm that everything is spot on, but I'll be back in a few hours, do some test and hopefully I'm going to close this issue/bug.

etamtlosz, 2023-01-07:
Closing it now.
1.2.0b7 solved this issue. I tested on two different PV systems and everything works perfect.
Thanks for the help and time you invest into this integration.

closed 2023-01-07 by etamtlosz

wlcrs, 2023-01-07:
🎉