wlcrs/huawei_solar #233: [Bug]: Unable to connect via serial on 1.2.0b5 1.1.6 is working

be04376, 2023-01-03:
### Describe the issue

Upgrading to 1.2.0b5 cause connectivity issues over serial.
Tried to complete remove and reinstall the integration, but then I'm unble to setup because
huawei_solar.exceptions.ReadException: Failed to read register 43006 after 6 tries

added diagnostics file from 1.1.6

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-4.6KTL-L1
Inverter Firmware version: V200R001C00SPC125
SDongle present: no
Power meter present: single phase
Battery: LUNA2000 10kWh
Battery Firmware version: V100R002C00SPC116

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.12.8",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Brussels",
    "os_name": "Linux",
    "os_version": "5.15.80",
    "supervisor": "2022.12.1",
    "host_os": "Home Assistant OS 9.4",
    "docker_version": "20.10.19",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "eufy_security": {
      "version": "2.10.0",
      "requirements": [
        "websocket-client==1.1.0"
      ]
    },
    "solcast_solar": {
      "version": "3.0.30",
      "requirements": [
        "aiohttp>=3.6.2",
        "datetime>=4.3",
        "isodate>=0.6.0"
      ]
    },
    "smartthinq_sensors": {
      "version": "0.27.1",
      "requirements": [
        "pycountry>=20.7.3",
        "xmltodict>=0.12.0",
        "cchardet>=2.1.7"
      ]
    },
    "huawei_solar": {
      "version": "1.1.6",
      "requirements": [
        "huawei-solar==2.1.6"
      ]
[... 412 lines cut ...]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 85, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 194, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 229, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 309, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 159, in retry
    await _call_handlers(on_giveup, **details, exception=e)
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 37, in _call_handlers
    await handler(details)
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 16, in f
    return coro_or_func(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 255, in backoff_giveup
    raise ReadException(f"Failed to read register {register} after {details['tries']} tries")
huawei_solar.exceptions.ReadException: Failed to read register 43006 after 6 tries
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I increased the logging configuration, restarted HA, and have copied all relevant log lines from the 'Full Logs' into the textarea.

wlcrs, 2023-01-04:
Possibly related to https://github.com/riptideio/pymodbus/issues/1214 .

wlcrs, 2023-01-05:
That issue is apparently fixed in the next release of pyModbus, I'll need wait until they releases a new version. Can I ping you to test again when that's the case?

I'll put of promoting the current beta to stable until the problem is confirmed to be resolved.

be04376, 2023-01-05:
Sure let me now when I can test for you.

bekovan, 2023-01-05:
Just tuning in to see when the next update is arriving.

I am trying to have my battery charging during the night & I can not make it work with the stable version.
Seems in beta a lot of stuff was happening.
Do you have somewere a writeout to the function of the new parameters?

Binda5000, 2023-01-15:
I had the same issue with 1.2.0b7. But after going back to 1.1.6 it still says failed to connect.

`Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py:160
First occurred: 11:22:45 (3 occurrences)
Last logged: 11:24:17

Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 154, in create_rtu
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 85, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 198, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 233, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 313, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 278, in _do_read
    raise ConnectionException(message)
huawei_solar.exceptions.ConnectionException: Modbus client is not connected to the inverter.`

wlcrs, 2023-01-19:
This should be fixed in release 1.2.0. Can you please test again?

bekovan, 2023-01-19:
Retrying setup: Failed to read register 43006 after 6 tries

SUN2000-3.68KTL-L1

ogger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py:160
First occurred: 11:26:17 (3 occurrences)
Last logged: 11:28:08

Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 282, in _do_read
    response = await self._client.protocol.read_holding_registers(
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/base.py", line 303, in execute
    resp = await asyncio.wait_for(req, timeout=self.params.timeout)
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 154, in create_rtu
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 85, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 200, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 235, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 314, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 159, in retry
    await _call_handlers(on_giveup, **details, exception=e)
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 37, in _call_handlers
    await handler(details)
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 16, in f
    return coro_or_func(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 261, in backoff_giveup
    raise ReadException(f"Failed to read register {register} after {details['tries']} tries")
huawei_solar.exceptions.ReadException: Failed to read register 43006 after 6 tries

be04376, 2023-01-19:
Hi,

serial connection still fails with 1.2.0

2023-01-19 16:47:36.103 DEBUG (MainThread) [pymodbus.client.serial] Starting serial connection
2023-01-19 16:47:36.105 DEBUG (MainThread) [pymodbus.client.base] Client connected to modbus server
2023-01-19 16:47:36.105 INFO (MainThread) [pymodbus.client.serial] Serial connected.
2023-01-19 16:47:36.106 INFO (MainThread) [pymodbus.client.serial] Connected to /dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0
2023-01-19 16:47:37.108 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2023-01-19 16:47:37.109 DEBUG (MainThread) [pymodbus.client.base] send: 0x1 0x3 0xa7 0xfe 0x0 0x1 0xc6 0x8e
2023-01-19 16:47:37.109 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2023-01-19 16:47:40.115 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-01-19 16:47:40.115 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 1.0 seconds after 1 tries
2023-01-19 16:47:41.115 DEBUG (MainThread) [pymodbus.client.base] send: 0x1 0x3 0xa7 0xfe 0x0 0x1 0xc6 0x8e
2023-01-19 16:47:41.116 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2023-01-19 16:47:44.117 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (asyncio.exceptions.TimeoutError)
2023-01-19 16:47:44.118 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 2 tries
2023-01-19 16:47:46.119 DEBUG (MainThread) [pymodbus.client.base] send: 0x1 0x3 0xa7 0xfe 0x0 0x1 0xc6 0x8e
2023-01-19 16:47:46.120 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2023-01-19 16:47:49.122 INFO (MainThread) [backoff] Backing off _do_read(...) for 4.0s (asyncio.exceptions.TimeoutError)
2023-01-19 16:47:49.122 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 4.0 seconds after 3 tries
2023-01-19 16:47:51.549 INFO (MainThread) [homeassistant.components.mqtt.binary_sensor] No matching payload found for entity: Radiator-bathroom window open with state topic: zigbee2mqtt/Radiator-bathroom. Payload: '{"auto_lock":"AUTO","away_mode":"OFF","away_preset_days":1,"away_preset_temperature":15,"battery_low":false,"boost_time":300,"child_lock":"LOCK","comfort_temperature":20,"current_heating_setpoint":25,"eco_temperature":15,"force":"close","holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"holidays_schedule":"06:00/20°C 08:00/15°C 11:30/15°C 12:30/15°C 17:30/20°C 22:00/15°C","linkquality":36,"local_temperature":15.5,"local_temperature_calibration":-1,"max_temperature":35,"min_temperature":5,"position":0,"preset":"manual","running_state":"idle","system_mode":"off","update":{"installed_version":87,"latest_version":87,"state":"idle"},"update_available":false,"valve_detection":null,"week":"5+2","window_detection":"OFF","window_detection_params":{"minutes":80,"temperature":91},"window_open":false,"workdays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":17,"minute":31,"temperature":20},{"hour":18,"minute":31,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"workdays_schedule":"06:00/20°C 08:00/15°C 11:30/15°C 17:31/20°C 18:31/20°C 22:00/15°C"}', template output: 'False', with value template 'Template("{{ value_json.window_open }}")'
2023-01-19 16:47:51.570 INFO (MainThread) [homeassistant.components.mqtt.binary_sensor] No matching payload found for entity: Radiator-bedroom3 window open with state topic: zigbee2mqtt/Radiator-bedroom3. Payload: '{"auto_lock":"AUTO","away_mode":"OFF","away_preset_days":1,"away_preset_temperature":15,"battery_low":false,"boost_time":300,"child_lock":"LOCK","comfort_temperature":20,"current_heating_setpoint":15,"eco_temperature":15,"force":"close","holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"holidays_schedule":"06:00/20°C 08:00/15°C 11:30/15°C 12:30/15°C 17:30/20°C 22:00/15°C","linkquality":45,"local_temperature":13.5,"local_temperature_calibration":-1,"max_temperature":35,"min_temperature":5,"position":0,"preset":"manual","running_state":"idle","system_mode":"off","update":{"installed_version":87,"latest_version":87,"state":"idle"},"update_available":false,"valve_detection":null,"week":"5+2","window_detection":"OFF","window_detection_params":{"minutes":249,"temperature":12},"window_open":false,"workdays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"workdays_schedule":"06:00/20°C 08:00/15°C 11:30/15°C 12:30/15°C 17:30/20°C 22:00/15°C"}', template output: 'False', with value template 'Template("{{ value_json.window_open }}")'
2023-01-19 16:47:51.595 INFO (MainThread) [homeassistant.components.mqtt.binary_sensor] No matching payload found for entity: Radiator-bedroom1 window open with state topic: zigbee2mqtt/Radiator-bedroom1. Payload: '{"auto_lock":"AUTO","away_mode":"OFF","away_preset_days":1,"away_preset_temperature":15,"battery_low":false,"boost_t [... cut ...]
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 37, in _call_handlers
    await handler(details)
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 16, in f
    return coro_or_func(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 261, in backoff_giveup
    raise ReadException(f"Failed to read register {register} after {details['tries']} tries")
huawei_solar.exceptions.ReadException: Failed to read register 43006 after 6 tries

wlcrs, 2023-01-20:
I will need to recreate my serial test-setup again to test locally. This will take some time as it's somewhat inconveniently located for me. For now, revert to 1.1.6 (or ideally: dig in yourself to help me find the issue with the upgrade to pymodbus v3.1.0)

wlcrs, 2023-01-24:
I'm having some trouble to get my serial test setup back online. I've already been swapping out components and wiring, restarting my inverter, bricking it by selecting 'sunspec' instead of 'modbus' in a settings menu and having to do a system reset as a result, etc. 

So for now, I have already spent a few hours on this problem without any progress. I will continue to work on it when I have more time (hopefully this weekend). If someone with some Python experience wants to help with this issue, they can head over to https://gitlab.com/Emilv2/huawei-solar/-/tree/develop/ to fetch the latest huawei-solar lib code. The develop branch has the latest code (with the bug), the master branch contains the version used by v1.1.6 of this integration.

iH8c0ff33, 2023-01-25:
> So for now, I have already spent a few hours on this problem without any progress. I will continue to work on it when I have more time (hopefully this weekend). If someone with some Python [...]

Thanks for all your work with this integration, it's been really useful for me.
I'll try to find the issue in the huawei-solar library and keep you updated if I make some useful progress.

iH8c0ff33, 2023-01-25:
@wlcrs The [old pymodbus async client](https://github.com/pymodbus-dev/pymodbus/blob/c5772b35ae3f29d1947f3ab453d8d00df846459f/pymodbus/client/asynchronous/async_io/__init__.py#L744) used a default baudrate of 9600 instead of the one specified in constants.py, but in version >3.0.0 this seems to be fixed and the new default is 19200.

This integration could just always use 9600 as baudrate (i.e. hardcode it in __get_rtu_client) or add a baudate entry in the HA config flow. I already applied the first solution as a temporary fix (I'm working on my HA instance from a remote location so can't change inverter settings easily) and it is currently working (on version 1.2.0), but if you'd like to I can make a PR to add the config entry.

wlcrs, 2023-01-25:
Hey @iH8c0ff33 , thank you for identifying the issue!! I will hardcode it to 9600 for now, as this is the quickest fix.

wlcrs, 2023-01-25:
Version 1.2.1 has been released, which should fix this issue.

closed 2023-01-25 by wlcrs

be04376, 2023-01-26:
Version 1.2.1 does indeed fix this issue.
Thx for the quick support !