wlcrs/huawei_solar #243: [Bug]: Logg off from wifi after successfull initialization

Miroslavgapa, 2023-01-06:
### Describe the issue

Hi, i mean i found a bug with which i am dealing already 2 days.  When i succesfully login to inverter AP everything seems fine but after app 3 minutes - wifi is log off. After that integration is showing initializing but is not able to connect through pymodbus to inverter.

I try latest beta 7 version.

2023-01-06 19:24:42.704 WARNING (MainThread) [pymodbus.client.tcp] Failed to connect: 
2023-01-06 19:24:43.903 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-01-06 19:24:43.903 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 123, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 85, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 198, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 233, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 313, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 278, in _do_read
    raise ConnectionException(message)
huawei_solar.exceptions.ConnectionException: Modbus client is not connected to the inverter.
2023-01-06 19:24:43.908 WARNING (MainThread) [homeassistant.config_entries] Config entry 'SUN2000-8KTL-M1 01074300-002' for huawei_solar integration not ready yet: Modbus client is not connected to the inverter.; Retrying in background
2023-01-06 19:24:44.435 WARNING (MainThread) [pymodbus.client.tcp] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 6607, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 6607)
2023-01-06 19:24:50.033 WARNING (MainThread) [pymodbus.client.tcp] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 6607, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 6607)
2023-01-06 19:24:50.114 WARNING (MainThread) [pymodbus.client.tcp] Failed to connect: 
2023-01-06 19:24:51.407 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-01-06 19:24:51.408 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 123, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 85, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 198, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 233, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 313, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 278, in _do_read
    raise ConnectionException(message)
huawei_solar.exceptions.ConnectionException: Modbus client is not connected to the inverter.

### Bescribe your Huawei Solar Setup

Inverter Type: Huawei SUN200L 8KTL M1
Inverter Firmware version:V100R001C00SPC146
SDongle present: yes but connection through him is not working, firmware V100R001C00SPC125
Power meter present: no
Battery: None
Battery Firmware version: none

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

I dont have there diagnostic package because initializing status of integration

### Upload your relevant debug logs

```text
2023-01-06 19:24:42.704 WARNING (MainThread) [pymodbus.client.tcp] Failed to connect: 
2023-01-06 19:24:43.903 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-01-06 19:24:43.903 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 123, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 85, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 198, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 233, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 313, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 278, in _do_read
    raise ConnectionException(message)
huawei_solar.exceptions.ConnectionException: Modbus client is not connected to the inverter.
2023-01-06 19:24:43.908 WARNING (MainThread) [homeassistant.config_entries] Config entry 'SUN2000-8KTL-M101074300-002' for huawei_solar integration not ready yet: Modbus client is not connected to the inverter.; Retrying in background
2023-01-06 19:24:44.435 WARNING (MainThread) [pymodbus.client.tcp] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 6607, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 6607)
2023-01-06 19:24:50.033 WARNING (MainThread) [pymodbus.client.tcp] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 6607, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 6607)
2023-01-06 19:24:50.114 WARNING (MainThread) [pymodbus.client.tcp] Failed to connect: 
2023-01-06 19:24:51.407 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-01-06 19:24:51.408 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 123, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 85, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 198, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 233, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 313, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 278, in _do_read
    raise ConnectionException(message)
huawei_solar.exceptions.ConnectionException: Modbus client is not connected to the inverter.
```

### Please confirm the following:

- [ ] The problem is still present in the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I increased the logging configuration, restarted HA, and have copied all relevant log lines from the 'Full Logs' into the textarea.

wlcrs, 2023-01-06:
This is with a probability of 99% a network problem. The integration will periodically retry to restore the connection, little more it can do.

Miroslavgapa, 2023-01-06:
hm i updated sdongle to version V100R001C00SPC133, nmap showing to me that the port is open on port 502 but connection unsucesfully. I am totaly lost and siting behind PC whole day.