wlcrs/huawei_solar #245: [Bug]: Entity using native unit of measurement 'None' which is not a valid unit for the device class

alecsmro, 2023-01-07:
### Describe the issue

Even everything seems to work properly (except a bit longer delayed sensors values) i've got this persistent warning in log:

Entity sensor.inverter_power_factor (<class 'custom_components.huawei_solar.sensor.HuaweiSolarSensorEntity'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('power_factor') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.

Entity sensor.power_meter_power_factor (<class 'custom_components.huawei_solar.sensor.HuaweiSolarSensorEntity'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('power_factor') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.

Entity sensor.power_meter_reactive_power_2 (<class 'custom_components.huawei_solar.sensor.HuaweiSolarSensorEntity'>) is using native unit of measurement 'kVarh' which is not a valid unit for the device class ('reactive_power') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1
Inverter Firmware version: V100R001C00SPC148
SDongle present: yes  / Firmware 
Power meter present: three phase
Battery: None
Battery Firmware version: NONE

Huawei Solar version 1.2.0b7 -  also tried 1.2.0b5

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.1.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Bucharest",
    "os_name": "Linux",
    "os_version": "5.15.76-v8",
    "supervisor": "2022.12.1",
    "host_os": "Home Assistant OS 9.4",
    "docker_version": "20.10.19",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "sonoff": {
      "version": "3.3.1",
      "requirements": [
        "pycryptodome>=3.6.6"
      ]
    },
    "hacs": {
      "version": "1.29.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "fullykiosk": {
      "version": "1.1.0",
      "requirements": [
        "python-fullykiosk==0.0.11"
      ]
    },
    "tapo_control": {
      "version": "4.2.1",
      "requirements": [
        "pytapo==2.8",
        "onvif-zeep-async==1.2.0"
[... 286 lines cut ...]
      "active_grid_C_power": [
        -61,
        "W"
      ]
    },
    "slave_0_config_data": {}
  }
}

### Upload your relevant debug logs

```text
Entity sensor.inverter_power_factor (<class 'custom_components.huawei_solar.sensor.HuaweiSolarSensorEntity'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('power_factor') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.

Entity sensor.power_meter_power_factor (<class 'custom_components.huawei_solar.sensor.HuaweiSolarSensorEntity'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('power_factor') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.

Entity sensor.power_meter_reactive_power_2 (<class 'custom_components.huawei_solar.sensor.HuaweiSolarSensorEntity'>) is using native unit of measurement 'kVarh' which is not a valid unit for the device class ('reactive_power') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
```

### Please confirm the following:

- [ ] The problem is still present in the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I increased the logging configuration, restarted HA, and have copied all relevant log lines from the 'Full Logs' into the textarea.

tarmor1, 2023-01-09:
Connected to https://github.com/home-assistant/core/pull/85287 and fixed in HA [2023.1.2](https://github.com/home-assistant/core/releases/tag/2023.1.2)

alecsmro, 2023-01-15:
@tarmor1 now it's fixed, like you said, but there is another one with a similar issue:
Entity sensor.power_meter_reactive_power_2 (<class 'custom_components.huawei_solar.sensor.HuaweiSolarSensorEntity'>) is using native unit of measurement 'kVarh' which is not a valid unit for the device class ('reactive_power') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.

wlcrs, 2023-01-19:
Fixed in 1.2.0

closed 2023-01-19 by wlcrs