wlcrs/huawei_solar #248: [Bug]: Periodic Dropping of a slave inverter

JohnMcLear, 2023-01-08:
### Describe the issue

Periodically my slave 3 drops off.  I have 1 master (0[dongle], 6[master],2[slave],3[slave])

Could it be a wiring issue?  

### Bescribe your Huawei Solar Setup

Inverter Type: 6KL1-1
Inverter Firmware version: 125
SDongle present: yes
Power meter present: single phase 
Battery: LUNA2000 10kWh x 3 (2 on each inverter)
Battery Firmware version: 130 afaik

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

[config_entry-huawei_solar-e7dae940ac3400b9f936d9adb24424b7.json.txt](https://github.com/wlcrs/huawei_solar/files/10368350/config_entry-huawei_solar-e7dae940ac3400b9f936d9adb24424b7.json.txt)

### Upload your relevant debug logs

```text
Logger: custom_components.huawei_solar
Source: helpers/update_coordinator.py:168
Integration: Huawei Solar
First occurred: 6 January 2023 at 08:11:11 (289 occurrences)
Last logged: 11:55:24

    Error fetching TA2270270016_data_update_coordinator data: Could not update TA2270270016 values: Modbus client is not connected to the inverter.
    Error fetching TA2270269658_data_update_coordinator data: Could not update TA2270269658 values: Got error while reading from register 32064 with length 52: Exception Response(131, 3, IllegalAddress)
    Timeout fetching TA2270269658_data_update_coordinator data
    Timeout fetching TA2270270016_data_update_coordinator data
    Timeout fetching 102180067169_data_update_coordinator data
```
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I increased the logging configuration, restarted HA, and have copied all relevant log lines from the 'Full Logs' into the textarea.

wlcrs, 2023-01-08:
From the limited information that you are giving me, it looks like a hardware problem. Otherwise it would also fail for the other slaves

bigmoby, 2023-02-21:
Hi @JohnMcLear the same issue in my father's HA installation but in the same my own installation it's ok. My and my father's Huawei solar plant are literally the same and in the same version of this integration, and the same dongle firmware. What's the in different thing betweene and my father? Probably the fact that I never access to the Solar Fusion app but my father very often! Could it be the cause?! I don't know! When the integration is off I have to reboot the dongle (I've created a rest API calling from HA) and then I have to restart the integration itself.

JohnMcLear, 2023-02-24:
> When the integration is off I have to reboot the dongle (I've created a rest API calling from HA) and then I have to restart the integration itself.

Can you clarify what you mean here please by "the integration is off", do you mean "unavailable"?

Also what steps do you take to "reboot the dongle"?

I'm going to uninstall the "FusionSolar" app and I'll see if things improve, I doubt it but I'm open minded :)

bigmoby, 2023-02-24:
> > When the integration is off I have to reboot the dongle (I've created a rest API calling from HA) and then I have to restart the integration itself.
> 
> Can you clarify what you mean here please by "the integration is off", do you mean showing as "unavailable" in Home Assistant?
> 
> Also what steps do you take to "reboot the dongle"?
> 
> I'm going to uninstall the "FusionSolar" app and I'll see if things improve, I doubt it but I'm open minded :)

Hi @JohnMcLear , sure I mean the integration appears to me as unavailable (and all the meters related on that of course).

In order to reboot the dongle please try to call this as a service:

```
rest_command:
  reset_ew11:
    url: http://[your-dongle-ip]/cmd
    method: POST
    username: !secret ew_11_username
    password: !secret ew_11_password
    payload: 'msg={"CID":20003,"PL":{}}'
    content_type:  'application/json; charset=utf-8'
```

And after that I have to restart the integration (in Devices menu).
I hope it could be useful to you 👍

ikxdf, 2023-02-25:
Hi,
thanks for your great work. Me, and some other friends, have the same issue with this integration. It works well for 3-4 days and after that the starts to have 5-6 Hours of delay and the inverter result disconnected from cloud. When we notice that stops the integration and all start to work normaly again. 
we have basically two type of config:
1. Raspberry as a Wi-Fi bridge 
2. Home assistant directly Connect with wifi

seem like the integration is sendigs a lot of request to the inverter and the inverter slow down.

thanks in advance

JohnMcLear, 2023-03-02:
Theory: At some point someone (I forgot who) speculated that if the fusion solar app was uninstalled from a phone it would have an impact on availability.  

Conclusion: **There is no impact from having the app installed/not installed on Inverter "Availability"**

I'm going to attempt the restart script as per above next.

wlcrs, 2023-03-02:
> seem like the integration is sendigs a lot of request to the inverter and the inverter slow down.

Having multiple slaves will of course add an extra burden on the master to process all the update requests. It is very well possible that it's too much for the system to handle.

Does lowering the polling frequency solve the problem? Either follow the instructions here: https://github.com/wlcrs/huawei_solar#inverter-polling-frequency, or change [the relevant constants in `const.py`]( https://github.com/wlcrs/huawei_solar/blob/main/const.py#L19)

JohnMcLear, 2023-03-02:
@wlcrs I'll take a look at that today as well as the restart script.  Thanks.

ikxdf, 2023-03-02:
> Having multiple slaves will of course add an extra burden on the master to process all the update requests. It is very well possible that it's too much for the system to handle.
>
> Does lowering the [...]

I was thinking the same: the system have some  problems to handle all this requests.

i have notice that. After a inverter and cloud Connectivity down caused by this integration i have a 4hours of data delay on the cloud. This delay slowly disappear in 3 days with integration powered off. Inverter reboot have no effect. Seems like the inverter have ha cache and with the integration polling this cache become huge and hard to handle for it. 

Hope this can help.

wlcrs, 2023-03-02:
> I tried using the example automation but that didn't work.

Make sure that the target entity_id exists. The naming of entities has changed due to HA policy changes, so it depends on when you first installed the integration.  You can use the Developer Tools -> Services to make a valid automation action:

[image]

You'll need to add an entity from every inverter that you have

After clicking on YAML Mode you will receive the yaml code to use:

[image]

JohnMcLear, 2023-03-02:
Good point, I edited my above comment.

JohnMcLear, 2023-03-02:
> rest_command:
>   reset_ew11:
>     url: http://[your-dongle-ip]/cmd
>     method: POST
>     username: !secret ew_11_username
>     password: !secret ew_11_password
>     payload: 'msg={"CID":20003,"PL":{}}'
>     content_type:  'application/json; charset=utf-8'

Afaik I think the default username is ``admin`` and ``00000a`` here.

```
rest_command:
  reset_inverter_dongle:
    url: http://SUN2000-102180067169.lan/cmd
    method: POST
    username: admin
    password: 00000a
    payload: 'msg={"CID":20003,"PL":{}}'
    content_type:  'application/json; charset=utf-8'
```

For example would be correct if the default username and pass is still set?  cc @bigmoby

bigmoby, 2023-03-03:
> > rest_command:
> > reset_ew11:
> > url: http://[your-dongle-ip]/cmd
> > method: POST
> > username: !secret ew_11_username
> > password: !secret ew_11_password
> > payload: 'msg={"CID":20003,"PL":{}}'
> > content_type:  'application/json; charset=utf-8'
> 
> Afaik I think the default username is `admin` and `00000a` here.
> 

Hi @JohnMcLear I remember that the username and password for the Dongle are: admin/admin.

JohnMcLear, 2023-03-03:
note that I'm getting Connection refused, not authentication failed but I'll try admin/admin

JohnMcLear, 2023-03-03:
`` Client error. Url: http://SDongleA-HV2170115623.lan/cmd. Error: Cannot connect to host sdonglea-hv2170115623.lan:80 ssl:default [Connect call failed ('10.0.0.100', 80)] ``

JohnMcLear, 2023-03-04:
@ikxdf @wlcrs Just by way of initial update

Theory: Less frequent polling will lead to less incorrect ``Unavailable`` values.
**Prelim Conclusion: Changing to less frequent polling does not solve the problem.**  I only have two days of data but given the patterns of "Unavailable" look exactly the same I'm going to make the preliminary conclusion that reducing the poll rate to /1 doesn't fix the problem.  I'm going to attempt /2 minutes now (4th of March).

[image]

JohnMcLear, 2023-03-05:
[image]

``/2`` didn't work, going to attempt ``/5``

ikxdf, 2023-03-06:
> [image]
> 
> `/2` didn't work, going to attempt `/5`

I have a friend and eh is using sucessfully your integration but his inverter is not connected to the cloud. No delay and no inverter issues.

JohnMcLear, 2023-03-07:
I can confirm that adjusting the frequency of querying the inverters has no notable impact on the availability of an inverter.  I tested at ``/1``, ``/2`` & ``/5``

[image]

So this issue is related to something different, any suggestions for alternative approaches @wlcrs ?

wlcrs, 2023-03-07:
Thank you for your clear reporting. 

Can you enable debug logging for a while on this integration (available in the same menu where you fetched the diagnostics file). Once the problem has presented itself you can disable it again and upload the resulting file.

thbiela, 2023-03-09:
I think I have the same error. Sometimes I get timeouts and the entities become unavailable. Otherwise the integration works fine. I have two inverters, a battery and a power meter connected.

```
Logger: custom_components.huawei_solar
Source: helpers/update_coordinator.py:215
Integration: Huawei Solar
First occurred: 8. März 2023 um 21:33:20 (78 occurrences)
Last logged: 09:10:22

Error fetching HV21B0270250_data_update_coordinator data: Could not update HV21<redacted> values: Got error while reading from register 37760 with length 28: Exception Response(131, 3, SlaveFailure)
Error fetching HV2110009960_data_update_coordinator data: Could not update HV21<redacted>  values: Got error while reading from register 32000 with length 20: Exception Response(131, 3, SlaveFailure)
Timeout fetching HV211<redacted>_data_update_coordinator data
Timeout fetching HV21B<redacted>_configuration_update_coordinator data
Timeout fetching HV21B<redacted>_data_update_coordinator data
```

I have disabled automatic polling for updates and created an automation script. But I have noticed, that even with disabled polling, the values are still being updated (for example the power meter values). I have 109 entities that are being read by the integration, I think that is too much.

[image]

This is my automation script:

```
alias: Huawei Update Trigger
description: ""
trigger:
  - platform: time_pattern
    seconds: /20
condition: []
action:
  - service: homeassistant.update_entity
    data: {}
    target:
      entity_id:
        - sensor.battery_charge_discharge_power_2
        - sensor.inverter_input_power_2
        - sensor.inverter_active_power_1
        - sensor.inverter_active_power_2
        - sensor.inverter_daily_yield_1
        - sensor.battery_day_charge_2
        - sensor.battery_day_discharge_2
        - sensor.battery_state_of_capacity_2
        - switch.battery_charge_from_grid_2
mode: single
```

Dongle:

SDongleA-05
Version V100R001C00SPC133
Connected to 5KTL
Battery

Luna2000 5kWh
Version V100R002C00SPC116
Connected to 10KTL
5KTL

SUN2000-5KTL-M1
Version V100R001C00SPC148
Modbus ID address 1
No battery
10KTL

SUN2000-10KTL-M1
Version V100R001C00SPC150
Modbus ID address 2
Battery

[config_entry-huawei_solar-8f917a2ab63c2ca9b71f691605f3713d.json.txt](https://github.com/wlcrs/huawei_solar/files/10929450/config_entry-huawei_solar-8f917a2ab63c2ca9b71f691605f3713d.json.txt)
[home-assistant_huawei_solar_2023-03-09T08-37-14.051Z.log](https://github.com/wlcrs/huawei_solar/files/10929451/home-assistant_huawei_solar_2023-03-09T08-37-14.051Z.log)

JohnMcLear, 2023-03-09:
I enabled debug logs at 14:18 pm on the 9th of Mar, I'll come back to you once I have something useful @wlcrs - - thanks

wlcrs, 2023-03-09:
Thanks @thbiela for your logs. I see some SlaveFailure responses, which are different from SlaveBusy errors that I already handle well. I can/should more aggressively retry the calls to the inverter in case of a SlaveFailure.

Note that this is clearly an inverter-side issue, and that I cannot guarantee any improvement. There is already so much retrying going on at multiple levels in the process...

I'll also wait for John to come back with this logs to verify if we're seeing the same errors appearing in his case.

thbiela, 2023-03-09:
Thanks for looking into this. Do you know why disabling polling does not work?

All 109 entities are still being updated even if I disable it here:

[image]

wlcrs, 2023-03-09:
No.

thbiela, 2023-03-09:
OK, after disabling my automation for polling, the entities stopped refreshing. So it has to do with my automation. Even if I try to refresh only one entity, all entities are being polled from the inverter. Is this behaviour intended? If yes, could you tell me if disabling all entities that I do not need will reduce the amount of data that is being read by the integration? Many thanks.

wlcrs, 2023-03-09:
I leverage the DataUpdateCoordinator from Home Assistant to optimize the updating of entities. All registers are updated in a batched fashion, independent of if they are disabled are not. Only if you have multiple inverters it can make sense to disable all entities of one of the inverters to stop it being queried.

thbiela, 2023-03-09:
So currently the only solution for me seems to be to clone the huawei-solar repository at gitlab and remove the not needed values from the arrays at the bottom of this file, correct?  https://gitlab.com/Emilv2/huawei-solar/-/blob/develop/src/huawei_solar/bridge.py 

I think that polling these 109 entites could be too much to handle and that this is the reason for the error.

wlcrs, 2023-03-09:
Yes. Feel free to experiment and report back!

JohnMcLear, 2023-03-10:
Contents of 

``cat home-assistant_huawei_solar_2023-03-10T10-47-19.729Z.log | grep huawei > wlcr.txt``

and
[wlcr.txt](https://github.com/wlcrs/huawei_solar/files/10941261/wlcr.txt)

``cat home-assistant_huawei_solar_2023-03-10T10-47-19.729Z.log | grep inverter > inverter.txt``
[inverter.txt](https://github.com/wlcrs/huawei_solar/files/10941264/inverter.txt)

JohnMcLear, 2023-03-10:
RE batch collection  or # of entities being an issue, one approach to testing this theory could be to stagger data gathering in an automation IE 
minutes: 1,4,7,10... for inverter 1
minutes: 2,5,8,11... for inverter 2
minutes: 3,6,9,12... for inverter 3

I can run this test next week if needs be, for now it appears I don't get a "busy" error but a "fails" fwiw @wlcrs

ikxdf, 2023-03-10:
> RE batch collection or # of entities being an issue, one approach to testing this theory could be to stagger data gathering in an automation IE minutes: 1,4,7,10... for inverter 1 minutes: [...]

@wlcrs 
The problem is not related at the number of iverter. I have only one inverter and i have the issue. i am in a group and some users, with one inverter, are a le to use the integration for >1 week without delay or issues without cloud sync. Other users, with cloud sync, can use the integration for only 3 or 4 days. After that the cloud result disconnected and when stop the integration the cloud sync restare with 2 Hours of delay of data. This delay on the cloud will be reduced to 0 after 2 days. Stop and start the inverter have no effects to resolve the delay.

wlcrs, 2023-03-10:
@JohnMcLear you filtered out the  stacktraces with your grep. Can you please provide a sample containing a few full stack traces at least?

JohnMcLear, 2023-03-10:
@wlcrs  - I cherry picked some out for you, I think #1 and #3 might be duplicates but I left it in anyway.

```
2023-03-09 15:53:21.224 ERROR (MainThread) [custom_components.huawei_solar] Unexpected exception from <bound method DataUpdateCoordinator.async_refresh of <custom_components.huawei_solar.H
uaweiSolarUpdateCoordinator object at 0xffff80614eb0>>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/debounce.py", line 109, in _handle_timer_finish
    await task
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 215, in async_refresh
    await self._async_refresh(log_failures=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 330, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 135, in async_update_listeners
    update_callback()
  File "/config/custom_components/huawei_solar/sensor.py", line 851, in _handle_coordinator_update
    value = self.coordinator.data[self._register_key].value
KeyError: 'power_meter_active_power'
```
```
2023-03-09 15:56:23.405 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{{ (((states('sensor.battery_state_of_capacity_1') | float) + (states('s
ensor.battery_state_of_capacity_2') | float) + (states('sensor.battery_state_of_capacity_3') | float))/3) | round(2)}}")
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1771, in forgiving_float_filter
    return float(value)
ValueError: could not convert string to float: 'unknown'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 458, in async_render
    render_result = _render_with_context(self.template, compiled, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2007, in _render_with_context
    return template.render(**kwargs)
  File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 1301, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 936, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1774, in forgiving_float_filter
    raise_no_default("float", value)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1472, in raise_no_default
    raise ValueError(
ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (((states('sensor.battery_state_of_capacity_1') | float) + (states('sensor.battery_state_of_capaci
ty_2') | float) + (states('sensor.battery_state_of_capacity_3') | float))/3) | round(2)}}' but no default was specified

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 576, in async_render_to_info
    render_info._result = self.async_render(variables, strict=strict, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 460, in async_render
    raise TemplateError(err) from err
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (((states('sensor.battery_state_of_capacity_1') | float) +
 (states('sensor.battery_state_of_capacity_2') | float) + (states('sensor.battery_state_of_capacity_3') | float))/3) | round(2)}}' but no default was specified
2023-03-09 15:56:23.422 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ 
(((states('sensor.battery_state_of_capacity_1') | float) + (states('sensor.battery_state_of_capacity_2') | float) + (states('sensor.battery_state_of_capacity_3') | float))/3) | round(2)}}'
 but no default was specified') while processing template 'Template("{{ (((states('sensor.battery_state_of_capacity_1') | float) + (states('sensor.battery_state_of_capacity_2') | float) + 
(states('sensor.battery_state_of_capacity_3') | float))/3) | round(2)}}")' for attribute '_attr_native_value' in entity 'sensor.solarbattery'
2023-03-09 15:56:23.430 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{{ (states('sensor.battery_charge_discharge_power_1') | float) + (states
('sensor.battery_charge_discharge_power_1') | float) + (states('sensor.battery_charge_discharge_power_3') | float)}}")
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1771, in forgiving_float_filter
    return float(value)
ValueError: could not convert string to float: 'unknown'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 458, in async_render
    render_result = _render_with_context(self.template, compiled, **kwargs)
[... 28 lines cut ...]

```
2023-03-09 16:00:29.625 ERROR (MainThread) [homeassistant.helpers.entity] Update for sensor.inverter_daily_yield_2 fails
Traceback (most recent call last):
[... cut ...]
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 330, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 135, in async_update_listeners
    update_callback()
  File "/config/custom_components/huawei_solar/sensor.py", line 851, in _handle_coordinator_update
    value = self.coordinator.data[self._register_key].value
KeyError: 'power_meter_active_power'
2023-03-09 16:00:29.632 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
```

wlcrs, 2023-03-10:
Hi all, can you please test again with 1.2.6 beta 1?

JohnMcLear, 2023-03-10:
Thanks for the fix man I'll test asap and feedback.  ETA 5 days

JohnMcLear, 2023-03-12:
[image]

I'll grab logs later

JohnMcLear, 2023-03-13:
[image]

[cat home-assistant_huawei_solar_2023-03-13T11-53-27.712Z.log | grep huawei | grep -v register > 1.log
](https://github.com/wlcrs/huawei_solar/files/10956980/1.log)

Oh wait I'm an idiot, I had automatic checking on AND my automation was running so probably spamming the inverter hard.  I'm going to disable my automation.

JohnMcLear, 2023-03-14:
[image]

Minor feedback, still some drops today.  Only a few events so I'll provide more feedback on Thurs including logs.

wlcrs, 2023-03-16:
Hi @thbiela , an `IllegalAddress` modbus error normally signals a permanent failure, as it says as much as "the registers that you are requesting do not exist at all". There is no point in aggressively retrying to read from the inverter at that point. The sensors going to "Unavailable" at that point is correct behavior in my opinion. 

Can you please leave the debug logging enabled for a while to capture some more information? Now I see that you enabled it only very shortly, meaning that it didn't capture anything useful.

I see another very worrying warning in your logs:

```
2023-03-14 08:55:39.481 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.SlaveBusyException: Mismatch between number of requested registers (52) and number of received registers (1))
```

This means that the integration probably receives answers to modbus read requests it didn't do. Do you have anything else connected to your installation on the RS485 ports for instance?

thbiela, 2023-03-16:
Hi, I have enabled debug logging and will disable it in a few hours. I have nothing else connected to the RS485 ports, only the huawei solar integration is accessing the inverter via modbus TCP.

JohnMcLear, 2023-03-16:
Bug is confirmed still present

[image]

RE Modbus, I have the huawei energy sensor thingy.

``cat home-assistant_huawei_solar_2023-03-16T12-57-25.244Z.log  | grep solar | grep -v Reading | grep 16 > 16.txt`` [
[16.txt](https://github.com/wlcrs/huawei_solar/files/10990983/16.txt)
](url)

I have "Enable polling for updates" enabled and my automation to get the data disabled.

```
2023-03-15 06:11:20.472 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-15 06:11:20.473 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'asyncio.exceptions.TimeoutError'>: backing off reading for 1.0 seconds after 1 tries
2023-03-15 06:11:27.449 ERROR (MainThread) [custom_components.huawei_solar] Timeout fetching TA2270270016_data_update_coordinator data
```

This appears to be the main stack trace.  Nothing notable after

ikxdf, 2023-03-17:
> Interesting! Thank you for your input!
> 
> Sadly this is something that pyModbus does not support for the moment.
> 
> Before I do the effort of creating a PR, can you test the following patch please?
> 
> In the **init**.py file of the integration, can you please paste the following lines just above `async def async_setup_entry`:
> 
> ```python
> patch_applied = False
> 
> 
> def monkey_patch_delay_after_connect():
>     """
>     Dirty fix to force a delay after each connection
>     """
> 
>     global patch_applied
> 
>     if patch_applied:
>         return
> 
>     from pymodbus.client.base import ModbusBaseClient
>     from time import sleep
> 
>     orig_connection_made = ModbusBaseClient.connection_made 
> 
>     def patched_connection_made(self, transport):
>         orig_connection_made(self, transport)
>         sleep(2)
> 
>     ModbusBaseClient.connection_made  = patched_connection_made
>     patch_applied = True
> ```
> 
> Restart HA after making the changes. Feel free to play around with the sleep time: `sleep(2) means waiting for 2 seconds, which [this OpenHab user](https://community.openhab.org/t/reading-data-from-huawei-inverter-sun-2000-3ktl-10ktl-via-modbus-tcp-and-rtu/87670/118) suggests as a good value.

My inverter was unresponsive after 4 or 5 days. I will made the changes this night and report back after 5 days.

thank you

ikxdf, 2023-03-23:
> @ikxdf did this patch resolve the issue for you? I haven't tested it on my setup.

I am at the 6th day without any issue: no delay and no cloud disconnection.
I have update the inverter at latest version before apply the fix…i cant said if the good result is because fix or update. 

At latest firmware update There is another problem (i have already open an issue but now is closed…). Now the integration show energumeni consumption at yield energy sensor.
This is the update.
[image]

JohnMcLear, 2023-03-23:
Oh actually I noticed two of my inverters are on 125 not 130..  I guess I have a decent amount of updating to do today...

wlcrs, 2023-03-25:
I've released 1.2.6 beta 3 which now has a 1.5 seconds delay built in after each (re)connection to the inverter. This makes the patch that I shared earlier in this issue unnecessary. 

Please report back if this solves the problem for you.

If necessary, you can change the delay by adding the following to the top of `__init__.py`:

```
import huawei_solar.modbus
huawei_solar.modbus.WAIT_ON_CONNECT = 2000 # in milliseconds
```

Please let me know if you do so, and which number works for you.

inigoml, 2023-03-26:
Same problem here since months ago. Was triggered after Huawei moved TCP port. Rising refresh time helps, but does not fix the problem. Will test this patch too and will report.

JohnMcLear, 2023-03-29:
[image]

Note how the unavailable is often 20+ mins now which is super weird...  

I can't grab logs right now but will in a few days.

inigoml, 2023-03-29:
After upgrade to 1.2.6b6, Fusionsolar keeps disconnecting. So, didn't work, unfortunately.

My current configuration is:
- AP in client mode connected to Huawei inverter wifi. Port NATting applied. 4 meters away from inverter, no obstacles.
- AP connected via ethernet to local LAN. 
So, working like a ethernet to wifi router.

So, should not be related with connectivity issues.

My theory is some kind of bug at inverter side that keeps TCP connections in a NON-CLOSE status that eventually affects both huawei_solar and fusionsolar connectivity.

inigoml, 2023-03-29:
> Please try with 2 seconds. With two seconds is working for me.

OK. Modified __init__.py
Will test again for a couple of days.

wlcrs, 2023-03-29:
> Please try with 2 seconds. With two seconds is working for me.

But with 1.5 seconds you were still having issues?

ikxdf, 2023-03-29:
> > Please try with 2 seconds. With two seconds is working for me.
> 
> But with 1.5 seconds you were still having issues?

I have never used 1.5s

inigoml, 2023-04-01:
> > Please try with 2 seconds. With two seconds is working for me.
> 
> OK. Modified **init**.py Will test again for a couple of days.

Bad luck. Today FusionSolar locked again with increased timouet to 2s.

`import huawei_solar.modbus
huawei_solar.modbus.WAIT_ON_CONNECT = 2000 # in milliseconds`

ikxdf, 2023-04-01:
> > > Please try with 2 seconds. With two seconds is working for me.
> > 
> > 
> > OK. Modified **init**.py Will test again for a couple of days.
> 
> Bad luck. Today FusionSolar locked again with increased timouet to 2s.
> 
> `import huawei_solar.modbus huawei_solar.modbus.WAIT_ON_CONNECT = 2000 # in milliseconds`

What version of inverter and battery firmware do you have?

inigoml, 2023-04-01:
> What version of inverter and battery firmware do you have?

Latest!

Inverter: SPC130
Battery: SPC119

ikxdf, 2023-04-01:
> > What version of inverter and battery firmware do you have?
> 
> Latest!
> 
> Inverter: SPC130 Battery: SPC119

I have the same firmware but i am with 2.5.0 integration version and this patch.

https://github.com/wlcrs/huawei_solar/issues/248#issuecomment-1474034533

ikxdf, 2023-04-05:
sorry, bad news. with 2 seconds sleep the inverter stop working after 10 days. @wlcrs what happen if I increase the sleep to 10 seconds? any ideas?

JohnMcLear, 2023-04-12:
Sorry about the radio silence, this is the performance on b4:

[image]

I'm going to test b5.

As I updated to b4 the patch was no longer applied.  If you want me to test w/ patch I can just bump me to let me know.

wlcrs, 2023-04-12:
Hi all,

As the wait-after-connect did not solve the issue completely, I've implemented a "nuclear option": if the retry-mechanism receives 2 timeouts in a row, it will disconnect and try to reconnect to the inverter before continuing with trying to read from the inverter.

You can test this by updating the huawei-solar library version to **2.2.7b6** by changing the following line in `manifest.json`  and restarting your HA:

```
 "requirements": ["huawei-solar==2.2.7b6"],
```

If this doesn't improve things, I honestly do not have a clue on what else I can do...

inigoml, 2023-04-13:
Testing right now with huawei-solar library. Will report if solves the problem.
Wait-after-connect and raising update time to 300s has keep my FusionSolar working for 9 days.... so, it does not fix the problem but helps.

[19 more replies]

closed 2023-05-08 by wlcrs