wlcrs/huawei_solar #258: [Feature Request]: can you explain what this is.?

Rhysgh, 2023-01-13:
### Describe your feature request

Huawei Solar: Set Fixed Charge Periods

Hi, sorry. Can you point me in the direction of more info regarding this service? i was able to set it using HA, but I cannot find the settings anywhere in my battery settings.

### Proper usage

- [X] I confirm that this is not a bug report

wlcrs, 2023-01-13:
I implemented it as documented in the Modbus Definitions. I don't know anything more

closed 2023-01-13 by wlcrs

Rhysgh, 2023-01-13:
would you be able to point me at the definition document I couldn't find it?

wlcrs, 2023-01-14:
Please stop making low-effort requests. I release this software for free, that doesn't mean I need to provide support to everyone who asks for this for free.

Consider this your final warning. I will block you from interacting with me if necessary.

Rhysgh, 2023-01-14:
Only asked a question that I could not find an answer to in any literature having read and searched both the API and Modbus documents provided by Huawei. I thought maybe you had made a mistake in the naming of it and that my feedback may have helped improve what a is fantastic integration, it deserves sponsorship and I will contribute in due course.

I'll re-read the documents again and see if I can work it out. Thank you for your time.