wlcrs/huawei_solar #260: [Feature Request]: Possibility to configure timeout period of modbus requests

letsid, 2023-01-18:
### Describe your feature request

Hi wlcrs,

Thank you very much for your work in this integration.
I am wondering if you can add the possibility to configure the timeout period of Modbus requests.
I would like to see this as I am planning to add an electrical car to my household and I want to surplus charge it with evcc.io
As both, your integration and evcc need connection to the modbus I need to use a modbus proxy.
As soon as I use both, your integration and evcc the integration starts to drop out for a while and does not give any values back.

You can see a similar issue here where the developer of EVCC suggested looser timeouts (Topic is in german, maybe Google Translate can help you to understand that?)
https://github.com/evcc-io/evcc/discussions/5094#discussioncomment-4291407

br,
letsid

### Proper usage

- [X] I confirm that this is not a bug report

wlcrs, 2023-01-18:
Rather than making this configurable, I think it's better to just have a sane default? Otherwise it just makes it harder to use the integration.

I would need some input on what a sane default might be. Just increasing it to a big number can potentially create issues for other users.

A very hacky way to change the timeout, but which will allow you to experiment with it, is to change the global defaults of pyModbus by adding the following lines to the top of `huawei_solar/__init__.py`:

```py
from pymodbus.constants import Defaults

Defaults.Timeout = 5
```

This changes the current default: https://github.com/pymodbus-dev/pymodbus/blob/e459ca52b5ee7cf4dab56585654351fff2a49738/pymodbus/constants.py#L140 . 

(I did not test this code)

Please play around with the timeout value and report back your findings

letsid, 2023-01-19:
Hi, 
 
 Code works! 
 before changing the timeout, I did disable all sensors from the integration that I am not using and changed the polling rate to 1 minute. That helped a little as well as.
 I then changed the timeout and had some drop-outs with a timeout of 7s and then put it to 10s and overnight I did not have any drop out overnight now.

I think my issue is very isolated, and most users of your integration can live with the default 3s timeout. 
For that reason, an optional method to configure the timeout value to a higher one would be nice I think. 
I don't know if it is possible, but it would be nice to have an option to add something like this to configuration.yaml

```
huawei_solar:
   timeout: 10
```
Which then overrides the default timeout if set

If this is too much effort for the gain - I can live with changing the timeout manually in the huawei_solar/__init__.py when the integration updates.

wlcrs, 2023-01-19:
It is not possible to mix legacy and config flow configurations. 

Can you also test timeouts of 8 and 9 seconds? Feel free to take a few days to test for stability issues.

letsid, 2023-01-26:
I played around wit different values and even with 10 sec's it sometimes has issues. But it is much better than before. 
Thank you!

wlcrs, 2023-01-26:
I'll keep it on 10 seconds in that case. In case of wrong connection parameters it now already takes one minute to finish detection... (6 retries * 10 seconds)

closed 2023-01-26 by wlcrs

mptr8956, 2024-04-29:
Hi, tried to add this to my init.py but the integration then refuses to load when homeassistant starts. where exactly on top do I need to put the code?